espressif/esp-idf #10255: Esp32-C2 boudrate output error

Eltopinovatif, 2022-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp32-c2 Uart error Does not match, standard boudrate is 115200 but actual is 74880. Test for example "Uart event" boudrate is changed to a value other than 115200 resulting in corrupt data output. can anyone help with the above problem
[image]

nopnop2002, 2022-11-29:
Try changing the crystal frequency to 26MHz in the following menuconfig menu:

[image]

Eltopinovatif, 2022-11-30:
Thanks @ nopnop 2002.
the problem was resolved after the Xtal frequency config setting was changed to 26MHZ.
[image]

closed 2022-11-30 by Eltopinovatif