espressif/esp-idf #10256: wake up timer partially respected

sergioaiello, 2022-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

esp_sleep_enable_timer_wakeup
it uses uint64_t
I should be able to set a time of millions of billions of us

### What is the actual behavior?

I set 2,148,000,000 (or higher), it stops but it does not sleep (it reboots immediately: "DEEPSLEEP_RESET")
If I reduce to 2,147,500,000 it works as expected. 

#define INACTIVEPERIOD 2147.5
sleepDuration = ( INACTIVEPERIOD * 1000000);
esp_sleep_enable_timer_wakeup( sleepDuration );

Since it is a very easy to reproduce, I would think either something I made wrong, or a similar case already exists. I'd be sorry for both.

### Steps to reproduce.

1.  working setup / set the code:
#define INACTIVEPERIOD 2147.5
sleepDuration = ( INACTIVEPERIOD * 1000000);
esp_sleep_enable_timer_wakeup( sleepDuration );
2.  not working setup / set the code:
#define INACTIVEPERIOD 2148
sleepDuration = ( INACTIVEPERIOD * 1000000);
esp_sleep_enable_timer_wakeup( sleepDuration );

igrr, 2022-11-27:
@sergioaiello The issue you are running into is that in C the expression `2148 * 1000000` is evaluated as signed integer multiplication. Signed ints are from -2147483648 to 2147483647. The multiplication above produces a signed integer overflow, and the resulting value is -2146967296.

You need to convert one of the operands to 64-bit int to get the result you expect:
```c++
#define INACTIVEPERIOD 2148
sleepDuration = ( INACTIVEPERIOD * 1000000ULL);
```

By the way, the compiler should produce a warning about the code you have posted, along the lines of:
```
warning: integer overflow in expression of type 'int' results in '-2146967296' [-Woverflow]
   14 |     sleepDuration = ( INACTIVEPERIOD * 1000000);
```

sergioaiello, 2022-11-28:
Thank you and sorry for my mistake
No, I had only this warning "#warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
Tested with 3600 and worked flawlessly. 
I'm going to close this case.

closed 2022-11-28 by sergioaiello