espressif/esp-idf #10257: run idf.py -p /dev/ttyS5 flash error

BingxuShi, 2022-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Use idf.py flash to upload to the board

### What is the actual behavior?

Use idf.py flash to upload to the board and report an error

### Steps to reproduce.

Follow the ESP-IDF programming documentation, compile the hello world project, and report an error when downloading to the board
cmd：idf.py -p /dev/ttyS5 flash

### Build or installation Logs.

```plain
esptool.py v3.3.2-dev
Serial port /dev/ttyS5
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:22:d8:ea:e5:a0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Traceback (most recent call last):
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5399, in <module>
    _main()
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5392, in _main
    main()
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4814, in main
    esp.flash_set_parameters(flash_size_bytes(args.flash_size))
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 1056, in flash_set_parameters
    self.check_command("set SPI params", ESP32ROM.ESP_SPI_SET_PARAMS,
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 468, in command
    p = self.read()
  File "/mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
CMake Error at run_serial_tool.cmake:56 (message):
  /home/bingxushi/.espressif/python_env/idf4.4_py3.10_env/bin/python
  /mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32
  failed

FAILED: CMakeFiles/flash /mnt/d/esp/esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash
cd /mnt/d/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/mnt/d/esp/esp-idf" -D SERIAL_TOOL="/home/bingxushi/.espressif/python_env/idf4.4_py3.10_env/bin/python /mnt/d/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/mnt/d/esp/esp-idf/examples/get-started/hello_world/build" -P /mnt/d/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

I'm using the Windows subsystem Ubuntu
My usart information
[image]

FanhuaCloud, 2022-12-03:
I have same ploblem, in IDF 5.0

dobairoland, 2022-12-06:
Hi @FanhuaCloud. The device shows up in Windows but not inside WSL. So this is a tunneling issues. I suggest you to search for the solution in the WSL documentation. Nothing can be done at the ESP-IDF side if the device is not showing up - it is like if you would have not connected and expect it to get flashed.

BTW, I don't think `COM5` will be translated to `/dev/ttyS5` (just based on the number). I would expect something like `/dev/ttyUSB0`.

closed 2022-12-06 by dobairoland

dobairoland, 2022-12-06:
Here are some steps which might help: https://esp32.com/viewtopic.php?t=8673

dobairoland, 2022-12-06:
There is a related request: https://github.com/microsoft/WSL/issues/7657