espressif/esp-idf #10258: SO_LINGER implementation does no propagate immediately RST to client when l_onoff set to 1 and l_linger set to 0

ves011, 2022-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Expected behavior
Closing the socket with linger enabled and linger timeout set to 0 should abort the connection immediately and send a TCP segment with RST flag set, so the peer is aware the connection is force dropped by the other side.
Any further read by the peer should return 0, or -1 with errno = ECONNRESET
Any further write should return -1 with errno = ECONNRESET or EPIPE  

Observed behavior
Closing the socket with linger enabled and linger timeout set to 0 send FIN segment to peer 
Peer send back ACK and is aware that a graceful shutdown is started.
Any further read by the peer returns 0 with errno = 0 -> which is fine.
The first write returns success (no of bytes written) and errno = 0 -> which is confusing;
Only after the first write peer receives RST segment and further writes returns -1 with errno = EPIPE 

environment
 - esp-idf: v4.4.3
 - linux kernel: 5.15.0-53-generic
 - esp32-wroom-32 dev kit
 
Steps to reproduce
Take tcp_server example provided in idf (\esp-idf-v4.4.3\examples\protocols\sockets\tcp_server) and modify tcp_server.c starting line 149
```
//do_retransmit(sock);
struct linger ling;
ling.l_linger = 0;
ling.l_onoff = 1;
shutdown(sock, 0);
err = setsockopt(sock, SOL_SOCKET, SO_LINGER, (const char*)&ling, sizeof(struct linger));
ESP_LOGI(TAG, "linger: %d %d", err, errno);
close(sock);
 ESP_LOGI(TAG, "close socket: %d %d", err, errno);
```
For the client side i have a small linux application which connects to the server and do some read/write
```
int main(void)
	{
	int sockfd, rc;
	struct sockaddr_in serv_addr;
	char message[32];
	strcpy(message, "test tcp server");
	bzero((char *) &serv_addr, sizeof(serv_addr));
	serv_addr.sin_family = AF_INET;
	serv_addr.sin_port = htons(3333);
	//192.168.1.52
	serv_addr.sin_addr.s_addr = htonl(0xc0a80134);
	while(1)
		{
		sockfd = socket(AF_INET, SOCK_STREAM, 0);
		if(sockfd > 0)
			{
			//usleep(10000);
			rc = connect(sockfd,(struct sockaddr *)&serv_addr, sizeof(serv_addr));
			if(!rc)
				{
				rc = recv(sockfd, &message, sizeof(message), 0);
				rc = send(sockfd, &message, sizeof(message), 0);
				rc = recv(sockfd, &message, sizeof(message), 0);
				rc = send(sockfd, &message, sizeof(message), 0);
				}
			close(sockfd);
			sleep(2);
			}
		}
	return 0;
	}
```
1. start the server on esp32, 
2. run linux app in debug mode and break the execution on the first recv
3. wait server app on esp32 to go back to listen mode
```
I (00:04:58.735) example: Socket listening
I (00:06:14.876) example: Socket accepted ip address: 192.168.1.71
I (00:06:14.877) example: linger: 0 0
I (00:06:14.880) example: close socket: 0 0
I (00:06:14.882) example: Socket listening
```
4. step by step on linux app and inspect rc and errno variables
 - after 1st recv: rc = 0, errno = 0
 - after 1st send: rc = 32, errno = 0
      --> here whireshark shows PSH and RST segments
 - after 2nd recv: rc = 0, errno = 0
 - after 2nd send: rc = -1 errno = 32

ESP-YJM, 2022-11-29:
@ves011 Can you please remove `shutdown(sock, 0);` code before set `SO_LINGER` option and try again.

ves011, 2022-11-29:
@ESP-YJM thx for looking into.
Tried several combinations with shutdown(soc,k,  0), shutdwon(sock, 2) and wo shutdown.
The behavior is the same.

ESP-YJM, 2022-11-29:
Hi @ves011 , i check with the LWIP code and find the reason. When you set `SO_LINGER` and close socket, LWIP does not directly send RST to disconnect, only when the socket has `unsent` or `unacked` data. The code logic please refer to https://github.com/espressif/esp-lwip/blob/33912690ea005e8ff05b2c579ebbf0a4189c3b00/src/api/api_msg.c#L1009. So in your test code, it meets the LWIP logic code and send `FIN` to peer. You can add `send(sock, "test", 4, 0);` code before close socket. This will send RST to peer. Because now socket has `unacked` data.

ves011, 2022-11-29:
Hi @ESP-YJM 
Yes. With send() before closing, RST segment is received on the client side.
However its not clear for me why unsent or unhacked data are needed in order to send RST.
The way i read RFC 1122, it doesnt  say anything about pending data
````
4.2.2.13  Closing a Connection: [RFC-793 Section 3.5]

            A TCP connection may terminate in two ways: (1) the normal
            TCP close sequence using a FIN handshake, and (2) an "abort"
            in which one or more RST segments are sent and the
            connection state is immediately discarded.  If a TCP
            connection is closed by the remote site, the local
            application MUST be informed whether it closed normally or
            was aborted.
````
Compiling and running the tcp_server app, on a linux machine, it sends RST wo the need of pending data.

ESP-YJM, 2022-11-29:
@ves011 I think `unsent` or `unhacked` data are needed in order to send RST，just the way lwip protocol stack implemented. Of course, send RST no need of `unsent` or `unhacked` data is also OK. Each protocol stack is implemented differently.

ves011, 2022-11-29:
OK, from my point of view, now its understood and will take it as an implementation specific.
Will close this topic as a non issue.

Thanks for your support!

closed 2022-11-29 by ves011