espressif/esp-idf #10259: `detect_python.sh` fails to detect correct Python binary

gregjesl, 2022-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1908-g439a709c42

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### What is the expected behavior?

`detect_python.sh` should detect Python version 3.7

### What is the actual behavior?

`detect_python.sh` returns version 3.6.9

### Steps to reproduce.

On a Linux system with Python version with both 3.6.9 and 3.7 installed, attempt to install the idf. 

### Build or installation Logs.

```plain
Detecting the Python interpreter
Checking "python3" ...
Checking "python" ...
Python 3.6.9
"python" has been detected
Checking Python compatibility
Traceback (most recent call last):
  File "/home/appveyor/projects/simpleson-l4m5a/esp-idf/tools/python_version_checker.py", line 45, in <module>
    check()
  File "/home/appveyor/projects/simpleson-l4m5a/esp-idf/tools/python_version_checker.py", line 39, in check
    _ver_to_str(OLDEST_PYTHON_SUPPORTED), _ver_to_str(sys.version_info[:3])
RuntimeError: ESP-IDF supports Python 3.7 or newer but you are using Python 3.6.9. Please upgrade your installation as described in the documentation.
Setting IDF_PATH to '/home/appveyor/projects/simpleson-l4m5a/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Checking "python" ...
Python 3.6.9
"python" has been detected
Checking Python compatibility
Traceback (most recent call last):
  File "/home/appveyor/projects/simpleson-l4m5a/esp-idf/tools/python_version_checker.py", line 45, in <module>
    check()
  File "/home/appveyor/projects/simpleson-l4m5a/esp-idf/tools/python_version_checker.py", line 39, in check
    _ver_to_str(OLDEST_PYTHON_SUPPORTED), _ver_to_str(sys.version_info[:3])
RuntimeError: ESP-IDF supports Python 3.7 or newer but you are using Python 3.6.9. Please upgrade your installation
```

### More Information.

Full log can be found [here on Appveyor](https://ci.appveyor.com/project/gregjesl/simpleson-l4m5a/builds/45501913#L303)

The [Appveyor Linux build environment](https://www.appveyor.com/docs/linux-images-software/) has multiple versions of Python installed. Looking at `detect_python.sh`, the script should iterate through the versions until it finds a suitable version; however, the script is exiting with an unsupported version, as shown here in the logs:
```
Python 3.6.9
"python" has been detected
```
I suspect the problem is here: 
https://github.com/espressif/esp-idf/blob/439a709c4298b0f613d09b47fc0b7b9728fc5733/tools/detect_python.sh#L20

gregjesl, 2022-11-28:
Root cause: by initializing `ESP_PYTHON` with `python`, the script will successfully return an unsupported version of Python if `python` is an unsupported version but no other supported versions are found.

dobairoland, 2022-11-28:
Hi @gregjesl. Thank you for your investigation.

Returning "python" is desired so even when there is no suitable version, the issue wouldn't be transformed to something else: "python script.py" would be attempted to run instead of "script.py" which could result in another issue - eg. script is not executable, not linked to the right Python.

Lets focus on why hasn't Python 3.7 being detected on your machine. `python3.7` is not accessible in the PATH? 

These are the Pythons checked (in this order):
```bash
for p_cmd in python3 python python3.7 python3.8 python3.9 python3.10 python3.11 python3.12; do
```

gregjesl, 2022-11-28:
Hello @dobairoland, thank you for your review. 

Based on your comment `detect_python.sh` is working as intended and that `python_version_checker.py` should be the script that determines if the installed version of python is a supported version. 

If that is the case, then I would assert that `detect_python.sh` and `python_version_checker.py` are largely redundant. I've submitted pull request #10260 that implements the desired behavior in `python_version_checker.py` in `detect_python.sh` and `detect_python.fish`. 

> Returning "python" is desired so even when there is no suitable version, the issue wouldn't be transformed to something else: "python script.py" would be attempted to run instead of "script.py" which could result in another issue - eg. script is not executable, not linked to the right Python.

The install script will eventually fail if the only version of Python installed is not a supported version. As implemented in #10260, `python_version_checker.py` becomes unnecessary and the install script will exit if no supported version of Python is found in `detect_python.sh` or `detect_python.fish`. 

The other approach I could see is keeping `python_version_checker.py` and removing the following:
https://github.com/espressif/esp-idf/blob/175e5fc260bedaa17d0e55788fa5c36ba3076a72/tools/detect_python.sh#L17-L20
https://github.com/espressif/esp-idf/blob/175e5fc260bedaa17d0e55788fa5c36ba3076a72/tools/detect_python.fish#L12-L15

 > Lets focus on why hasn't Python 3.7 being detected on your machine. python3.7 is not accessible in the PATH?

I can't find specific documentation on where the different Python binaries are installed on the Appveyor image. They provide the location of the virtual environments, but the IDF creates an virtual environment and Python does not support creating a virtual environment from a virtual environment. Virtual environment support for this IDF is probably a separate enhancement. 

My current work-around is to have my CI install script download and install Python 3.7.15.

dobairoland, 2022-11-28:
> If that is the case, then I would assert that detect_python.sh and python_version_checker.py are largely redundant. I've submitted pull request https://github.com/espressif/esp-idf/pull/10260 that implements the desired behavior

Yes, I've seen that PR but I'm sorry but that is not the desired behavior. You are right that the scripts have similarities but used for different purpose. The shell scripts intends to find something in the PATH which can be used (don't opt out if `python3` is not present but `python` is a suitable one; `pythonX.Y` are there for cases when Python installed additionally to meet ESP-IDF requirements). The PY script is invoked from the build system as well.

> They provide the location of the virtual environments, but the IDF creates an virtual environment and Python does not support creating a virtual environment from a virtual environment.

That is the problem. ESP-IDF doesn't directly support connecting to existing virtual environments (would result in error if attempted). The requirement is to have a globally available python installed (not virtual environment). 

In your case indeed installing a Python manually and using that with ESP-IDF is the way to go.

closed 2022-11-28 by dobairoland