espressif/esp-idf #10262: Using idf.py without network connectivity

markand, 2022-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-4037-g9b8c558e63-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Hi,

Working with idf.py on poor network connectivity seems complicated because idf.py seems to download updates by itself on startup. Is there a way to disable that? We are sometimes working on very low network connectivity (or sometimes not at all).

```
$ idf.py monitor
... hangs ...
^C
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt
Destination: /home/markand/.espressif/espidf.constraints.v5.0.txt.tmp
```

### What is the actual behavior?

Provide a way to avoid automatic downloading.

### Steps to reproduce.

Any command from idf.py without networking.

dobairoland, 2022-11-28:
Could you please try the latest v5.0 version? This has already been fixed in 737cdcbaa6fb7089791c2a954e8e96f4beb9c835

markand, 2022-11-28:
Thanks, I'll try in a couple of minutes.

closed 2023-02-01 by dobairoland