espressif/esp-idf #10264: Error in esp_http_server

laura-nat, 2022-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

don't know how to check it

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### What is the expected behavior?

I am following the Application Example in this link "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html" in the Arduino IDE and ESP32. I want to create a very simple webserver page in AP mode to print out on a webpage values read by a sensor.

### What is the actual behavior?

I am following the Application Example in this link "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html" in the Arduino IDE and ESP32. I always get the error

"assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)"

### Steps to reproduce.

just run the tutorial as it is .

ESP-YJM, 2022-11-29:
@laura-nat It seems your code not call `esp_netif_init()` to init netif and lwip protocol stack. You can run this example https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/simple.

hmalpani, 2022-12-06:
Closing this issue as the above suggestion should fix your problem. Please feel free to reopen if the issue persists.

closed 2022-12-06 by hmalpani