espressif/esp-idf #10265: httpd_sess_free_ctx in httpd_req_cleanup uses wrong context pointer

sforconi, 2022-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.x

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

The code should not crash when a previously allocated session context is reset to NULL in another uri handler.
As pointed in the documentation: "By default, if you change the sess_ctx in some URI handler, the http server will internally free the earlier context (if non NULL), after the URI handler returns."

### What is the actual behavior?

The firmware crashes.

### Steps to reproduce.

1. create a HTTP Server using esp_http_server API
2. create a uri handler where you setup a session context: `req->sess_ctx = malloc(...);`
3. create another uri handler where you clear the session context: `req->sess_ctx = NULL` 

### Debug Logs.

```plain
assert failed: heap_caps_free heap_caps.c:339 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace:0x40375f36:0x3fcca3c00x4037d7d9:0x3fcca3e0 0x4038565d:0x3fcca400 0x403761a3:0x3fcca520 0x4038568d:0x3fcca540 0x4205b56b:0x3fcca560 0x4205af05:0x3fcca580 0x4205b115:0x3fcca5b0 0x4205b9f8:0x3fcca600 0x42059d7c:0x3fcca620 0x420bbda7:0x3fcca650 0x4205a0b3:0x3fcca670 0x4205a152:0x3fcca6c0 0x40381759:0x3fcca6e0 
0x40375f36: panic_abort at /home/samuele/data/esp/esp-idf/components/esp_system/panic.c:402

0x4037d7d9: esp_system_abort at /home/samuele/data/esp/esp-idf/components/esp_system/esp_system.c:128

0x4038565d: __assert_func at /home/samuele/data/esp/esp-idf/components/newlib/assert.c:85

0x403761a3: heap_caps_free at /home/samuele/data/esp/esp-idf/components/heap/heap_caps.c:339 (discriminator 1)

0x4038568d: free at /home/samuele/data/esp/esp-idf/components/newlib/heap.c:39

0x4205b56b: httpd_sess_free_ctx at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:232

0x4205af05: httpd_req_cleanup at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_parse.c:697

0x4205b115: httpd_req_delete at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_parse.c:825

0x4205b9f8: httpd_sess_process at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:416 (discriminator 9)

0x42059d7c: httpd_process_session at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_main.c:178 (discriminator 9)

0x420bbda7: httpd_sess_enum at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)

0x4205a0b3: httpd_server at /home/samuele/data/esp/esp-idf/components/esp_http_server/src/httpd_main.c:228
[... 413 lines cut ...]
D (1298) esp_core_dump_elf: Add segment size=344, start_off=0x842c
D (1305) esp_core_dump_elf: Add stack for task 0x3fcf310c: addr 0x3fcf2e70, sz 660
D (1312) esp_core_dump_elf: Add segment size=660, start_off=0x8584
D (1320) esp_core_dump_elf: Found 0 bad task out of 19
D (1323) esp_core_dump_elf: ================ Processing coredump info ================
D (1331) esp_core_dump_elf: Application SHA256='23dce878fc18915c', length=17.
D (1339) esp_core_dump_elf: Add note size=104, start_off=0x8818
D (1345) esp_core_dump_elf: Add note size=172, start_off=0x8818
D (1350) esp_core_dump_elf: Add NOTE segment, size=276, start_off=0x8818
D (1357) esp_core_dump_elf: =========== Data written size = 35136 bytes ==========
D (1365) esp_core_dump_checksum: Total length of hashed data: 35136
I (1371) esp_core_dump_flash: Write end offset 0x8944, check sum length 4
I (1378) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.
```

### More Information.

In function `httpd_req_cleanup(httpd_req_t *r)`, in file `httpd_parse.c`, the function `httpd_sess_free_ctx(void **ctx, httpd_free_ctx_fn_t free_fn)` is invoked as:

`httpd_sess_free_ctx(ra->sd->ctx, ra->sd->free_ctx);`

instead it should be:

`httpd_sess_free_ctx(&ra->sd->ctx, ra->sd->free_ctx);`

hmalpani, 2022-11-29:
Hello @sforconi,
I tried to reproduce the issue using the steps you provided but was unable to reproduce. My application didn't crashed. Can you please share your code here to which can reproduce this issue?

sforconi, 2022-11-29:
Hi,
I have a `/login` URI where I malloc() a session context, then I have a `/logout` URI where the session context should be freed.
Reading the docs (if I understand correctly), it should be sufficient to set `req->sess_ctx = NULL`, then the esp_http_server should manage the deallocation of the previous allocated session context. 
Doing this, the code crashes producing the output above.
Using `eq->ignore_sess_ctx_changes = true` and manually freeing the session context works correctly.

In my opinion, the `httpd_sess_free_ctx(ra->sd->ctx, ra->sd->free_ctx)` in `httpd_req_cleanup(httpd_req_t *r)` is not correct as `ra->sd->ctx` is a `void *`, instead the function `httpd_sess_free_ctx(void **ctx, httpd_free_ctx_fn_t free_fn)` works on a `void **`.

hmalpani, 2022-11-29:
@sforconi Got it. Was able to reproduce it. Will fix it. Thanks!

closed 2022-12-22 by espressif-bot