espressif/esp-idf #10266: Create several Virtual Com ports

beckmx, 2022-11-28:
### Is your feature request related to a problem?

Sometimes having to attach several devices to USB takes a lot of space for settting up new serial ports, being able to create virtual com ports inside the hardware could be a great idea

### Describe the solution you'd like.

When connecting an esp32 usb we could be able to setup new ports or a number of serial ports that we could handle inside the code

### Describe alternatives you've considered.

creating several pcbs to individual setup serial ports in the computer host

igrr, 2023-01-16:
@beckmx to clarify your request, is your request related to ESP32-S2 and ESP32-S3?

Other ESP32 series chips don't have a programmable USB peripheral which could be used to implement multiple virtual serial ports.

If yes, then this example should allow you to create two CDC ports: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_serial_device if you set CONFIG_TINYUSB_CDC_COUNT=2.

Increasing the number of CDC ports might not be possible beyond that, though, since the chip only supports a limited number of USB endpoints (https://github.com/espressif/esp-idf/pull/8011#issuecomment-1006291715).

beckmx, 2023-01-17:
Nvm I just noticed you answered everything

closed 2023-01-17 by beckmx