espressif/esp-idf #10267: BLE Stops Advertising after a while in

kovid-dozee, 2022-11-29:
Hello All,
I am using ESP32 for GATTC, GATTS and WiFi. While I am using Bluetooth functionality I don't work on WiFi. I am facing the same issue in my device as well, whenever the device is in the gattc mode and takes the data from server on Notification. After taking few values the bluetooth advertising stops and Bluetooth stops working. I need to restart the device to make it work again. I am using IDF v4.3-dirty as the toolchain.

_Originally posted by @kovid-dozee in https://github.com/espressif/esp-idf/issues/513#issuecomment-1330056380_

xiewenxiang, 2022-11-29:
Can you describe the appearance of not working in detail?

kovid-dozee, 2022-11-30:
My accessory is sending the data on Notification.
I have a task which on an interval of 60secs call the esp_ble_gap_start_scanning() and then connects with the accessory -> take data on Notification -> inside the notification if we recieve 50 data sets we call esp_ble_gap_disconnect() -> which disconnects the accessory and starts Advertising(since we call esp_ble_gap_start_advertising() in Disconnect event).
Now the problem is after 7 or 8 times of above cycle the BLE though disconnects the accessory but couldn't start advertise. Also we are getting ESP_OK as return from esp_ble_gap_start_advertising() but still the advertising is not starting. The advertising only starts after power reset.

kovid-dozee, 2022-12-07:
Hello any updates on this?

xiewenxiang, 2022-12-09:
Can you provide the reproduce code?
From the current description, it is impossible to determine the exact cause

closed 2024-12-17 by Alvin1Zhang