espressif/esp-idf #10268: [Crash][v4.4.3][ESP32-S3] assert failed: spi_flash_disable_interrupts_caches_and_other_cpu when using KConfig 16KB Data Cache

chipweinberger, 2022-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-215-g9f3f82f54b

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

I should save 16KB of Internal RAM by setting Data Cache Size to 16KB.

[image]

### What is the actual behavior?

Shortly after boot, I hit this assert:

`assert failed: spi_flash_disable_interrupts_caches_and_other_cpu cache_utils.c:137 (esp_task_stack_is_sane_cache_disabled())`

### Steps to reproduce.

1. Set CONFIG_ESP32S3_DATA_CACHE_16KB=y 
2. try to initialize wifi
3. (possibly other unknown steps)
4. crash

### Debug Logs.

```plain
I (9407) pd wifi: esp_event_loop_create_default()
I (9417) pd wifi: esp_netif_create_default_wifi_ap()
I (9417) pd wifi: esp_wifi_init()
I (9427) pp: pp rom version: e7ae62f
I (9427) net80211: net80211 rom version: e7ae62f

assert failed: spi_flash_disable_interrupts_caches_and_other_cpu cache_utils.c:137 (esp_task_stack_is_sane_cache_disabled())

Backtrace: 0x40376507:0x3c002680 |<-CORRUPTED
0x40376507: panic_abort at /esp-idf/components/esp_system/panic.c:429
```

### More Information.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10109856/sdkconfig.txt)

chipweinberger, 2022-11-29:
Related: https://github.com/espressif/esp-idf/issues/9393

chipweinberger, 2022-11-29:
In` memory_layout.c`, this code looks fishy.

```
#if CONFIG_ESP32S3_DATA_CACHE_16KB || CONFIG_ESP32S3_DATA_CACHE_32KB
    { 0x3FCF0000,           0x8000,                                     SOC_MEMORY_TYPE_DRAM,       0}, //Level 9, DRAM, DMA is accessible but retention DMA is inaccessible
#endif
#if CONFIG_ESP32S3_DATA_CACHE_16KB
    { 0x3C000000,           0x4000,                                     SOC_MEMORY_TYPE_DRAM,       0}, //Level 10, DRAM, DMA is accessible but retention DMA is inaccessible
#endif
```

I would have expected the fist line to be `#if CONFIG_ESP32S3_DATA_CACHE_32KB`, not both 32KB and 16KB

chipweinberger, 2022-11-29:
Also this code looks fishy. Is the else for 16KB or 64KB? There are 3 options, not 2.

```
#if CONFIG_ESP32S3_DATA_CACHE_32KB
    Cache_Occupy_DCache_MEMORY(CACHE_MEMORY_DBANK1, CACHE_MEMORY_INVALID);
    cache_size = CACHE_SIZE_HALF;
#else
    Cache_Occupy_DCache_MEMORY(CACHE_MEMORY_DBANK0, CACHE_MEMORY_DBANK1);
    cache_size = CACHE_SIZE_FULL;
#endif
```

chipweinberger, 2022-11-29:
Or perhaps we need to check for sanity like this?

```
static inline bool esp_task_stack_is_sane_cache_disabled(void)
{
    const void *sp = (const void *)esp_cpu_get_sp();

    return esp_ptr_in_dram(sp)
#if CONFIG_ESP_SYSTEM_ALLOW_RTC_FAST_MEM_AS_HEAP
        || esp_ptr_in_rtc_dram_fast(sp)
#endif
#if CONFIG_ESP32S3_DATA_CACHE_16KB
        || ((void*) SOC_DROM_LOW <= sp && sp <= (void*)(SOC_DROM_LOW + 0x4000))
#endif
    ;
}
```

To be honest, I dont understand why `0x3C000000`, which is the ESP32-S3 `SOC_DROM_LOW` address is being mapped to `SOC_MEMORY_TYPE_DRAM`. This makes no sense to me. `DROM` is in flash memory, meant for string constants, etc, not the heap.

But in `cpu_start.c` I see this code, which seems to confirm the that 'Data Cache' uses DROM? Very confusing.

```
#if CONFIG_ESP32S3_DATA_CACHE_16KB
    Cache_Invalidate_DCache_All();
    Cache_Occupy_Addr(SOC_DROM_LOW, 0x4000);
#endif
```

chipweinberger, 2022-11-30:
The more I learn using the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf), the more I realize 16KB Data Cache is just not supported.

`Cache_Set_DCache_Mode` only supports "FULL(64KB)" and "HALF(32KB)", but AFAICT not "QUARTER(16KB)". This function is implemented in ROM, so I don't know for sure.

chipweinberger, 2023-05-25:
this was fixed here: https://github.com/espressif/esp-idf/commit/3118120659869ea79ecdb33c45f9d4e20fded563

chipweinberger, 2023-05-25:
@mahavirj can you look at this comment: https://github.com/espressif/esp-idf/issues/10268#issuecomment-1330125529

Sherry616, 2024-03-05:
Hi @chipweinberger, let me introduce the confusing map.

In esp32s3, our cache supports cache occupy feature, that is force occupy some cache lines in the cache data memory. The occupied cache lines cannot be replaced, in such ways the cache lines can be treat as internal memory. But, we should must use cache virtual address to access these "internal memory", or we can call it "FAKE dram"
 
The issue is obvious clear for the assert log: esp_task_stack_is_sane_cache_disabled()
I think this is because somehow your task's stack is located in the "FAKE dram", but IDF does not support good of the address region. Thanks.

closed 2026-03-14 by chipweinberger