espressif/esp-idf #10270: [ESP32S3] Value of 'CONFIG_ESP32S3_USE_FIXED_STATIC_RAM_SIZE' can only been set to '0x34000' max in 'menuconfig'

sbithinx, 2022-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
'CONFIG_ESP32S3_USE_FIXED_STATIC_RAM_SIZE' can maximal been set to `0x34000` via `menuconfig`. I would expect to be able to set it to maximal available RAM of 512KB.

[image]

Is this intended? If yes, why? If not, this is properly a bug.

Greetings
Sebastian

sbithinx, 2022-11-29:
Also, if the maximal value of `0x34000` is configured, the compilation always fails.

ma89vik, 2022-11-30:
Yeah, I think you are right. Seems like we have done a poor job of updating this option for new chips.

Thanks for reporting it, we'll take a look.

chipweinberger, 2022-12-01:
@sbithinx, can you explain what this setting is supposed to do?

See my related issue: https://github.com/espressif/esp-idf/issues/10281

ESP-Marius, 2022-12-01:
@chipweinberger I think the idea is that it should give you a fixed and reliable heap size, e.g. if you add one more bss variable it wont change your total heap memory, as it would already have reserved a static amount for this.

ESP-Marius, 2022-12-01:
Unfortunately the option seems to have been broken during refactoring, and no longer does what it is supposed to do as far as I can see. Ill look at it today...

ESP-Marius, 2022-12-01:
@sbithinx With regards to your original question: this value will be less than the chips total memory, as not all of it can be used for this purpose. E.g. parts are used by the software bootloader, and will only be available as heap memory after app startup.

sbithinx, 2022-12-01:
@ESP-Marius Thanks for the support. I'm aware of this. I was just to lazy to calculate the exact values. This is better done by you :-) But it actually can't exceed the maximal value of available RAM, even though it is less than that. :-D

closed 2023-03-22 by espressif-bot