espressif/esp-idf #10273: BUG - Ota update to ESP-IDF v4.4.3 makes MCPWM stop producing a signal

Sjurinho, 2022-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-Wroom32

### Power Supply used.

External 5V

### What is the expected behavior?

MCPWM provides a proper motor signal and runs a pair of motors

### What is the actual behavior?

The motors do not run at all before a physical power cycle.

### Steps to reproduce.

1. Use IDF v4.2.3
2. Run motors - works
3. Upgrade to ESP-IDF v4.4.3 via Ota update
4. Motors don't work
5. Do a physical reset (plug power in and out)
6. Motors work again, and never stops working

### More Information.

I recently began migrating from ESP-IDF version v4.2.3 to ESP-IDF version v4.4.3. When I flashed to the board via USB everything worked as a charm and it worked fine.
However, when I updated my application using the app_update module (since I have some externally installed units) the motors on my units stopped working. The actual API still responds the way I expect, with ESP_OK returning from mcpwm_init, mcpwm_gpio_init, mcpwm_set_duty and mcpwm_set_duty_mode. The actual signal produced, however, does not run the motors. The code I have tested is the exact same code, but the IDF version is updated without a hardware reset in between, only an `esp_restart` call. 

The weird part is that after I perform a physical power cycle of the mcu, then it actually starts working again!

Has anyone else had some similar experience? Could there be something in the physical registers that need flushing? If so, is there any way to do that via software?

Sjurinho, 2022-11-29:
Here is my initialisation code
```
// Initialise MCPWM unit, timer and gpio pins (Clockwise then Counter-Clockwise)
    err_resp = mcpwm_gpio_init(_mc_unit, _cw_signal, _cw_pin);
    if (err_resp != ESP_OK) {
        ESP_LOGE(TAG, "<< mcpwm_gpio_init(_mc_unit, _cw_signal, _cw_pin): %d %s", err_resp, esp_err_to_name(err_resp));
    }
    err_resp = mcpwm_gpio_init(_mc_unit, _ccw_signal, _ccw_pin);
    if (err_resp != ESP_OK) {
        ESP_LOGE(TAG, "<< mcpwm_gpio_init(_mc_unit, _ccw_signal, _ccw_pin): %d %s", err_resp, esp_err_to_name(err_resp));
    }

    // Initialise MCPWM configuration
    mcpwm_config_t mc_config;
    mc_config.frequency = 18000;                // Frequency in Hz (16-20kHz to remove motor squeal)
    mc_config.cmpr_a = 0;                       // Duty cycle of PWMxA = 0
    mc_config.cmpr_b = 0;                       // Duty cycle of PWMxb = 0
    mc_config.duty_mode = MCPWM_DUTY_MODE_0;    // Active high duty, percentage of time high equals duty cycle percentage
    mc_config.counter_mode = MCPWM_UP_COUNTER;

    // Configure PWMxA/B with above settings
    err_resp = mcpwm_init(_mc_unit, _mc_timer, &mc_config);
    if (err_resp != ESP_OK) {
        ESP_LOGE(TAG, "<< mcpwm_init(_mc_unit, _mc_timer, &mc_config): %d %s", err_resp, esp_err_to_name(err_resp));
    }
```

Sjurinho, 2022-11-30:
I tested reverting my IDF version to IDF v4.3.4, and that solves the problem without any external interaction.

suda-morris, 2022-11-30:
@Sjurinho Thank you very much for reporting the issue. 

Do you mean the MCPWM stops generating the PWM waves after you call `esp_restart`? Sorry, I asked because we want to create a minimal project that can reproduce the issue stably.

Sjurinho, 2022-11-30:
Hi @suda-morris, thanks for the quick response. I haven't been able to check on a simple dev-board, but I have a board connected to a set of motors running on two gpio-pins. 
What I can say is that after performing an Ota update from ESP-IDF v4.2.3 to ESP-IDF v4.4.3, the motors stopped running. Performing a simple esp-restart did not solve the problem. However, when I actually pulled the external power source low, thus performing a hardware restart, the motors started running again. After the hardware restart, the functionality worked as expected, also after performing esp-restarts and doing new updates to my application image (still using ESP-IDF v4.4.3).
In short, it seems like it only occurs on the `esp_restart`s after booting the application image with the updated ESP-IDF. 

If I have understood it correctly `esp_restart` does not restart the peripherals, which could explain why the issue only occurs before a hardware restart.

suda-morris, 2022-12-05:
Luckily we have a "complicated" example to spin a DC motor. After I tweaked the example by a little (don't want to use the console CLI), let's spin the motor for 5 seconds and then restart the system. I can see the motor can resume after restart. It never stops working. :thinking: 
```diff
diff --git a/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/cmd_mcpwm_motor.c b/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/cmd_mcpwm_motor.c
index 93f683e939..277453cdd4 100644
--- a/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/cmd_mcpwm_motor.c
+++ b/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/cmd_mcpwm_motor.c
@@ -287,25 +287,28 @@ void cmd_mcpwm_motor_init(mcpwm_motor_control_t *motor_ctrl)
     ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&usbjtag_config, &repl_config, &repl));
 #endif

-    register_motor_ctrl_config();
-    register_motor_ctrl_expt();
-    register_motor_ctrl_pid();
-    register_motor_ctrl_motor();
-
-    printf("\n =================================================================\n");
-    printf(" |             Example of Motor Control                          |\n");
-    printf(" |                                                               |\n");
-    printf(" |  1. Try 'help', check all supported commands                  |\n");
-    printf(" |  2. Try 'config' to set control period or pwm frequency       |\n");
-    printf(" |  3. Try 'pid' to configure pid paremeters                     |\n");
-    printf(" |  4. Try 'expt' to set expectation value and mode              |\n");
-    printf(" |  5. Try 'motor' to start motor in several seconds or stop it  |\n");
-    printf(" |                                                               |\n");
-    printf(" =================================================================\n\n");
-
-    printf("Default configuration are shown as follows.\nYou can input 'config -s' to check current status.");
-    print_current_status();
-
-    // start console REPL
-    ESP_ERROR_CHECK(esp_console_start_repl(repl));
+    // register_motor_ctrl_config();
+    // register_motor_ctrl_expt();
+    // register_motor_ctrl_pid();
+    // register_motor_ctrl_motor();
+
+    // printf("\n =================================================================\n");
+    // printf(" |             Example of Motor Control                          |\n");
+    // printf(" |                                                               |\n");
+    // printf(" |  1. Try 'help', check all supported commands                  |\n");
+    // printf(" |  2. Try 'config' to set control period or pwm frequency       |\n");
+    // printf(" |  3. Try 'pid' to configure pid paremeters                     |\n");
+    // printf(" |  4. Try 'expt' to set expectation value and mode              |\n");
+    // printf(" |  5. Try 'motor' to start motor in several seconds or stop it  |\n");
+    // printf(" |                                                               |\n");
+    // printf(" =================================================================\n\n");
+
+    // printf("Default configuration are shown as follows.\nYou can input 'config -s' to check current status.");
+    // print_current_status();
+
+    // // start console REPL
+    // ESP_ERROR_CHECK(esp_console_start_repl(repl));
+
+    mc->cfg.running_sec = 5;
+    brushed_motor_start(mc);
 }
diff --git a/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/mcpwm_brushed_dc_control_example.c b/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/mcpwm_brushed_dc_control_example.c
index ba382bc71b..4d00496512 100644
--- a/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/mcpwm_brushed_dc_control_example.c
+++ b/examples/peripherals/mcpwm/mcpwm_brushed_dc_control/main/mcpwm_brushed_dc_control_example.c
@@ -26,6 +26,8 @@

 #include "mcpwm_brushed_dc_control_example.h"

+#include "esp_system.h"
+
 #define MOTOR_CTRL_MCPWM_UNIT   MCPWM_UNIT_0
 #define MOTOR_CTRL_MCPWM_TIMER  MCPWM_TIMER_0

@@ -283,6 +285,7 @@ static void mcpwm_brushed_motor_expt_thread(void *arg)
             if (motor_ctrl.sec_cnt > 20 * motor_ctrl.cfg.running_sec && motor_ctrl.cfg.running_sec != 0) {
                 brushed_motor_stop(&motor_ctrl);
                 printf("\nTime up: motor stoped\n");
+                esp_restart();
             }
         }
```

Sjurinho, 2022-12-05:
@suda-morris The problem only occures when running an older IDF-version (older than v4.4) and then updating the app to an image built with IDF version newer than v4.4 via the `app_update` functions. The first boot after this transition will not work until doing a hardware reset (that is plugging out the power). After having performed the hardware reset, it works fine again, but since some deployments might be remote it might be cumbersome to go out and do an actual hardware reset.

In my case it happened after I updated my unit from ESP-IDF v4.2.3-dirty to ESP-IDF v4.4.3, using `esp_ota_write` to update the image. After I did this, the motor did not run until after I performed a hardware reset. 
I have also tested the transition from v4.2.3-dirty --> v4.3.4, and in this case it works completely fine.

I understand that the situation is quite edge case, but as far as I can see there is no warning in the docs about this.

suda-morris, 2022-12-06:
Right, this issue is not easy to reproduce on my side. I choose this [example](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/peripherals/mcpwm/mcpwm_brushed_dc_control). Compile two bin files based on release/v4.2 and release/v4.4 SDK. And flash 4.2 bin to OTA slot 0, 4.4 bin to OTA slot1 (using the otatool.py).

Firstly I switch to slot 0 to run, checking the PWM output is correct. Then I switch to slot 1 (also using the otatool.py), without a hardware reset in the middle. I checked the PWM output, it's also correct.

Sjurinho, 2022-12-06:
Hmm, that example has alot of similarity to my project, so it should display the same behavior. May I ask how you do the actual switch between the slots? I see that the `otatool.py` only writes the necessary data to the `otadata` partition. In my case at least, performing a hardware reset (which on the development board would be the same as pressing the reset button) would clear the problem. The problem only arises when doing an ota update, meaning that the mcu does the switch by calling `esp_ota_set_boot_partition` and `esp_restart`. Could you test switching between the two images using these calls?

Sjurinho, 2022-12-06:
> Hmm, that example has alot of similarity to my project, so it should display the same behavior. May I ask how you do the actual switch between the slots? I see that the `otatool.py` only writes the [...]

Something like this
```cpp
static void mcpwm_example_brushed_motor_control(void *arg)
{
    //1. mcpwm gpio initialization
    mcpwm_example_gpio_initialize();

    //2. initial mcpwm configuration
    printf("Configuring Initial Parameters of mcpwm...\n");
    mcpwm_config_t pwm_config;
    pwm_config.frequency = 1000;    //frequency = 500Hz,
    pwm_config.cmpr_a = 0;    //duty cycle of PWMxA = 0
    pwm_config.cmpr_b = 0;    //duty cycle of PWMxb = 0
    pwm_config.counter_mode = MCPWM_UP_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);    //Configure PWM0A & PWM0B with above settings
    // 3. Test Motors
    do {
        brushed_motor_forward(MCPWM_UNIT_0, MCPWM_TIMER_0, 50.0);
        vTaskDelay(2000 / portTICK_RATE_MS);
        brushed_motor_backward(MCPWM_UNIT_0, MCPWM_TIMER_0, 30.0);
        vTaskDelay(2000 / portTICK_RATE_MS);
        brushed_motor_stop(MCPWM_UNIT_0, MCPWM_TIMER_0);
        vTaskDelay(2000 / portTICK_RATE_MS);
    } while (false);
    // 4. Switch Partition
    esp_ota_set_boot_partition(esp_ota_get_next_update_partition(NULL)); //depending on partition configuration
    esp_restart();
}
```

dami9814n, 2024-03-13:
hi, i am experiencing the same problem with pwm signals updating from esp-idf v4.4.6 to esp-idf v5.0.4. Resetting the device by shutting the power down fixes the problem.
Any news on this bug?

suda-morris, 2025-01-20:
@dami9814n Starting from esp-idf v5.0, the MCPWM driver has been redesigned. Do you still face the issue with the new driver? If so, could you describe the steps for use to reproduce the issue?

closed 2025-03-12 by Alvin1Zhang