espressif/esp-idf #10274: Does the ESP32 Support BLE Scanning or Connecting While Sending/Receiving Data Over BLE?

Aaron-Beal-CardinalPeak, 2022-11-29:
Hello,

I've seen vague mentions of this when searching through forums/issues, but I wanted an explicit response. Does the ESP32 (through the ESP-IDF framework) support running these BLE operations simultaneously when it's acting as a central:
* Scan while receiving data from an existing connection.
* Scan while sending data over an existing connection.
* Connecting to another device while receiving data over an existing connection.
* Connecting to another device while sending data over an existing connection.

Based on various pieces of example code, I currently halt scans when attempting a connect. I've been trying to get the above scan-while-connected scenarios to work with limited luck so far (the new connection usually causes the existing connection to disconnect). I want to make sure I'm not attempting something that's completely unsupported.

If any of the above are supported, are there any limitations/restrictions I should be aware of to minimize the chance of connections failing?

xiewenxiang, 2022-11-30:
Because ESP32 is single antenna, it is impossible to be in the sending or receiving state at the same time.

The four scenarios you mentioned above are all supported, and our coexistence strategy is time-sharing multiplexing.

Aaron-Beal-CardinalPeak, 2022-12-05:
@xiewenxiang Any response to my last question?

xiewenxiang, 2022-12-06:
@Aaron-Beal-CardinalPeak  Yes, it was advised to wait until the event returns before performing the next step.

closed 2022-12-06 by Aaron-Beal-CardinalPeak