espressif/esp-idf #10275: Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

SunnyRuss, 2022-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Wi-Fi to initialize.

### What is the actual behavior?

crashes with cache access fault

### Steps to reproduce.

Turn the ESP32 on.  It happens every time, so far.

### Debug Logs.

```plain
I (14995) wifi_init: WiFi/LWIP prefer SPIRAM
I (14995) wifi_init: WiFi IRAM OP enabled
I (15005) wifi_init: WiFi RX IRAM OP enabled
I (15025) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4018d4b4  PS      : 0x00060034  A0      : 0x40084530  A1      : 0x3ffb0de0  
A2      : 0x3ffd7828  A3      : 0x3ffb2044  A4      : 0x00000000  A5      : 0x4008e626  
A6      : 0x3ffb2fb8  A7      : 0x00802000  A8      : 0x80085278  A9      : 0x00000000  
A10     : 0x3ffb2044  A11     : 0x3ffb0de0  A12     : 0x3ff42000  A13     : 0x6c000002  
A14     : 0xe0000020  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4018d4b1:0x3ffb0de0 0x4008452d:0x3ffb0e20 0x40092ae4:0x3ffdf7f0 0x4009bc65:0x3ffdf820 0x4008794b:0x3ffdf8a0 0x4011c2ad:0x3ffdf8e0 0x4011db7b:0x3ffdf910 0x4018d36e:0x3ffdf930 0x4011eaf5:0x3ffdf950 0x4011d361:0x3ffdf9e0 0x4011d3d7:0x3ffdfa60 0x4011da09:0x3ffdfac0 0x4011c4c5:0x3ffdfae0 0x4011c889:0x3ffdfb10 0x4015d10b:0x3ffdfb30 0x4015d384:0x3ffdfba0 0x4015d486:0x3ffdfbd0 0x4015d57d:0x3ffdfc00 0x40144676:0x3ffdfc20 0x40144de5:0x3ffdfc40 0x40143818:0x3ffdfc60 0x400983f5:0x3ffdfc80 0x40091da1:0x3ffdfcb0

C:\Sunsight\projects\AntennAlign>C:\Sunsight\esp\tools\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-addr2line.exe -p -C -e build\AntennAlign.elf
0x4018d4b1
??:?
0x4008452d
C:/Sunsight/esp/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1111
0x40092ae4
C:/Sunsight/esp/esp-idf-v4.4/components/hal/esp32/include/hal/spi_flash_ll.h:345
0x4009bc65
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:218
0x4008794b
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/esp_flash_api.c:832 (discriminator 4)
0x4011c2ad
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/partition.c:424
0x4011c2ad
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/partition.c:424
0x4011db7b
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_partition.cpp:45
0x4018d36e
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_page.cpp:879
0x4011eaf5
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_page.cpp:306
0x4011d361
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_storage.cpp:481
0x4011d3d7
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_storage.cpp:548
0x4011da09
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_handle_simple.cpp:36
0x4011c4c5
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_api.cpp:509
0x4011c889
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_api.cpp:519
0x4015d10b
C:/Sunsight/esp/esp-idf-v4.4/components/esp_phy/src/phy_init.c:599
0x4015d384
C:/Sunsight/esp/esp-idf-v4.4/components/esp_phy/src/phy_init.c:517
```

### More Information.

I used IDF.PY to disable the second core. (#define CONFIG_FREERTS_UNICORE 1 in the SDKCONFIG.H file)   It runs fine.

SunnyRuss, 2022-11-29:
A slightly different version.

I (15234) wifi_init: tcp mbox: 6
I (15234) wifi_init: tcp tx win: 5744
I (15244) wifi_init: tcp rx win: 5744
I (15244) wifi_init: tcp mss: 1440
I (15254) wifi_init: WiFi/LWIP prefer SPIRAM
I (15254) wifi_init: WiFi IRAM OP enabled
I (15264) wifi_init: WiFi RX IRAM OP enabled
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4018cda4  PS      : 0x00060034  A0      : 0x40084070  A1      : 0x3ffb0ba0  
A2      : 0x3ffd5614  A3      : 0x3ffb15b8  A4      : 0x00000000  A5      : 0x4008dd6e  
A6      : 0xdaa575cb  A7      : 0x00000001  A8      : 0x80084c9c  A9      : 0x00000000  
A10     : 0x3ffb15b8  A11     : 0x3ffb0ba0  A12     : 0x80091d21  A13     : 0x3ffdd580  
A14     : 0x3ffdd5d0  A15     : 0x3ffdd590  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x40092bdb  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4018cda4

Backtrace: 0x4018cda1:0x3ffb0ba0 0x4008406d:0x3ffb0be0 0x40008544:0x3ffdd5f0 0x400874b9:0x3ffdd610 0x4009b4b1:0x3ffdd630 0x4009b3b5:0x3ffdd660 0x4009af81:0x3ffdd680 0x4008720d:0x3ffdd700 0x4011b8fa:0x3ffdd750 0x4011d393:0x3ffdd770 0x4011d9a1:0x3ffdd790 0x4011dc14:0x3ffdd7b0 0x4011c7c1:0x3ffdd820 0x4011ccfd:0x3ffdd8a0 0x4011d1b5:0x3ffdd910 0x4011c03b:0x3ffdd930 0x40145aad:0x3ffdd960 0x40143bd6:0x3ffdd980 0x40144b41:0x3ffdd9e0 0x4014312c:0x3ffdda50 0x40097651:0x3ffdda70 0x4009115d:0x3ffddaa0

0x4018cda1:0x3ffb0ba0
??:?
0x4008406d:0x3ffb0be0
C:/Sunsight/esp/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1111
0x40008544:0x3ffdd5f0
??:0
0x400874b9:0x3ffdd610
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/spi_flash_os_func_app.c:148
0x4009b4b1:0x3ffdd630
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:443
0x4009b3b5:0x3ffdd660
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:241
0x4009af81:0x3ffdd680
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:266
0x4008720d:0x3ffdd700
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/esp_flash_api.c:907
0x4011b8fa:0x3ffdd750
C:/Sunsight/esp/esp-idf-v4.4/components/spi_flash/partition.c:466
0x4011d393:0x3ffdd770
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_partition.cpp:59
0x4011d9a1:0x3ffdd790
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_page.cpp:101
0x4011dc14:0x3ffdd7b0
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_page.cpp:247
0x4011c7c1:0x3ffdd820
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_storage.cpp:221 (discriminator 4)
0x4011ccfd:0x3ffdd8a0
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_storage.cpp:320
0x4011d1b5:0x3ffdd910
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_handle_simple.cpp:52
0x4011c03b:0x3ffdd930
C:/Sunsight/esp/esp-idf-v4.4/components/nvs_flash/src/nvs_api.cpp:425
0x40145aad:0x3ffdd960
??:?
0x40143bd6:0x3ffdd980
??:?
0x40144b41:0x3ffdd9e0
??:?
0x4014312c:0x3ffdda50
??:?
0x40097651:0x3ffdda70
??:?
0x4009115d:0x3ffddaa0
C:/Sunsight/esp/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131

AxelLin, 2023-02-09:
@SunnyRuss Do you still hit the issue with the latest v4.4.4 release?

SunnyRuss, 2023-02-09:
I am still running 4.4.3.

For some strange reason, I have not encountered this in quite a while.  Obviously, this problem is influenced by everything else that is happening within the ESP32.  As such, when I hit it, I hit for a week or so, then I change the something (the project is not yet complete) and it goes away.

So, for right now, I am happy.  But thanks for staying on top of this.

-Russ

ESP-Marius, 2023-03-01:
Closing this for now then, if you reproduce this again on the latest release of v4.4 then please re-open it and we'll try to investigate and see if we can find the root-cause.

closed 2023-03-01 by ESP-Marius