espressif/esp-idf #10276: Cannot read Reserved RTC Memory from Factory App

Delzye, 2022-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-DevKit-C v4 (Wrover-E) | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I want to use the reservable section in RTC fast memory to communicate data between bootloader and factory application between resets (custom bootloader override).

According to [this PR](https://github.com/espressif/esp-idf/pull/8480), I can use:
```
  rtc_retain_mem_t* mem = bootloader_common_get_rtc_retain_mem();
  uint8_t x = mem->custom[0];
```
to read and write data to the RTC section. I expected to use this function in both the bootloader and the factory app (it's a bootloader function, but I could not find any other fuction to achieve the same functionality in the factory app) to read and write data to this section in memory.

To use the function in the factory app, I created a task and pinned it to the PRO CPU Core.

### What is the actual behavior?

I can read and write data to this section in the bootloader and read the written data after a reset.

I can also write data to the section in the factory app and read it in the bootloader after a reset, but I can't read any data in the factory app.

Whenever I access the array in the factory app, it just returns 0, even though I know there is data written there (I can read it in the bootloader before and after accessing it in the factory app).

I also made sure the pointer always points to the same location in memory (which makes sense, since I can write data to the array).

### Steps to reproduce.

1. Activate Reserved RTC Fast Memory (16 Bytes in my case)
2. Step: Create a custom bootloader and add:
```
    rtc_retain_mem_t* mem = bootloader_common_get_rtc_retain_mem();
    esp_rom_printf( "[%s], First byte of rtc memory: %x Validation: %x\n", TAG, mem->custom[0], mem->custom[1] );
```
after `bootloader_init()`.
3. Step: Add the same code in the factory app in a task pinned to CPU0 (PRO Core) (different output, why?)
4. Step: Add this to a factory app (Again CPU0):
```
    rtc_retain_mem_t* mem = bootloader_common_get_rtc_retain_mem();
    mem->custom[0] = 0xE1;
```
Repead 1./2., 0xE1 is read in 1., but not in 2.

boarchuz, 2022-11-30:
The CRC will be invalid after changing that byte, which will cause the bootloader to reset the 'corrupt' contents.
Update the CRC with `bootloader_common_update_rtc_retain_mem(NULL, false);`. The bootloader does this automatically*.

See:
https://github.com/espressif/esp-idf/blob/99e2722aabce89e6477c1e1f105b4db45db6d330/components/bootloader_support/src/bootloader_common_loader.c#L185-L204

*I do see what appears to be an unrelated bug(?): the bootloader will not update rtc_retain_mem_t if the Test app is booted via either of these paths:
https://github.com/espressif/esp-idf/blob/99e2722aabce89e6477c1e1f105b4db45db6d330/components/bootloader_support/src/bootloader_utility.c#L491
https://github.com/espressif/esp-idf/blob/99e2722aabce89e6477c1e1f105b4db45db6d330/components/bootloader_support/src/bootloader_utility.c#L521

Delzye, 2022-11-30:
Ah, thank you, that makes sense!
Is this documented anywhere?
I tried to read through the descriptions of all functions in that file before using them, but this wasn't mentioned?

Kind regards

boarchuz, 2022-11-30:
Not sure but definitely not if you were looking at more recent docs. In newer IDF versions the CRC no longer includes the custom user region:
https://github.com/espressif/esp-idf/blob/593335578f3893f8f6921f8a21079cedb9afa15a/components/bootloader_support/src/bootloader_common_loader.c#L140-L149

Delzye, 2022-11-30:
Ah, in v4.4 that function does not exist yet.
Thanks for the quick help!

closed 2022-11-30 by Delzye

o-marshmallow, 2022-12-06:
@Delzye The commit that fixes this bug has been backported to `release/v5.0` and `release/v4.4` branches. It will be synced to github as soon as internal tests pass.