espressif/esp-idf #10277: Sleep state on esp32-wrover-ie

ASabovic, 2022-11-30:
### Is your feature request related to a problem?

Hi guys,

I am wondering is there a possibility of putting my esp32-wrover-ie board into a sleep state without using an external wake-up once I want it to perform something (for example take a photo using a camera module)? I am looking for some solution similar to what Arduino has with their delay function and turning off all peripherals that are not necessary?

ASabovic, 2022-12-07:
In my case, I just want to test the person detection model where the esp32 board will take a photo go to low power state (I hope below 1mA) for 5 seconds let’s say and then perform the inference task and again sleep 5 seconds to repeat the cycle. The best will be to turn off WiFi and BLE as the communication part in this project is not important. Is that possible? And also, does this work with esp32-cam as well? Thanks!

igrr, 2022-12-07:
If you don't need to keep Wi-Fi or BLE working, you can use the Light Sleep feature. Light Sleep current on the ESP32 is around 1mA. After wake-up the program continues running from the same place.
You may also use Deep Sleep, then the current will be lower. Exact value will depend on the development board; the ESP32 chip itself will consume around 5uA. After waking up from deep sleep, the program will start from the beginning, so deep sleep is more like a reset from software perspective.
Both methods are described in the first link from my previous comment.

ESP-Marius, 2023-04-27:
Closing this as the original question has been answered.

closed 2023-04-27 by ESP-Marius