espressif/esp-idf #10278: Bluedroid  gattserver callback issue

nouwon, 2022-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I capture ESP_GATTS_DISCONNECT_EVT after client connection is closed?I am using esp-idf ver 4.4.3.
Currently an android client performs connection,sends data,receives data successfully  but when the android client disconnects
gatt server call back event ESP_GATTS_DISCONNECT_EVT is never triggered.
Similar problems have been reported by other users but there is no satisfying solution proposal for this issue?
Any help will be highly apprecited.
Thanks

nouwon, 2022-12-01:
Thank you very much for your response xulogzhuang.
Following are relevant initialization and registration functions.
I can provide further code if nothing wrong is present here.

Project brief:
1-First I walk through an initialization sequence in my main.c file with app_state_handler() function call.

	static void app_state_handler(){

		/* Check the application's current state. */
		switch (appData.state)
		{
				/* Application's initial state. */
			case APP_STATE_INIT:
			{
				app_init();
				ESP_LOGI(SYS_TAG, "%s System is initialized: %s\n", __func__, esp_err_to_name(ESP_OK));
				break;
			}
			case APP_STATE_TASKS_INIT:
			{
				tasks_init();
				break;
			}
			case APP_STATE_ERROR:
			{
				ESP_LOGE(SYS_TAG, "%s System encountered error!: %s\n", __func__, esp_err_to_name(ESP_FAIL));
				break;
			}
			/* The default state should never be executed. */
			default:
			{
				
				appData.state = APP_STATE_INIT;

				break;
			}
		}
	}
2-app_init function call initializes ble
	
	static void app_init(){
   
		if(ble_init()){
			 ESP_LOGI(SYS_TAG, "%s ble initialized !: %s\n", __func__, esp_err_to_name(ESP_OK));
			appData.state=APP_STATE_TASKS_INIT;
			send_system_event(APP_STATE_CHANGED);
		}
		else
			 ESP_LOGE(SYS_TAG, "%s BLE init failed: %s\n", __func__, esp_err_to_name(ESP_FAIL));

		}
	}
	here is my ble initializaton method copied as it is given in the examples.
	bool ble_init(void)
	{
    esp_err_t ret;
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    // Initialize NVS
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));

    ret = esp_bt_controller_init(&bt_cfg);
[... 254 lines cut ...]
				spp_profile_tab[SPP_PROFILE_APP_IDX].gatts_if = gatts_if;
			}
			else
			{
				ESP_LOGI(GATTS_TABLE_TAG, "Reg app failed, app_id %04x, status %d\n", param->reg.app_id, param->reg.status);
				return;
			}
		}

		do
		{
			int idx;
			for (idx = 0; idx < SPP_PROFILE_NUM; idx++)
			{
				if (gatts_if == ESP_GATT_IF_NONE || /* ESP_GATT_IF_NONE, not specify a certain gatt_if, need to call every profile cb function */
					gatts_if == spp_profile_tab[idx].gatts_if)
				{
					if (spp_profile_tab[idx].gatts_cb)
					{
						spp_profile_tab[idx].gatts_cb(event, gatts_if, param);
					}
				}
			}
		} while (0);
	}

xulongzhuang, 2022-12-01:
The above code does not look much different from ble_spp_server. Since the above code is not complete, I used [ble_spp_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server) to do the following tests:
1. ble_spp_server_demo.h opens the `SUPPORT_HEARTBEAT` macro
2. In `case ESP_GATTS_DISCONNECT_EVT: ` add and print as follows
`ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x%x", param->disconnect.reason);`
3. The BLE app on the mobile phone scans and connects to the BLE_SPP_SERVER device
4. After waiting more than ten seconds, the BLE_SPP_SERVER device will call [esp_ble_gap_disconnect(spp_remote_bda)](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_spp_server/main/ble_spp_server_demo.c#L423) to actively disconnect

[image]
As can be seen in the figure above, there is a disconnection event

If your code behavior is inconsistent with the above process, please tell me more detailed information, such as the Android client used, the reproduced demo and process, etc.

nouwon, 2022-12-01:
I am migrating to esp-idf ver 5.0 current version is 4.4.3.
As soon as I make it run with the new esp-idf version I will comment about the result.
Thank you very much for your kind efforts to resolve the issue.
May I learn your esp-idf version?

nouwon, 2022-12-02:
@xulongzhuang 
Problm seemsto be a hardware problem.I could not figure out why but I changed the board and now the callback event is fired properly.
Thank you for the help.