espressif/esp-idf #10279: Change to IDF v5.1 (master) causing SPI related link errors

ammaree, 2022-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2063-g593335578f

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### What is the expected behavior?

Compile and link to be successful

### What is the actual behavior?

Compiles successfully but linker fails with errors.

### Steps to reproduce.

Part of a complex app with many IDF and private components but build fails with any config including the SPI adapter.

### Build or installation Logs.

```plain
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.literal.spi_master_init_driver+0x40): undefined reference to `spi_hal_init'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.literal.spi_master_init_driver+0x4c): undefined reference to `spi_hal_deinit'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.6.literal+0x0): undefined reference to `spi_hal_fetch_result'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.0.literal+0x0): undefined reference to `spi_hal_setup_device'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.5.literal+0x0): undefined reference to `spi_hal_setup_trans'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.5.literal+0x4): undefined reference to `spi_hal_prepare_data'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.5.literal+0x8): undefined reference to `spi_hal_user_start'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.iram1.7.literal+0x18): undefined reference to `spi_hal_usr_is_done'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj):(.literal.spi_bus_add_device+0x48): undefined reference to `spi_hal_cal_clock_conf'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_master_init_driver':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:241: undefined reference to `spi_hal_init'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:255: undefined reference to `spi_hal_deinit'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_master_deinit_driver':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:277: undefined reference to `spi_hal_deinit'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_post_trans':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:604: undefined reference to `spi_hal_fetch_result'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_setup_device':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:502: undefined reference to `spi_hal_setup_device'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_new_trans':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:589: undefined reference to `spi_hal_setup_trans'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:590: undefined reference to `spi_hal_prepare_data'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:595: undefined reference to `spi_hal_user_start'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_intr':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:619: undefined reference to `spi_hal_usr_is_done'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/driver/libdriver.a(spi_master.c.obj): in function `spi_bus_add_device':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:371: undefined reference to `spi_hal_cal_clock_conf'
```

wanckl, 2022-12-01:
Hello?I have a try with your question but it is ok for me. but what the version you mean as `v5.1-dev-2063-g593335578f`, is it the tag `v5.1-dev` or you use a installation pkg? in my experience, you can:   
1. try to update your tool-chain    
2. try to build a example to see if there is the same issue   
3. `esp-idf` use a components-manager now, check if you add `REQUIRES driver` in `main/CMakeLists.txt` of your project   
4. check if you add some special compile option flags

ammaree, 2022-12-01:
@wanckl 
Thanks for the suggestions.
1. Toolchain test done, using latest.
2. not sure building of an example will be conclusive proof.
3. "driver" is not an external/managed component so not component manager related as far as I know. But, "driver" is listed as a REQUIRE in main/CMakeLists.txt
4. Compile flags, difficult to say where to start.

The messages seem to indicate, at least to me, that the "driver" component is being built and that the "spi_master.c" APIs are included, but that "spi_hal_????" APIs, all part of the "hal" component cannot be found.

Similarly "ledc.c" cannot find "led_hal_??????" APIs, all of which also form part of the "hal" component.
Only a single component in my codebase refer to "ledc.c" APIs and that module has "PRIV_REQUIRES driver"

My code does not in any way reference "hal" functions directly, only through "spi" and "led" as being part of "drivers"
So, to me it looks like component "driver" not finding component "hal" at link time.

@igrr maybe you can in your wisdom point me at my mistake, or identify this as a bug?

Icarus113, 2022-12-05:
Could you help check the hal component CMakeLists.txt file in your local project? To see if the needed `spi_hal_x.c`s and `ledc_hal_x.c`s are in the source file list?
- You may directly do `message(SRCS="${srcs}")` to message out the source file list, 
- or by
  ```
  idf_component_get_property(my_srcs hal SRCS)
  message(STATUS "my_srcs is: ${my_srcs}")
  ```
to check the registered source files. 
Then do a `idf.py reconfigure` to re-run the cmake to see the needed messages. 

Note if you do it in the latter way, you need to put the `idf_component_get_property` after the hal component `idf_component_register` is called.

@ammaree

ammaree, 2022-12-05:
@wanckl
Confirmed that `REQUIRES hal` is indeed present in `components/driver/CMakeFiles.txt`

@Icarus113  
Result of adding `message(SRCS="${srcs}")` into `components/hal/CMakelists.txt` yields...

`SRCS="mpu_hal.cefuse_hal.cesp32/efuse_hal.cmmu_hal.cwdt_hal_iram.crtc_io_hal.cspi_hal.cspi_hal_iram.cspi_slave_hal.cspi_slave_hal_iram.ctimer_hal.ctimer_hal_iram.cgpio_hal.cuart_hal.cuart_hal_iram.cspi_flash_hal.cspi_flash_hal_iram.cspi_flash_encrypt_hal_iram.cadc_hal_common.cadc_oneshot_hal.cledc_hal.cledc_hal_iram.ci2c_hal.ci2c_hal_iram.crmt_hal.cpcnt_hal.cmcpwm_hal.ctwai_hal.ctwai_hal_iram.ci2s_hal.csdm_hal.cadc_hal.csha_hal.csdio_slave_hal.ctouch_sensor_hal.caes_hal.cesp32/brownout_hal.cesp32/touch_sensor_hal.cesp32/gpio_hal_workaround.c"`

So, it seems the required `spi_hal????.c` and `ledc_hal???.c` are there, or am I missing something?

Icarus113, 2022-12-05:
This is kinda weird. The error logs show that those `spi_hal_xx` symbols are to be resolved, but the references cannot be found. Seeing all of the `spi_hal_xx` functions in `spi_hal.c`/`spi_hal_iram.c` are missing, I can only suspect that these two source files are not in the build.

Could you please help check in your local project build directory `<your-project-dir>/build/esp-idf/hal/CMakeFiles/__idf_hal.dir` to see if `spi_hal.c.obj`, `spi_hal_iram.c.obj`, `ledc_hal.c.obj`, `ledc_hal_iram.c.obj` are there? I believe these obj files are to be dealt by the linker. 

If there you find the needed .obj files, then you do a for example
```
nm spi_hal.c.obj
```
to list the symbol table, to see if you can find the needed function symbols.

So till now you'll know if it's the whole `.c` file missing, or the definition of those functions are missing. 

@ammaree

ammaree, 2022-12-05:
@Icarus113 

All 4 the .obj files are there.
`__idf_hal.dir % nm spi_hal.c.obj                                                                                                                                                                                                                
         U SPI1
         U SPI2
         U SPI3
00000000 r SPI_HAL_TAG
00000000 r __FUNCTION__$0
         U __assert_func
         U __divdf3
         U __floatsidf
00000000 r __func__$1
         U esp_log_write
         U memset
00000000 t s_spi_hal_dma_init_config
00000000 T spi_hal_cal_clock_conf
00000000 T spi_hal_cal_timing
00000000 T spi_hal_deinit
00000000 T spi_hal_get_freq_limit
00000000 T spi_hal_init
00000000 T spi_hal_master_cal_clock`

`__idf_hal.dir % nm spi_hal_iram.c.obj
         U __assert_func
00000000 r __func__$0
00000000 r __func__$1
         U lldesc_setup_link_constrained
         U memcpy
00000000 T spi_hal_fetch_result
00000000 T spi_hal_prepare_data
00000000 T spi_hal_setup_device
00000000 T spi_hal_setup_trans
00000000 T spi_hal_user_start
00000000 T spi_hal_usr_is_done`

`__idf_hal.dir % nm ledc_hal.c.obj
         U LEDC
         U __assert_func
00000000 r __func__$0
00000000 T ledc_hal_get_clk_cfg
00000000 T ledc_hal_init`

`__idf_hal.dir % nm ledc_hal_iram.c.obj
00000000 T ledc_hal_clear_fade_end_intr_status
00000000 T ledc_hal_get_duty
00000000 T ledc_hal_get_fade_end_intr_status
00000000 T ledc_hal_ls_channel_update
00000000 T ledc_hal_set_duty_cycle
00000000 T ledc_hal_set_duty_direction
00000000 T ledc_hal_set_duty_num
00000000 T ledc_hal_set_duty_scale
00000000 T ledc_hal_set_hpoint`

At a quick glance it seems all the symbols are there as well...

ammaree, 2022-12-05:
@Icarus113 

Also checked in libhal.a that all the same symbols for all 4 file-sets are present there.

o-marshmallow, 2022-12-05:
Are you storing your IDF project in a Dropbox-synchronized folder? I see `Dropbox/` in your path?

If that's the case, can you try moving your project out of it just to have a try? It may be unrelated but I saw a lot of weird errors related to files, symbolic links, hard links, permissions etc when using Dropbox or Google Drive-synchronized folders.

o-marshmallow, 2022-12-05:
@ammaree Are you storing your IDF project in a Dropbox-synchronized folder? I see Dropbox/ in your path?

If that's the case, can you try moving your project out of it just to have a try? It may be unrelated but I saw a lot of weird errors related to files, symbolic links, hard links, permissions etc when using Dropbox or Google Drive-synchronized folders.

ammaree, 2022-12-09:
After having cleaned my build environment as far a REQUIRES, PRIV_REQUIRES and COMPONENTS are concerned the link problems are becoming more and more strange.

The latest errors refer to symbols that do not exist anywhere in my code, but they all appear to be concatenations of symbols or some form of .obj or .a file corruption.

`/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x94): undefined reference to '_Z11xEpGDI_InitP19tskTaskControlBlock'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0xa0): undefined reference to '_Z13vEpGDI_DeInitv'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function 'app_main':
/Users/andremaree/DevSpace/z-projects/irmacs/main/main.cpp:149: undefined reference to '_Z11xEpGDI_InitP19tskTaskControlBlock'
/Users/andremaree/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/andremaree/DevSpace/z-projects/irmacs/main/main.cpp:198: undefined reference to '_Z13vEpGDI_DeInitv'`

This `_Z11xEpGDI_InitP19tskTaskControlBlock` appear to include 2 separate symbols, `xEpGDI_Init` and `tskTaskControlBlock` joined together with `_Z11` and `P19` to miraculously create a new symbol.

Similarly `_Z13vEpGDI_DeInitv` comprises of `vEpGDI_DeInit` ( a valid symbol) with `_Z13` and `v` added to the start and end.

Any suggestions....

ammaree, 2022-12-10:
Suspecting it might be related to a corrupt build environment I have now deleted the complete esp-idf as well as the .espressif directories, reinstalled everything from scratch, recompiled and the exact same error at link stage.

The symbols are the same ones listed in the post above.

If it was memory corruption I would expect the results to be random, based on applications loaded and process running, but there is nothing inconsistent in the result.

Any suggestions?

@igrr @o-marshmallow @david-cermak

igrr, 2022-12-11:
The names of the missing symbols are not corrupted. They are the result of [C++ name mangled](https://en.wikipedia.org/wiki/Name_mangling#C++) representation of the following function signatures:
* `xEpGDI_Init(tskTaskControlBlock*)`
* `vEpGDI_DeInit(void)`

Are these functions defined (implemented) in a .c or a .cpp source file?

Are these functions defined in the same component as your `main.cpp`?

If they are in a different component, is that component present in the `REQURES` or `PRIV_REQUIRES` list of your `main` component?

ammaree, 2022-12-11:
@igrr 

You once again saved my backside....

My first attempt to convert some modules over to C++ caused this learning curve. I successfully made the required changes converting my "main.c" to "main.cpp" but in the larger scale restructuring I created separate header files, some of which did not get the c++ guards.

Added those and the problem gone, now I can continue testing and fixing the rest....

Thanks again!

closed 2022-12-11 by ammaree