espressif/esp-idf #10284: Error sending large packet wss

saiquaan5, 2022-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

commit 6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7 (HEAD, tag: v4.4.3)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32s3

### Power Supply used.

External 3.3V

### What is the expected behavior?

I am developing a gateway suite with esp32s3. Use pppos USB with SIM7600 but can test with wifi and ethernet.
I use protobuf-c to send binary packets on outgoing.
The size of each packet I send is more than 2000 bytes.
I have increased the buffer size in the websocket_client.c file but after 2 successful submissions I get this error again. Look forward to the help

I (74228) MAIN: Has event schedule rs485_handle
E (74236) APP_WEBSOCKET: Size of schedule message 2263
E (74237) esp-aes: Failed to allocate memory
E (74244) esp-tls-mbedtls: write error :-0x0001:
E (74249) TRANSPORT_BASE: esp_tls_conn_write error, errno=Success
E (74256) TRANSPORT_WS: Error write header
E (74260) WEBSOCKET_CLIENT: Network error: esp_transport_write() returned -1, errno=0
E (74270) TRANSPORT_BASE: poll_read select error 0, errno = Success, fd = 54
I (74270) WEBSOCKET_CLIENT: Reconnect after 10000 ms
E (74277) WEBSOCKET_CLIENT: Network error: esp_transport_poll_read() returned -1, errno=119
I (74283) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (74297) MAIN: disconnected wss event
I (74292) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (74307) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (74313) MAIN: disconnected wss event
I (74312) MAIN: Has event log rs485_handle
I (74322) MAIN: mtfc Offline
I (74375) MAIN: Has event log rs485_handle
I (74376) MAIN: mtfc Offline

I use Ringbuffer. and my submit function

app_websocket_handle_t app_websocket_init(app_websocket_config_t* config) {
    app_websocket_handle_t handle = malloc(sizeof(app_websocket_t));
    handle->network_handle = config->network_handle;
    handle->app_rs485_handle = config->rs485_handle;
    handle->callback = config->callback;
    handle->websocket_connected = false;
    handle->websocket_url = strdup(config->websocket_url);
    handle->is_start_websocket = false;
    handle->ws_tx_buffer = malloc(WS_BUFFER_SIZE);
    handle->rx_rb = xRingbufferCreate(DEFAULT_MESH_RINGBUFFER_SIZE, RINGBUF_TYPE_NOSPLIT);
    handle->tx_rb = xRingbufferCreate(DEFAULT_MESH_RINGBUFFER_SIZE, RINGBUF_TYPE_NOSPLIT);
    handle->device_id = NULL;
    handle->last_login_tick = -LOGIN_TIMEOUT_MS;
    handle->last_ping_tick = -PING_TIMEOUT_MS;
    handle->last_timeout_ping = -PING_TIMEOUT_MS;
[... 471 lines cut ...]
E (427804) esp-tls: create_ssl_handle failed
E (427804) esp-tls: Failed to open new connection
E (427810) TRANSPORT_BASE: Failed to open a new connection
E (427816) TRANSPORT_WS: Error connecting to host dev.hue-traffic.sitech.com.vn:443
E (427824) WEBSOCKET_CLIENT: Error transport connect
I (427830) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (427836) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (427841) MAIN: disconnected wss event
I (427857) MAIN: Has event log rs485_handle
I (427857) MAIN: mtfc Offline
E (438744) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (438744) esp-tls: create_ssl_handle failed
E (438745) esp-tls: Failed to open new connection
E (438750) TRANSPORT_BASE: Failed to open a new connection
E (438757) TRANSPORT_WS: Error connecting to host dev.hue-traffic.sitech.com.vn:443
E (438765) WEBSOCKET_CLIENT: Error transport connect
I (438770) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (438776) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (438782) MAIN: disconnected wss event
I (438797) MAIN: Has event log rs485_handle
I (438797) MAIN: mtfc Offline
E (449647) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (449647) esp-tls: create_ssl_handle failed
E (449648) esp-tls: Failed to open new connection
```

saiquaan5, 2022-12-01:
ALL LOG

saiquaan5, 2022-12-01:
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (117 compressed) at 0x00008000 in 0.1 seconds (effective 252.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/mac/project/GW-Traffic-SDKD-1.0/GW-Traffic-sdkd_ver1.1/usb_cdc_4g_module
Executing "/Users/mac/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/mac/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Users/mac/project/GW-Traffic-SDKD-1.0/GW-Traffic-sdkd_ver1.1/usb_cdc_4g_module/build/cdc_4g.elf -m '/Users/mac/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/Users/mac/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0001' '-b' '921600'"...
--- idf_monitor on /dev/cu.usbserial-0001 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
nt 3: paddr=ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17c0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x2fec
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (25) boot: compile time 10:27:38
I (25) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2d9d8h (186840) map
I (133) esp_image: segment 1: paddr=0003da00 vaddr=3fc9d230 size=02618h (  9752) load
I (135) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=9af7ch (634748) map
I (235) esp_image: segment 3: paddr=000dafa4 vaddr=3fc9f848 size=01b64h (  7012) load
I (237) esp_image: segment 4: paddr=000dcb10 vaddr=40378000 size=1522ch ( 86572) load
I (258) esp_image: segment 5: paddr=000f1d44 vaddr=50000000 size=00010h (    16) load
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
I (278) cpu_start: Pro cpu up.
I (278) cpu_start: Starting app cpu, entry point is 0x40379600
0x40379600: call_start_cpu1 at /Users/mac/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (292) cpu_start: Pro cpu start user code
I (292) cpu_start: cpu freq: 240000000
I (292) cpu_start: Application information:
I (295) cpu_start: Project name:     cdc_4g
I (300) cpu_start: App version:      db54cff-dirty
I (305) cpu_start: Compile time:     Dec  1 2022 10:27:34
I (311) cpu_start: ELF file SHA256:  8ed56125a6a0cf0f...
I (317) cpu_start: ESP-IDF:          v4.4.3-dirty
I (323) heap_init: Initializing. RAM available for dynamic allocation:
I (330) heap_init: At 3FCA4D10 len 00044A00 (274 KiB): D/IRAM
I (336) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (343) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (350) spi_flash: detected chip: generic
I (354) spi_flash: flash io: qio
W (358) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (371) sleep: Configure to isolate all GPIO pins in sleep state
I (378) sleep: Enable automatic switching of GPIO sleep configuration
I (385) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (400) MAIN: HardwareId: GW-3C:4A:D9:FA
W (422) MAIN: System using 4g/wifi
I (422) APP_NETWORK: ----------------------- System using 4G -----------------------
[... 558 lines cut ...]
E (74250) TRANSPORT_WS: Error write header
E (74254) WEBSOCKET_CLIENT: Network error: esp_transport_write() returned -1, errno=0
E (74264) TRANSPORT_BASE: poll_read select error 0, errno = Success, fd = 54
I (74264) WEBSOCKET_CLIENT: Reconnect after 10000 ms
E (74271) WEBSOCKET_CLIENT: Network error: esp_transport_poll_read() returned -1, errno=119
I (74277) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (74291) MAIN: disconnected wss event
I (74286) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (74301) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (74307) MAIN: disconnected wss event
I (74307) MAIN: Has event log rs485_handle
I (74316) MAIN: mtfc Offline
I (74322) MAIN: Has event log rs485_handle
I (74324) MAIN: mtfc Offline
E (84797) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (84797) esp-tls: create_ssl_handle failed
E (84798) esp-tls: Failed to open new connection
E (84803) TRANSPORT_BASE: Failed to open a new connection
E (84809) TRANSPORT_WS: Error connecting to host dev.hue-traffic.sitech.com.vn:443
E (84817) WEBSOCKET_CLIENT: Error transport connect
I (84823) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (84828) APP_WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (84834) MAIN: disconnected wss event
I (84849) MAIN: Has event log rs485_handle
I (84849) MAIN: mtfc Offline

chipweinberger, 2022-12-03:
Your post is a little hard to read.

You should format it using expanding sections 

https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab

and with code blocks

https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

saiquaan5, 2022-12-04:
Ok i try. thanks for rep :d

ESP-YJM, 2022-12-05:
@saiquaan5 From your log, it seems it failed with `memory allocation failed`. You need check whether your memory is enough and whether there is a memory leak in the application.

Alvin1Zhang, 2023-01-30:
@saiquaan5 Thanks for reporting and sharing the updates. Would you please help share if any further updates for the issue? Thanks.

mahavirj, 2023-02-14:
Closing. Please drop a comment if you are still facing the issue.

closed 2023-02-14 by mahavirj