espressif/esp-idf #10286: How can a BLE mesh node leave the mesh network by itself ?

tonyho, 2022-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Sometimes, the node would want to leave the network and do a factory reset. 
What is the leaving and removing (keys) API of a Node? 
If we use a ESP Provisioner, how to reset and delete the whole network and all nodes information as an action of factory reset?

chegewara, 2024-03-02:
```
/**
 * @brief         Reset the provisioning procedure of the local BLE Mesh node.
 *
 * @note          All provisioning information in this node will be deleted and the node
 *                needs to be re-provisioned. The API function esp_ble_mesh_node_prov_enable()
 *                needs to be called to start a new provisioning procedure.
 *
 * @return        ESP_OK on success or error code otherwise.
 *
 */
esp_err_t esp_ble_mesh_node_local_reset(void);
```

Alvin1Zhang, 2025-03-31:
Thanks for reporting, and thanks to @chegewara  for the answer, feel free to reopen if the issue still happens.

closed 2025-03-31 by Alvin1Zhang