espressif/esp-idf #10287: BLE Mesh Fast Provisioning "Too short message for OpCode 0x00c702e5"

Shane555, 2022-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-863-g550772ab32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32C3-Mini1 Dev Kit

### Power Supply used.

USB

### What is the expected behavior?

I used 3 ble_mesh_fast_prov_server and 1 ble_mesh_fast_prov_client demo to test. When i set `.max_node_num = 0x02,` in example_prov_info_t struct in client, i expect the provisioning to occur quicker than if it is set to 0x01. 

### What is the actual behavior?

The client got stuck in a ESP_BLE_MESH_CLIENT_MODEL_SEND_TIMEOUT_EVT loop with error `BLE_MESH: Too short message for OpCode 0x00c702e5`. It seems to me that there remains a server node provisioned directly by the client node that did not provision any other unprovisioned nodes and it sends a ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR_STATUS with 0 bytes. Hence the error. This should be accounted for and the case should be caught so as to allow the client to exit the loop and continue on with the fast provisioning sequence.

### Steps to reproduce.

1. Set .max_node_num = 0x02 in example_prov_info_t struct in client demo code
2. Flash fast provisioning server demo into 3 devkit
3. Flash fast provisioning client demo into 1 devkit
4.  Check uart output and see that client is stuck in a callback event.
...

### Debug Logs.

```plain
I (1133269) EXAMPLE: ESP_BLE_MESH_CLIENT_MODEL_SEND_TIMEOUT_EVT, opcode 0xc602e5, dst 0x0006
W (1133269) FAST_PROV_CLIENT: example_fast_prov_client_recv_timeout: Receive fast prov server status timeout
I (1133279) EXAMPLE: ESP_BLE_MESH_MODEL_SEND_COMP_EVT, err_code 0
E (1133289) BLE_MESH: Too short message for OpCode 0x00c702e5
I (1143279) EXAMPLE: ESP_BLE_MESH_CLIENT_MODEL_SEND_TIMEOUT_EVT, opcode 0xc602e5, dst 0x0006
W (1143279) FAST_PROV_CLIENT: example_fast_prov_client_recv_timeout: Receive fast prov server status timeout
I (1143289) EXAMPLE: ESP_BLE_MESH_MODEL_SEND_COMP_EVT, err_code 0
E (1143309) BLE_MESH: Too short message for OpCode 0x00c702e5
I (1153289) EXAMPLE: ESP_BLE_MESH_CLIENT_MODEL_SEND_TIMEOUT_EVT, opcode 0xc602e5, dst 0x0006
W (1153289) FAST_PROV_CLIENT: example_fast_prov_client_recv_timeout: Receive fast prov server status timeout
I (1153309) EXAMPLE: ESP_BLE_MESH_MODEL_SEND_COMP_EVT, err_code 0
E (1153329) BLE_MESH: Too short message for OpCode 0x00c702e5
I (1163309) EXAMPLE: ESP_BLE_MESH_CLIENT_MODEL_SEND_TIMEOUT_EVT, opcode 0xc602e5, dst 0x0006
W (1163309) FAST_PROV_CLIENT: example_fast_prov_client_recv_timeout: Receive fast prov server status timeout
I (1163319) EXAMPLE: ESP_BLE_MESH_MODEL_SEND_COMP_EVT, err_code 0
E (1163359) BLE_MESH: Too short message for OpCode 0x00c702e5
```

### More Information.

If i used 1 server nodes and 1 client nodes with max_node_num = 0x01 , error always occurs as the server node directly provisioned by client did not provision any other nodes. For the 3 server nodes and 1 client node case, error occurs randomly depending on which node provisioned which first when i set max_node_num = 0x02

pranav-anantharam, 2023-03-27:
@JiaLWang @Shane555 Hi all, I also face the exact same bug with the fast provisioning example. Are there any updates on this issue?

pranav-anantharam, 2023-03-27:
Hi all, I also face the exact same bug with the fast provisioning example. Are there any updates on this issue?

Shane555, 2023-03-28:
Hi @pranav-anantharam  for now the quick solution would be to ensure top provisioner only provisions a primary provisioner. Then always ensure that primary provisioner can provision at least one node. This means the example can only work with 3 or more nodes. Basically reduce the chances that the primary provisioner does not provision any nodes