espressif/esp-idf #10288: I2s ouput not working with psram enabled.

nikhil-robinson, 2022-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-173-g1f63dc70b8

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

Custom board esp32s3 N8R8

### Power Supply used.

USB

### What is the expected behavior?

Play an mp3 file from sdcard to i2s. which works fine without psram. ( the log of code without psram)

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1670
load:0x403c9700,len:0xbb0
load:0x403cc700,len:0x2f0c
entry 0x403c9950
I (24) boot: ESP-IDF v4.4.3-173-g1f63dc70b8 2nd stage bootloader
I (25) boot: compile time 15:48:23
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ef78h ( 61304) map
I (104) esp_image: segment 1: paddr=0001efa0 vaddr=3fc92600 size=01078h (  4216) load
I (105) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2d998h (186776) map
I (144) esp_image: segment 3: paddr=0004d9c0 vaddr=3fc93678 size=01970h (  6512) load
I (145) esp_image: segment 4: paddr=0004f338 vaddr=40374000 size=0e5f8h ( 58872) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (179) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x403752b0
0x403752b0: call_start_cpu1 at /home/nikhil/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000
I (194) cpu_start: Application information:
I (196) cpu_start: Project name:     main
I (201) cpu_start: App version:      f82540e
I (206) cpu_start: Compile time:     Dec  1 2022 15:48:23
I (212) cpu_start: ELF file SHA256:  ab48415abb72c62c...
I (218) cpu_start: ESP-IDF:          v4.4.3-173-g1f63dc70b8
[... 206 lines cut ...]
I (810) sleep: Enable automatic switching of GPIO sleep configuration
I (817) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (838) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (838) PC_STORAGE: Initializing SD card
I (848) PC_STORAGE: Using SDMMC peripheral
I (848) PC_STORAGE: Mounting filesystem
I (858) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (868) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (878) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (918) PC_STORAGE: Filesystem mounted
Name: 8310 
Type: SDSC
Speed: 20 MHz
Size: 481MB
I (938) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=4
I (938) I2S: I2S0, MCLK output by GPIO0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x39 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
0x403743c0: _DoubleExceptionVector at /home/nikhil/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:461

```
```

igrr, 2022-12-01:
@nikhil-robinson It looks like some of the pins you are using for I2S might be overlapping with pins reserved for the in-package PSRAM. Please check the ESP32-S3 Datasheet, section "2.6 Pin-to-Pin Mapping Between Chip and In-Package Flash/PSRAM". The chip package with octal PSRAM connects GPIO35 and GPIO37 of ESP32-S3 with DQ6 and DQS signals of PSRAM.

nikhil-robinson, 2022-12-15:
@igrr yes it was the problem

closed 2022-12-15 by nikhil-robinson