espressif/esp-idf #10291: Multiple ESP32 Openthread Border Routers on the same network cause devices not to be able to connect to the internet

meatherly, 2022-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

When 2 border routers are on the same network devices should be able to connect to the internet. 

### What is the actual behavior?

When 2 border routers are on the same network no device can connect to the internet.

### Steps to reproduce.

* Set up 2 border routers using the ESP32
* Allow them to form a thread network
* Get the preferred NAT64 Prefix using `netdata show`
* Program a 3rd OpenThread node to ping a external endpoint every 10 seconds. 
* Flash and allow the 3rd OpenThead node to join the network.

You'll notice the 3rd node will get ping timeouts when there are 2 ESP32 border routers on the same network.

### Debug Logs.

```plain
# 3rd OpenThread Node Ping CLI Logs

uart:~$ ot ping 8.8.8.8
Pinging synthesized IPv6 address: fdae:a713:fe2b:2:0:0:808:808
[00:03:45.245,025] <inf> [I] Icmp6---------: Sent echo request: (seq = 2)
[00:03:45.249,725] <inf> [I] MeshForwarder-: Sent IPv6 ICMP6 msg, len:56, chksum:f842, ecn:no, to:0x2000, sec:yes, prio:normal
[00:03:45.250,061] <inf> [I] MeshForwarder-:     src:[fdf8:2e54:ed95:1:e895:72be:d4a9:1aac]
[00:03:45.250,366] <inf> [I] MeshForwarder-:     dst:[fdae:a713:fe2b:2:0:0:808:808]
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done

# The ESP32 Border Router that received the request logs:

I(492429) OPENTHREAD:[I] MeshForwarder-: Received IPv6 ICMP6 msg, len:56, chksum:f842, ecn:no, from:0xa000, sec:yes, prio:normal, rss:-51.0
I(492429) OPENTHREAD:[I] MeshForwarder-:     src:[fdf8:2e54:ed95:1:e895:72be:d4a9:1aac]
I(492439) OPENTHREAD:[I] MeshForwarder-:     dst:[fdae:a713:fe2b:2:0:0:808:808]

# When there is only one border router on the network

# 3rd OpenThread Node Ping CLI Logs

uart:~$ ot ping 8.8.8.8
Pinging synthesized IPv6 address: fd8a:993c:b044:2:0:0:808:808
16 bytes from fd8a:993c:b044:2:0:0:808:808: icmp_seq=1 hlim=53 time=88ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 88/88.0/88 ms.
Done

# The ESP32 Border Router that received the request logs:

I(33174) OPENTHREAD:[I] MeshForwarder-: Received IPv6 ICMP6 msg, len:56, chksum:8ed7, ecn:no, from:0x6c01, sec:yes, prio:normal, rss:-43.0
I(33174) OPENTHREAD:[I] MeshForwarder-:     src:[fd8a:993c:b044:1:3db2:7a69:646b:edb6]
I(33194) OPENTHREAD:[I] MeshForwarder-:     dst:[fd8a:993c:b044:2:0:0:808:808]
I(33244) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 ICMP6 msg, len:56, chksum:8dd7, ecn:no, to:0x6c01, sec:yes, prio:normal
I(33254) OPENTHREAD:[I] MeshForwarder-:     src:[fd8a:993c:b044:2:0:0:808:808]
I(33264) OPENTHREAD:[I] MeshForwarder-:     dst:[fd8a:993c:b044:1:3db2:7a69:646b:edb6]
```

### More Information.

I'm using the [IDF openthread example code](https://github.com/espressif/esp-idf/tree/release/v5.0/examples/openthread/ot_br) to build both border routers. I haven't modified the code. Here's my `sdkconfig.defaults`

```
#
# libsodium
#
CONFIG_LIBSODIUM_USE_MBEDTLS_SHA=y
# end of libsodium

#
# Partition Table
#
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_PARTITION_TABLE_MD5=y
# end of Partition Table

#
# mbedTLS
#

CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECJPAKE=y
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
# end of TLS Key Exchange Methods

CONFIG_MBEDTLS_ECJPAKE_C=y
# end of mbedTLS

#
# OpenThread
#
CONFIG_OPENTHREAD_ENABLED=y
CONFIG_OPENTHREAD_BORDER_ROUTER=y
CONFIG_OPENTHREAD_BR_AUTO_START=y
CONFIG_OPENTHREAD_NETWORK_NAME="OpenThread"
CONFIG_OPENTHREAD_NETWORK_CHANNEL=15
CONFIG_OPENTHREAD_NETWORK_PANID=0x1234
CONFIG_OPENTHREAD_NETWORK_EXTPANID="dead00beef00cafe"
CONFIG_OPENTHREAD_NETWORK_MASTERKEY="00112233445566778899aabbccddeeff"
CONFIG_OPENTHREAD_NETWORK_PSKC="104810e2315100afd6bc9215a6bfac53"
# end of OpenThread

# FreeRTOS Kernel
# increasing timer stack to prevent stack overflow
CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=2560
# end of FreeRTOS Kernel

#
# lwIP
#
CONFIG_LWIP_IPV6_FORWARD=y
CONFIG_LWIP_IPV6_NUM_ADDRESSES=8
CONFIG_LWIP_MULTICAST_PING=y
CONFIG_LWIP_NETIF_STATUS_CALLBACK=y
CONFIG_LWIP_HOOK_IP6_ROUTE_DEFAULT=y
CONFIG_LWIP_HOOK_ND6_GET_GW_DEFAULT=y
CONFIG_LWIP_HOOK_IP6_INPUT_CUSTOM=y
CONFIG_LWIP_IPV6_AUTOCONFIG=y
CONFIG_LWIP_NETBUF_RECVINFO=y
# end of lwIP

#
# mDNS
#
CONFIG_MDNS_STRICT_MODE=y
CONFIG_MDNS_MULTIPLE_INSTANCE=y
# end of mDNS

#
# ESP System Settings
#
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=3584
# end of ESP System Settings

zwx1995esp, 2022-12-14:
Hi @meatherly The fix MR has already merged in our internal branch,  please just wait it syncs to github.

chshu, 2023-09-11:
Fixed by https://github.com/espressif/esp-idf/commit/98a1e9d78676cfe571034d939896c3908b2b8337

closed 2023-09-11 by chshu