espressif/esp-idf #10295: __attribute__((weak))  not work

sidhoo22, 2022-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

tag  : v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

expected print  log : __attribute__  work !

### What is the actual behavior?

the actual log :ops , __attribute__  not work !

### Steps to reproduce.

file1.c 
#include "file2.h"
__attribute__((weak)) int app_init(void)
{
    printf("ops , __attribute__  not work !\n");
    return 0;
}

// main entry
void main(void){
   app_init();
}

file2.c
 int app_init(void)
{
    printf(" __attribute__  work !\n");
    return 0;
}

expected print  log : __attribute__  work !

ma89vik, 2022-12-02:
This isnt really a bug, just how the linker works, .e.g see https://stackoverflow.com/questions/13089166/how-to-make-gcc-link-strong-symbol-in-static-library-to-overwrite-weak-symbol

IDF provides a convenient way of setting `-whole-archive` with `WHOLE_ARCHIVE`, see https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/plugins#ensuring-that-the-plugin-code-is-included-into-the-executable for some more explanation.

ESP-Marius, 2022-12-02:
This isnt really a bug as far as I can see, just how the linker works, .e.g see https://stackoverflow.com/questions/13089166/how-to-make-gcc-link-strong-symbol-in-static-library-to-overwrite-weak-symbol

IDF provides a convenient way of setting -whole-archive with WHOLE_ARCHIVE, see https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/plugins#ensuring-that-the-plugin-code-is-included-into-the-executable for some more explanation.

ljden, 2022-12-05:
@ESP-Marius @sidhoo22 Just a note - `WHOLE_ARCHIVE` is not available in v4.4.2 although it is fairly trivial to cherry pick (https://github.com/espressif/esp-idf/commit/273633ee310fbc18b17edfaeae3f3121508e3b8d). 
I had to this for my issue: https://github.com/espressif/esp-idf/issues/10103

o-marshmallow, 2022-12-06:
Hi @sidhoo22 ,

As v4.4.x doesn't have WHOLE_ARCHIVE, you can check this https://github.com/espressif/esp-idf/issues/9718#issuecomment-1311560206

It will force the linker to integrate your file containing the strong function in the binary.

sidhoo22, 2022-12-06:
@ljden @o-marshmallow oh i see. i'll test it later, thanks for your help

ESP-Marius, 2023-01-29:
Marking this as resolved then since the original question was answered and there doesnt seem to be any bugs related to IDF.

closed 2023-01-29 by ESP-Marius