espressif/esp-idf #10296: Using i2c clock speed at 8Mhz with esp32c2 + Main XTAL freequence=26MHz

nopnop2002, 2022-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-1840-gaad600c9ca-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

DOITING ESPC2-12 Development board

### Power Supply used.

USB

### What is the expected behavior?

Even with a Main XTAL freequence of 26MHz, it works fine with an i2c clock speed of 8MHz.

### What is the actual behavior?

With a Main XTAL freequence of 26MHz, the i2c clock speed of 8MHz does not work properly.

I used this sample under the following conditions:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

- ESP32C2

- Main XTAL freequence=26MHz
CONFIG_XTAL_FREQ_26=y

- i2c freequencey=8MHz
static uint32_t i2c_frequency = 800000;

### Steps to reproduce.

1. Set Main XTAL freequence=26MHz using menuconfig.    
2. Set the i2c frequency to 8MHz.    
3. Execute this example.    
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

### Debug Logs.

```plain
I ran ic2detect using i2c_frequency = 800000, but it doesn't scan properly.

i2c-tools> i2cdetect
I (6844) cmd_i2ctools: conf.master.clk_speed=800000
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
i2c-tools>

```
```

### More Information.

I ran ic2detect using i2c_frequency = 400000, it does scan properly.

```
I (17304) cmd_i2ctools: conf.master.clk_speed=400000
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- UU --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
```

chipweinberger, 2022-12-03:
just fyi, 

800,000 is 0.8MHz, not 8MHz.

nopnop2002, 2022-12-03:
@chipweinberger 

This is my typo.

Steps to reproduce.
Set Main XTAL freequence=26MHz using menuconfig.
Change the i2c frequency to 0.8MHz.
Execute this example.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

chipweinberger, 2022-12-03:
Most i2c devices only support 100kHz and 400kHz btw

which i2c device are you trying to communicate with?

nopnop2002, 2022-12-04:
@chipweinberger 

>which i2c device are you trying to communicate with?

ssd1306 128 x 64 Dot Matrix

krzychb, 2022-12-04:
Hi @nopnop2002,
From the datasheet https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf, it looks like the I2C clock of SSD1306 can run at 400 kHz max. See Table 13-6 :I2C Interface Timing Characteristics.

nopnop2002, 2022-12-04:
@krzychb 

>it looks like the I2C clock of SSD1306 can run at 400 kHz max. 

I understand.

closed 2022-12-04 by nopnop2002