espressif/esp-idf #10297: Memory overlap issue after upgrading from v4.0.1

afcec, 2022-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to updrage my project from esp-idf v4.0.1 to 4.1 with the purpose then to upgrade to the last stable release. We are using  a custom board based on ESP32-WROOM-32D that has been working successfully for the last two years. After adding the required changes to make the project work with esp-netif, I built the project and after flashing I first noticed a boot error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1812
load:0x40078000,len:12436
ho 0 tail 12 room 4
load:0x40080400,len:3360
entry 0x4008060c
ets Jun 8 2016 00:22:57

I enabled then the debug serial port and I can see the following:

(543) cpu_start: Pro cpu up.
I (543) cpu_start: Application information:
I (543) cpu_start: Project name: Conn_Board
I (574) cpu_start: App version: 2.2
I (629) cpu_start: Compile time: Dec 1 2022 13:50:34
I (702) cpu_start: ELF file SHA256: b0a602820b525504...
I (774) cpu_start: ESP-IDF: v4.1
I (830) cpu_start: Starting app cpu, entry point is 0x400813c0
I (0) cpu_start: App cpu up.
V (951) memory_layout: reserved range is 0x3f4429f8 - 0x3f442a20
D (1024) memory_layout: Checking 7 reserved memory ranges:
D (1091) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (1169) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffbda58
D (1247) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (1325) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (1403) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (1481) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (1559) memory_layout: Reserved memory range 0x40080000 - 0x400968ac
D (1637) memory_layout: Building list of available memory regions:
V (1712) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (1793) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (1908) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
V (1988) memory_layout: Examining memory region 0x3ffb0000 - 0x3ffb8000
V (2069) memory_layout: Region 0x3ffb0000 - 0x3ffb8000 inside of reserved 0x3ffb0000 - 0x3ffbda58
V (2176) memory_layout: Examining memory region 0x3ffb8000 - 0x3ffc0000
V (2256) memory_layout: Start of region 0x3ffb8000 - 0x3ffc0000 overlaps reserved 0x3ffb0000 - 0x3ffbda58
D (2372) memory_layout: Available memory region 0x3ffbda58 - 0x3ffc0000

** We are using flash encryption and this is our partition table:

nvs,		data,	nvs,		0x9000,		0x4000,	encrypted
otadata,	data,	ota,		0xd000,		0x2000,	encrypted
phy_init,	data,	phy,		0xf000,		0x1000,	encrypted
ota_0,		app,	ota_0,		0x10000,	1536K,
ota_1,		app,	ota_1,		0x190000,	1536K,
certs,		data,	255,		0x310000,	64K,	encrypted
nvs_key,	data,	nvs_keys,	0x320000,	0x1000,	encrypted

** I have already searched for similar issues with no luck. I have posted this issue on esp32.com forum but nobody replied there...

Could you please help me?

igrr, 2022-12-02:
Could you please attach your application .map file and your sdkconfig file?

> I am trying to updrage my project from esp-idf v4.0.1 to 4.1 with the purpose then to upgrade to the last stable release.

Not sure if this was a typo, but please note that v4.1 is no the latest stable release. v4.1 release branch is in maintenance period now and will be EoL next year. Please check the support period graph here: https://github.com/espressif/esp-idf#esp-idf-release-support-schedule.

afcec, 2022-12-02:
Yes, upgrading to 4.1 is a first step with the final purpose of reaching the last stable release step by step.

Please, finf the required files attached. Thanks for your help.

[map and sdkconfig.zip](https://github.com/espressif/esp-idf/files/10139748/map.and.sdkconfig.zip)

afcec, 2022-12-02:
Just one more comment: this last build was done using ESP-IDF v4.1-1-ge4f6ea73da after applying a bug fix suggested for a user having a similar issue, but the behavior was just the same with v4.1

igrr, 2022-12-02:
One more question, is that all the log output you get? I.e. is
```
D (2372) memory_layout: Available memory region 0x3ffbda58 - 0x3ffc0000
```
the last line printed? What happens after that, is there a watchdog reset?

afcec, 2022-12-02:
Yes, continuous watchdog reset. Find below all the lines:

cpu_start: Pro cpu up.
I (543) cpu_start: Application information:
I (543) cpu_start: Project name:     Conn_Board
I (573) cpu_start: App version:      2.2
I (629) cpu_start: Compile time:     Dec  2 2022 09:36:29
I (702) cpu_start: ELF file SHA256:  4ea564ebf1bc4a74...
I (773) cpu_start: ESP-IDF:          v4.1-1-ge4f6ea73da
I (844) cpu_start: Starting app cpu, entry point is 0x400813c0
I (0) cpu_start: App cpu up.
V (965) memory_layout: reserved range is 0x3f4429f8 - 0x3f442a20
D (1038) memory_layout: Checking 7 reserved memory ranges:
D (1105) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (1183) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffbda58
D (1261) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (1339) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (1417) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (1495) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (1573) memory_layout: Reserved memory range 0x40080000 - 0x400968a8
D (1651) memory_layout: Building list of available memory regions:
V (1727) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (1807) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (1922) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
V (2003) memory_layout: Examining memory region 0x3ffb0000 - 0x3ffb8000
V (2083) memory_layout: Region 0x3ffb0000 - 0x3ffb8000 inside of reserved 0x3ffb0000 - 0x3ffbda58
V (2190) memory_layout: Examining memory region 0x3ffb8000 - 0x3ffc0000
V (2270) memory_layout: Start of region 0x3ffb8000 - 0x3ffc0000 overlaps reserved 0x3ffb0000 - 0x3ffbda58
D (2386) memory_layout: Available memory region 0x3ffbda58 - 0x3ffc0000
V (2466) memory_layout: Examining memory region 0x3ffc0000 - 0x3ffc2000
D (2546) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
V (2626) memory_layout: Examining memory region 0x3ffc2000 - 0x3ffc4000
D (2707) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
V (2787) memory_layout: Examining memory region 0x3ffc4000 - 0x3ffc6000
D (2867) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
V (2947) memory_layout: Examining memory region 0x3ffc6000 - 0x3ffc8000
D (3028) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
V (3108) memory_layout: Examining memory region 0x3ffc8000 - 0x3ffca000
D (3188) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
V (3268) memory_layout: Examining memory region 0x3ffca000 - 0x3ffcc000
D (3348) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
V (3429) memory_layout: Examining memory region 0x3ffcc000 - 0x3ffce000
D (3509) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
V (3589) memory_layout: Examining memory region 0x3ffce000 - 0x3ffd0000
D (3669) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
V (3749) memory_layout: Examining memory region 0x3ffd0000 - 0x3ffd2000
D (3830) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
V (3910) memory_layout: Examining memory region 0x3ffd2000 - 0x3ffd4000
D (3990) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
V (4070) memory_layout: Examining memory region 0x3ffd4000 - 0x3ffd6000
D (4150) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
V (4231) memory_layout: Examining memory region 0x3ffd6000 - 0x3ffd8000
D (4311) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
V (4391) memory_layout: Examining memory region 0x3ffd8000 - 0x3ffda000
D (4471) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
V (4551) memory_layout: Examining memory region 0x3ffda000 - 0x3ffdc000
D (4632) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
V (4712) memory_layout: Examining memory region 0x3ffdc000 - 0x3ffde000
D (4792) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
V (4872) memory_layout: Examining memory region 0x3ffde000 - 0x3ffe0000
D (4953) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
V (5033) memory_layout: Examining memory region 0x3ffe0000 - 0x3ffe4000
V (5113) memory_layout: Start of region 0x3ffe0000 - 0x3ffe4000 overlaps reserved 0x3ffe0000 - 0x3ffe0440
V (5229) memory_layout: End of region 0x3ffe0440 - 0x3ffe4000 overlaps reserved 0x3ffe3f20 - 0x3ffe4350
D (5342) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
V (5422) memory_layout: Examining memory region 0x3ffe4000 - 0x3ffe8000
V (5503) memory_layout: Start of region 0x3ffe4000 - 0x3ffe8000 overlaps reserved 0x3ffe3f20 - 0x3ffe4350
D (5618) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
V (5698) memory_layout: Examining memory region 0x3ffe8000 - 0x3fff0000
D (5779) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
[... 3 lines cut ...]
D (6099) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
V (6180) memory_layout: Examining memory region 0x3fffc000 - 0x40000000
D (6260) memory_layout: Available memory region 0x3fffc000 - 0x40000000
V (6340) memory_layout: Examining memory region 0x40070000 - 0x40078000
V (6420) memory_layout: Region 0x40070000 - 0x40078000 inside of reserved 0x40070000 - 0x40078000
V (6528) memory_layout: Examining memory region 0x40078000 - 0x40080000
V (6608) memory_layout: Region 0x40078000 - 0x40080000 inside of reserved 0x40078000 - 0x40080000
[... cut ...]
V (7840) memory_layout: Examining memory region 0x4008c000 - 0x4008e000
V (7920) memory_layout: Region 0x4008c000 - 0x4008e000 inside of reserved 0x40080000 - 0x400968a8
V (8028) memory_layout: Examining memory region 0x4008e000 - 0x40090000
V (8108) memory_layout: Region 0x4008e000 - 0x40090000 inside of reserved 0x40080000 - 0x400968a8
V (8215) memory_layout: Examining memory region 0x40090000 - 0x40092000
V (8295) memory_layout: Region 0x40090000 - 0x40092000 inside of reserved 0x40080000 - 0x400968a8

igrr, 2022-12-02:
The last lines in this log (from `memory_layout`) don't indicate any problem, as far as I can tell.

Are you printing or logging anything from app_main? Could you compare the log output between the working and non-working version and say what is the first log line that is missing?

Another thing, you have the following options enabled in sdkconfig:
```
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
```

Are you calling `rtc_wdt_feed` or `rtc_wdt_disable` anywhere in the application code to disable the watchdog?

afcec, 2022-12-02:
> The last lines in this log (from `memory_layout`) don't indicate any problem, as far as I can tell.
>
> Are you printing or logging anything from app_main? Could you compare the log output between the [...]

No, I am not

igrr, 2022-12-02:
Thanks for checking this!

Could you please try to increase CONFIG_BOOTLOADER_WDT_TIME_MS to a larger value?
What seems to happen right now is that the start-up flow is interrupted approximately after 9 seconds, which incidentally is the RTC watchdog timeout configured by the bootloader. (One possible reason for such slow startup could be if you are using a slow console baud rate, such as 9600.)

Alternatively, you can temporarily disable CONFIG_BOOTLOADER_WDT_ENABLE option, just to check if the boot process can proceed further than it does now.

afcec, 2022-12-05:
Hi, increasing CONFIG_BOOTLOADER_WDT_TIME_MS to 12 seconds worked. 

Now I am facing different errors due to the changes made to move from tcpip_adater to netif, but the application starts correctly.

Thanks!

closed 2022-12-06 by igrr