espressif/esp-idf #10298: esp_task_wdt_reset called when the task list is empty

norventa, 2022-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 rc1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Not throw multiple wobblys.

### What is the actual behavior?

esp_task_wdt_reset in task_wdt.c is called repeatedly (correct). Many times after calling find_entry_from_task_handle_and_check_all_reset entry is NULL. handle is not NULL (how?). There are 5 tasks shown whilst debuging in the CALL STACK panel.

`esp_err_t esp_task_wdt_reset(void)
{
    ESP_RETURN_ON_FALSE(p_twdt_obj != NULL, ESP_ERR_INVALID_STATE, TAG, "TWDT was never initialized");
    esp_err_t ret;
    TaskHandle_t handle = xTaskGetCurrentTaskHandle();
    portENTER_CRITICAL(&spinlock);
    // Find entry from task handle
    bool all_reset;
    twdt_entry_t *entry;
    entry = find_entry_from_task_handle_and_check_all_reset(handle, &all_reset);
    ESP_GOTO_ON_FALSE_ISR((entry != NULL), ESP_ERR_NOT_FOUND, err, TAG, "task not found");
    // Mark entry as reset and issue timer reset if all entries have been reset
    entry->has_reset = true;    // Reset the task if it's on the task list
    if (all_reset) {    // Reset if all other tasks in list have reset in
        task_wdt_timer_feed();
    }
    ret = ESP_OK;
err:
    portEXIT_CRITICAL(&spinlock);
    return ret;
}`

### Steps to reproduce.

Simply by running my application. After initialization of various classes a SPIFFs partition is read. It is then that the following error message repeats:

E (xx) task_wdt: esp_task_wdt_reset(793): task not found

### Debug Logs.

```plain
Windows PowerShell
[... 272 lines cut ...]
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

norventa, 2022-12-03:
The watchdog code has changed and this was the cause of the problem. By calling esp_task_wdt_delete with a task handle something strange happens in that (I believe) the list of handles gets updated by zeroing the task list array item rather than getting removed from the array.

closed 2022-12-03 by norventa

CraigeHales, 2026-01-07:
Similar issue, using a highest-priority hard loop on cpu 1, using printf, with watchdog disabled on cpu1. Many different weird behaviors. Same _find_entry_from_task_handle_and_check_all_reset_/_task-not-found_ and some other issues, eventually including _out-of-stack-space_ as I tried different ideas.
Works great with more stack space requested when launching the task. 
`xTaskCreatePinnedToCore(cpu_burn_task, "cpu_burn_task", 1024 * 10 /* stack, 2K is too small */, 
                 NULL /*parm*/, configMAX_PRIORITIES - 1, NULL /*taskhandle*/, 1 /*core*/);`