espressif/esp-idf #10299: WiFi promiscuous timestamp differs by 140+ms

PavelDrabek, 2022-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Arduino ESP32 version 2.0.5 based on the ESP-IDF v4.4.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

DOIT ESP32 DEVKIT V1

### Power Supply used.

USB

### What is the expected behavior?

I expect timestamps differs by 10 microseconds at most.

### What is the actual behavior?

Timestamps differs by more than 140ms (140000 microseconds). 

Timestamp stored in wifi_promiscuous_pkt_t->wifi_pkt_rx_ctrl_t received in wifi promiscuous rx callback function differs from micros() called immediately in that function by more than 140,000 microseconds but code is running smoothly. 

### Steps to reproduce.

Run the following code: 

```
#include <WiFi.h>
#include <esp_wifi.h>

void promiscuous_rx_cb(void *buf, wifi_promiscuous_pkt_type_t type) {
  unsigned long t_cb = micros(); // local time when callback is called

  const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buf;
  unsigned int t_rx = ppkt->rx_ctrl.timestamp; // local time when this packet is received

  String type_str;
  switch(type){
    case WIFI_PKT_MGMT: type_str = "WIFI_PKT_MGMT"; break;
    case WIFI_PKT_CTRL: type_str = "WIFI_PKT_CTRL"; break;
    case WIFI_PKT_DATA: type_str = "WIFI_PKT_DATA"; break;
    case WIFI_PKT_MISC: type_str = "WIFI_PKT_MISC"; break;
  }

  Serial.printf("rx_cb is called: %s\t t_rx:%u t_cb: %lu, t_diff: %lu\n", type_str, t_rx, t_cb, (t_cb - t_rx));
}

void setup() 
{
  unsigned long t1 = micros();
  Serial.begin(115200);
  unsigned long t2 = micros();

  wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
  unsigned long t3 = micros();

  esp_wifi_init(&wifi_init_config);
  unsigned long t4 = micros();

  WiFi.mode(WIFI_STA);
  unsigned long t5 = micros();

  esp_wifi_set_promiscuous(true);
  unsigned long t6 = micros();

  esp_wifi_set_promiscuous_rx_cb(&promiscuous_rx_cb);
  unsigned long t7 = micros();

  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
  unsigned long t8 = micros();

  Serial.printf("t1: %lu, d: %lu\n", t1, t1);
  Serial.printf("t2: %lu, d: %lu\n", t2, t2-t1);
  Serial.printf("t3: %lu, d: %lu\n", t3, t3-t2);
  Serial.printf("t4: %lu, d: %lu\n", t4, t4-t3);
  Serial.printf("t5: %lu, d: %lu\n", t5, t5-t4);
  Serial.printf("t6: %lu, d: %lu\n", t6, t6-t5);
  Serial.printf("t7: %lu, d: %lu\n", t7, t7-t6);
  Serial.printf("t8: %lu, d: %lu\n", t8, t8-t7);
}
 
void loop() {

}
```

### Debug Logs.

```plain
18:55:49.198 -> t1: 33205, d: 33205
18:55:49.198 -> t2: 34503, d: 1298
18:55:49.198 -> t3: 34540, d: 37
18:55:49.198 -> t4: 52917, d: 18377
18:55:49.198 -> t5: 148575, d: 95658
18:55:49.198 -> t6: 148874, d: 299
18:55:49.198 -> t7: 148918, d: 44
18:55:49.198 -> t8: 149267, d: 349
18:55:50.681 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:1486147 t_cb: 1632695, t_diff: 146548
18:55:51.677 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:2483852 t_cb: 2630371, t_diff: 146519
18:55:52.645 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:3485173 t_cb: 3631713, t_diff: 146540
18:55:53.685 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:4484143 t_cb: 4630684, t_diff: 146541
18:55:54.647 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:5482801 t_cb: 5629336, t_diff: 146535
18:55:55.674 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:6486301 t_cb: 6632820, t_diff: 146519
18:55:56.665 -> rx_cb is called: WIFI_PKT_MGMT	 t_rx:7483587 t_cb: 7630110, t_diff: 146523
```

### More Information.

I am working with Arduino-ESP 2.0.5 and create the issue there but was redirected here because it's ESP-IDF function: https://github.com/espressif/arduino-esp32/issues/7540

based on this issue: https://github.com/espressif/esp-idf/issues/2468
I already disabled sleep mode 

based on this issue: https://github.com/espressif/esp-idf/issues/2706#issuecomment-441407932
`timer is started when WiFi clock is enabled. When esp_wifi_start()/esp_wifi_set_promiscuous() is returned, the timer is started.` 
I stored timestamp when this method is called and it also mismatches by 2 miliseconds (148575 vs 146548) so I can't even create a correction variable.

pkese, 2023-12-01:
Did you try to disable wifi power saving?

I'm getting better results if I call `esp_wifi_set_ps(WIFI_PS_NONE)` immediately after initializing wifi interface.

```c
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
```

zhangyanjiaoesp, 2024-04-08:
@PavelDrabek The `timestamp` in the `rx_ctrl` represents the time the packet was received by the hardware, the timer is started when `esp_wifi_start()` is returned. And the `micros()` will call `esp_timer_get_time()`, the esp timer is started when the system starts. These two timers have different time bases.

Sherry616, 2024-06-07:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product!

closed 2024-06-07 by Sherry616

emaayan, 2024-07-30:
> C

so if i wanted to capture the actual date and time (to be displayed in wireshar) i would need to constantly call gettimeofday 
like so, would that be problematic performance wise?
```
int r = gettimeofday(&p, NULL);
  if (r == 0) {     
  	pcap_rec.pcap_rec_hdr.ts_sec=p.tv_sec;
  	pcap_rec.pcap_rec_hdr.ts_usec=p.tv_usec;
  }
  if (_sock && pcap_rec.pcap_rec_hdr.orig_len > 0) {
    if (!onSend(_sock, msg, total_size)) {
      _sock = 0;
      sniffer_stop();
    }
  }
  sniffer_packet_t sniffer_packet = sniffer_to_packet_data(pcap_rec);
```