espressif/esp-idf #10300: CMD and PS shells complain of an error if installtion intentionally omoitted support for even one ESP32 target

JBHunterI, 2022-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Using Python in %IDF_TOOLS_PATH%\\python_env\\idf5.0_py3.8_env\\Scripts\\
Python 3.8.7
Using Git in %IDF_TOOLS_PATH%\\tools\\idf-git\\2.34.2\\cmd\\
git version 2.34.1.windows.1
Checking Python compatibility
Setting IDF_PATH: %IDF_TOOLS_PATH%\\esp-idf-v5.0

Adding ESP-IDF tools to PATH...
    %IDF_TOOLS_PATH%\\tools\\riscv32-esp-elf-gdb\\11.2_20220823\\riscv32-esp-elf-gdb\\bin
    %IDF_TOOLS_PATH%\\tools\\riscv32-esp-elf\\esp-2022r1-11.2.0\\riscv32-esp-elf\\bin
    %IDF_TOOLS_PATH%\\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    %IDF_TOOLS_PATH%\\tools\cmake\3.24.0\bin
    %IDF_TOOLS_PATH%\\tools\\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin
    %IDF_TOOLS_PATH%\\tools\\ninja\1.10.2\
    %IDF_TOOLS_PATH%\\tools\\idf-exe\1.0.3\
    %IDF_TOOLS_PATH%\\tools\\ccache\4.6.2\ccache-4.6.2-windows-x86_64
    %IDF_TOOLS_PATH%\\tools\\tools\dfu-util\0.9\dfu-util-0.9-win64
    %IDF_TOOLS_PATH%\\esp-idf-v5.0\tools

Checking if Python packages are up to date...
Constraint file: %IDF_TOOLS_PATH%\\espidf.constraints.v5.0.txt
Requirement files:
 - %IDF_TOOLS_PATH%\\tools\requirements\\requirements.core.txt
Python being checked: %IDF_TOOLS_PATH%\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing dfu-util, esp32ulp-elf, riscv32-esp-elf, riscv32-esp-elf-gdb, use command 'python.exe %IDF_TOOLS_PATH%\\esp-idf-v5.0\\tools\\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe %IDF_TOOLS_PATH%\\esp-idf-v5.0\\tools\\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

%IDF_TOOLS_PATH%\\esp-idf-v5.0> git describe --tag
v5.0

%IDF_TOOLS_PATH%\\esp-idf-v5.0> 

### What is the actual behavior?

Using Python in %IDF_TOOLS_PATH%\\python_env\\idf5.0_py3.8_env\\Scripts\\
Python 3.8.7
Using Git in %IDF_TOOLS_PATH%\\tools\\idf-git\\2.34.2\\cmd\\
git version 2.34.1.windows.1
Checking Python compatibility
Setting IDF_PATH: %IDF_TOOLS_PATH%\\esp-idf-v5.0

Adding ESP-IDF tools to PATH...
ERROR: tool xtensa-esp-elf-gdb has no installed versions. Please run '%IDF_TOOLS_PATH%\\esp-idf-v5.0\\install.bat' to install it.
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '%IDF_TOOLS_PATH%\\esp-idf-v5.0\\install.bat' to install it.
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '%IDF_TOOLS_PATH%\\esp-idf-v5.0\\install.bat' to install it.
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '%USERPROFILE%\\esp-idf-v5.0\\install.bat' to install it.
ERROR: tool esp32ulp-elf has no installed versions. Please run '%IDF_TOOLS_PATH%\\esp-idf-v5.0\\install.bat' to install it.
ERROR: tool dfu-util has no installed versions. Please run '%IDF_TOOLS_PATH%\\esp-idf-v5.0\\install.bat' to install it.

%IDF_TOOLS_PATH%\\esp-idf-v5.0> git describe --tag
v5.0

%IDF_TOOLS_PATH%\\esp-idf-v5.0> 

### Steps to reproduce.

1. Install ESP-IDF v5.0 using esp-idf-tools-setup-online-2.16.exe on Windows 10 without including support for every single ESP32 target, regardless if you're ever going to develop for them. 
3. Look at the ERRORs appear when ever a CMD and PS shell is opened, including at the end of the ESP-IDF installation.

### More Information.

It shouldn't be flagged as an error if support for one or more target(s) was intentionally selectively omitted during installation.

dobairoland, 2022-12-05:
Hi @JBHunterI. Could you please try to delete `idf-env.json` from `%IDF_TOOLS_PATH` and re-run the installer? Does this fix your issue?

Otherwise please inspect your environment variables and clean everything (ESP-IDF-related) from there before running the installer.

JBHunterI, 2022-12-06:
No, deleting %IDF_TOOLS_PATH\idf-env.json & re-running the universal online installer on a new PC that has never had the ESP-IDF installed on it (so there were no ESP* or IDF* variables in the environment prior to this installation attempt), pointing it to the existing %IDF_PATH% to avoid downloading & extracting that again, & selecting the exact same installation options again (only interested in the ESP32-C3 tools) does not get rid of the error messages after the "Adding ESP-IDF tools to PATH..." message in the new CMD & PS shells. Both still complain about xtensa-esp*, esp32ulp-elf, & dfu-util files not being installed.  I don't understand how you can't reproduce this with the v5.0 latest stable release.

dobairoland, 2022-12-06:
What version of ESP-IDF are you using? The `release/v5.0` of this Github repo? Or the tagged v5.0 version? Please make sure that it is not one of the betas or earlier. Because I think this issue has already been fixed earlier.

BTW, a new version of the installer (2.17) has just been released at https://dl.espressif.com/dl/esp-idf/. This is the first release since the final version of v5.0 of ESP-IDF has been released.

JBHunterI, 2022-12-06:
As the 'v5.0' output of my 'git describe --tag' command in the header of this issue shows, I was using the pre-release of v5.0 and updated to v5.0-rc1 before, but I waited for the official release of v5.0 to submit this issue. I don't know if it's the Github repo or the tagged version, but it's whatever the Universal Installer v2.16 installs when I select the very first/top-most version offered:
[image]

Are you suggesting this problem will disappear if I install ESP-IDF v5.0 (release version) using the Universal Installer v2.17?  BTW, what' the heck is the difference between "v5.0 (release version)" and "release/v5.0 (release branch)" five lines below, & which one should I select?  Why does this have to be so confusing? Can you not reproduce this issue with v2.16. or v2.17 already?

JBHunterI, 2022-12-06:
Thanks, @igrr. I've read that already, but my point was that the Universal Online Installer doesn't state it but, now from my experience, it would download & extract the "v5.0 (release version)" at the top of its list from an archived/zipped copy of the official Release v5.0, & I'm guessing "release/v5.0 (release branch)" near the bottom of its list would get a copy of the same thing but from a tag from the GitHub repo. All that is just not clear, at least to me, from the installer's list. Let me know if that's incorrect.

igrr, 2022-12-06:
I think the end result when installing these two versions would be:
- If you choose "v5.0 (release version)", you will get a copy of IDF at v5.0 tag. Indeed, it practically gets downloaded as an archived copy of the repo, with submodules included.
- If you choose "v5.0 (release branch)", you will get a copy of latest IDF release/5.0 branch. Since v5.0 release, there have been more commits to release/v5.0 branch, so that branch has more features and fixes than the v5.0 release tag. Eventually, there will be a v5.0.1 bugfix release on that branch. Then the installer will start offering "v5.0.1 (release version)" instead of "v5.0 (release version)" as the first choice. The point is, the release branch most of the time is ahead of the release tag. So it is more recent and has more features (+) but unlike the tags, it hasn't been subject to release testing by our QA (-).

Practically, when you choose the "release branch" option, the installer will download the archive with the latest tag on that branch (v5.0 in this case) and then update it to the tip of the release branch. This is purely a download speed optimization, though; the end result is the same as if you would do `git clone --branch release/v5.0 --recursive https://github.com/espressif/esp-idf.git`.

JBHunterI, 2022-12-06:
Thanks for clarifying that, @igrr. Maybe if the installer used the word "stable" in its list, as does the list at the top-left of the online ESP-IDF Programming Guide, it might make it clearer that the we're talking about a fixed point in the product's release (like an archive or a specific 'stable' tag), whereas a branch implies the possibility of getting something that might be changing with time.

JBHunterI, 2022-12-06:
FYI, getting back to my original question for this thread, I tried using the Universal Installer v2.17, after uninstalling the "v5.0 (release version)" installed with the Universal Installer v2.16 & manually getting rid of any remaining files/folders/environment variables left behind. Here is what I entered as the installtion configuration:
[image]
I also compared the %IDF_TOOLS_PATH%\releases\esp-idf-v5.0.zip from both installer versions: they were identical. 
Not sure if dfu-util is needed by the ESP32-C3, but do we agree that an installation configuration that excludes support for some ESP32 targets should not cause errors?

dobairoland, 2022-12-07:
> but do we agree that an installation configuration that excludes support for some ESP32 targets should not cause errors?

Nobody denies that. I was collecting more information so we could properly understand the issue and reproduce it.

mfialaf, 2022-12-20:
Hi @JBHunterI 
I am sorry for having this issue.
The fix for now is as follows:

1) Open the ESP-IDF 5.0 CMD
2) You'll see the same error as before. Copy the command the ERROR tells you. Running this command would install all the tools for all esp targets. Since you want to use only esp32c3, place `esp32c3` at the end of the command. 

Your fixing command according to the screenshots you provided should look like this: `C:\App\ESP\IDF\esp-idf-v5.0\install.bat esp32c3`

It seems that we have a problem with the Universal Installer, when parsing the user's choice of targets to the configuration file.
We will fix it in the next release.

Please try this, and let me know if it works for you.

georgik, 2022-12-20:
Problem is caused by idf_tools.py which is not flipping the slashes in the `--idf-path` parameter passed from the installer. The value is then stored in the JSON in incorrect format.

After discussion with @mfialaf we switched slashes to backslashes in the installer: https://github.com/espressif/idf-installer/releases/tag/online-2.18-beta-01

dobairoland, 2022-12-20:
Thank you @georgik and @mfialaf for your help!

georgik, 2022-12-20:
Meanwhile we discovered some other issue. Released new build https://github.com/espressif/idf-installer/releases/tag/online-2.18-beta-03
@mfialaf confirmed that it works well.

JBHunterI, 2022-12-20:
I had to go back to ESP-IDF v4.4.3 because the [Featured FreeRTOS IoT Integration on ESP32-C3 RISC-V MCU](https://github.com/FreeRTOS/iot-reference-esp32c3/#featured-freertos-iot-integration-on-esp32-c3-risc-v-mcu) won't seem to support ESP-IDF v5.0 for a while (see [this response in its issue #16](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/16#issuecomment-1359579124)) but I installed v4.4.3 using your Online Installer v2.18-beta3 mentioned above and it seemed to not complain when it only installed the ESP32-C3 RISC-V tools I requested. My system path only contains entries for that, 'where git python' reports the expected 3 paths, 'python --version', 'git --version', and 'git describe --tag' correctly responds with "v4.4.3".

dobairoland, 2023-01-02:
If I read the last comments correctly then this has been resolved the the new version of the installer.

closed 2023-01-02 by dobairoland