espressif/esp-idf #10301: ulp_riscv_i2c crash

Veldmonster, 2022-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-2061-g7869f4e151

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Olimex ESP32-S2-DevKit-Lipo

### Power Supply used.

USB

### What is the expected behavior?

The ESP should remain in deep sleep while polling the BMP180 every 20ms when either temperature or pressure exceeds the thresholds defined in bmp180_defs.h. The ESP should wakeup and print before going back to sleep.

### What is the actual behavior?

The ESP32-S2 enters deep sleep and then wakeup again. With message "Not a ULP-RISC V wakeup (cause = 11)". Thereafter it just hangs until physical reset.

### Steps to reproduce.

Step1: Use example, esp-idf-master\examples\system\ulp_riscv\i2c with bmp180 i2c sensor connected
Step2: Configure esp-idf for an esp32s2
**Step3**: Edit bmp180_defs.h set the value of EXAMPLE_UT_THRESHOLD and EXAMPLE_UP_THRESHOLD to a higher value such that it will not trigger an instant wakeup.
Step4: Build, flash and monitor

### Debug Logs.

```plain
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1784
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.1-dev-2061-g7869f4e151 2nd stage bootloader
I (21) boot: compile time Dec  3 2022 12:06:29
I (23) boot: chip revision: v0.0
I (27) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=09574h ( 38260) map
I (100) esp_image: segment 1: paddr=0001959c vaddr=3ff9e02c size=00004h (     4) load
I (101) esp_image: segment 2: paddr=000195a8 vaddr=3ffbe040 size=01590h (  5520) load
I (110) esp_image: segment 3: paddr=0001ab40 vaddr=40022000 size=054d8h ( 21720) load
I (123) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=14e34h ( 85556) map
I (143) esp_image: segment 5: paddr=00034e5c vaddr=400274d8 size=06b5ch ( 27484) load
I (150) esp_image: segment 6: paddr=0003b9c0 vaddr=40070000 size=0002ch (    44) load
I (156) boot: Loaded app from partition at offset 0x10000
I (157) boot: Disabling RNG early entropy source...
I (172) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (172) cpu_start: Pro cpu up.
I (192) cpu_start: Pro cpu start user code
I (192) cpu_start: cpu freq: 160000000 Hz
I (192) cpu_start: Application information:
I (195) cpu_start: Project name:     ulp-riscv-rtc-i2c-example
I (202) cpu_start: App version:      1
I (206) cpu_start: Compile time:     Dec  3 2022 12:05:52
I (212) cpu_start: ELF file SHA256:  674e393a2eb947d2...
I (218) cpu_start: ESP-IDF:          v5.1-dev-2061-g7869f4e151
I (225) cpu_start: Min chip rev:     v0.0
I (230) cpu_start: Max chip rev:     v1.99 
I (234) cpu_start: Chip rev:         v0.0
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FFBFE58 len 0003C1A8 (240 KiB): DRAM
I (253) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (259) heap_init: At 3FF9E030 len 00001FD0 (7 KiB): RTCRAM
I (266) spi_flash: detected chip: generic
I (270) spi_flash: flash io: dio
I (274) cpu_start: Starting scheduler on PRO CPU.
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
Reading calibration data from BMP180 ...
ac1 = 7854
ac2 = -1181
ac3 = -14525
ac4 = 33974
ac5 = 24866
ac6 = 15474
b1 = 6515
b2 = 47
mb = -32768
mc = -11786
md = 2901

Reading initial uncompensated temperature and pressure data ...
Uncompensated Temperature = 23241
Uncompensated Pressure = 42990

Real Temperature = 19.700001 deg celcius
Real Pressure = 1018.480000 hPa

Entering deep sleep

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x5 (DSLEEP),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1784
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.1-dev-2061-g7869f4e151 2nd stage bootloader
I (21) boot: compile time Dec  3 2022 12:06:29
I (23) boot: chip revision: v0.0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=09574h ( 38260) map
I (100) esp_image: segment 1: paddr=0001959c vaddr=3ff9e02c size=00004h (     4) 
I (100) esp_image: segment 2: paddr=000195a8 vaddr=3ffbe040 size=01590h (  5520) load
I (110) esp_image: segment 3: paddr=0001ab40 vaddr=40022000 size=054d8h ( 21720) load
I (123) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=14e34h ( 85556) map
I (143) esp_image: segment 5: paddr=00034e5c vaddr=400274d8 size=06b5ch ( 27484) load
I (150) esp_image: segment 6: paddr=0003b9c0 vaddr=40070000 size=0002ch (    44) 
I (156) boot: Loaded app from partition at offset 0x10000
I (156) boot: Disabling RNG early entropy source...
I (171) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (171) cpu_start: Pro cpu up.
I (183) cpu_start: Pro cpu start user code
I (183) cpu_start: cpu freq: 160000000 Hz
I (183) cpu_start: Application information:
I (188) cpu_start: Project name:     ulp-riscv-rtc-i2c-example
I (194) cpu_start: App version:      1
I (199) cpu_start: Compile time:     Dec  3 2022 12:05:52
I (205) cpu_start: ELF file SHA256:  674e393a2eb947d2...
I (211) cpu_start: ESP-IDF:          v5.1-dev-2061-g7869f4e151
I (217) cpu_start: Min chip rev:     v0.0
I (222) cpu_start: Max chip rev:     v1.99 
I (227) cpu_start: Chip rev:         v0.0
I (231) heap_init: Initializing. RAM available for dynamic allocation:
I (239) heap_init: At 3FFBFE58 len 0003C1A8 (240 KiB): DRAM
I (245) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (251) heap_init: At 3FF9E030 len 00001FD0 (7 KiB): RTCRAM
I (258) spi_flash: detected chip: generic
I (262) spi_flash: flash io: dio
I (266) cpu_start: Starting scheduler on PRO CPU.
Not a ULP-RISC V wakeup (cause = 11)
Initializing RTC I2C ...
```

### More Information.

The same problem is exhibited with another i2c device (vl53l1x) irrespective of using an ESP32-S2 or ESP32-S3.

schafon, 2022-12-06:
Having the exact same issue, esp32-s2-saola-1-v1.2

sudeep-mohanty, 2022-12-06:
Hi @schafon,
Thanks for reporting this. I have been able to reproduce the issue at my end as well and am looking into it. I will keep the post updated once I have some findings.

schafon, 2022-12-06:
> Hi @schafon,
> Thanks for reporting this. I have been able to reproduce the issue at my end as well and am looking into it. I will keep the post updated once I have some findings.

Hi there, I'm a complete noob with C, I usually code in Arduino so I'm way off my comfort zone.
I think this issue is related to this: https://github.com/espressif/esp-idf/issues/10164
I'm pretty sure that the crash coming from ulp_riscv_delay_cycles()

Thank you for looking into it! I'm looking forward to use this code!

Veldmonster, 2022-12-06:
> Hi there, I'm a complete noob with C, I usually code in Arduino so I'm way off my comfort zone. I think this issue is related to this: #10164 I'm pretty sure that the crash coming from [...]

I don't think it is related (at least not directly). I have tested it by placing ulp_riscv_delay_cycles() in a loop. When a count is reached, it wakes main cpu and it runs fine. In an infinite loop, the cocpu stays asleep. That said, I have had trouble with the internal temperature sensor using the main cpu in the past. It uses an infinite loop to scan for a condition (scary).

Which brings me to another issue, I see ulp_riscv_i2c.c also uses infinite loops to scan for the tx/rx interrupt. If that interrupt never arrives unfortunate things happen. For my purpose, I have modified ulp_riscv_i2c.c to at least make use the ack error and timeout flags provided by RTC_I2C_INT_ST_REG but, even that does not fix the reported issue.

The closest, possibly related, issue I could find was: #9455 
But, that is for a C3 using another clock (which does not work with ulp_riscv_i2c either). Also, rtc_sleep.c is implemented differently for the S2/3.

schafon, 2022-12-19:
Hi guys, I know it's been only two weeks and that you probably have other issues with higher priority but I'm developing a product that I want to sell and this is the only thing standing in my way.
Is there an ETA on this issue? 
Thanks!

sudeep-mohanty, 2022-12-19:
Hi @schafon,
Thanks for reporting the issue. And apologies for the late reply. We've been debugging the issue and have narrowed it down to the reset sequence of  the ULP core which seems to be causing this crash that is observed. There is an internal MR which is currently being reviewed and will make it into the repository as soon as it is approved. While I can't give a definite timeline for the same, I estimate that it should be available soon (in the coming week or two). The fix will also be backported to our stable release branches.
However, to unblock your development process, please try this change locally and do let us know if it helps. Here is the patch file attached - 

[ulp_riscv_i2c_crash.patch](https://github.com/espressif/esp-idf/files/10256487/ulp_riscv_i2c_crash.patch)

schafon, 2022-12-19:
> 

Thank you so much!! It works!
I tested the temp with heat gun I have at my desk and I think it goes crazy above certain temp, the int16_t is not enough.
But I don't really care because you saved me so much trouble!
Keep on the amazing job!

Veldmonster, 2022-12-23:
Thank you @sudeep-mohanty. I too can confirm that the patch works.

closed 2022-12-23 by Veldmonster

sudeep-mohanty, 2022-12-23:
Thank you for confirming @Veldmonster !