espressif/esp-idf #10305: esp_bt_hid_device_connect() returns ESP_OK even if there was not possibility to connect to the BD

artik15, 2022-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi!
I am trying to make my project (BT CLASSIC HIDD keyboard) connect to the last connected/paired device. To do so I am saving its BD address in NVS and reading it at proper moment. I have created a task function that gets the address and tries to connect defined amount of times (currently 3 times) to the device. I am using `esp_bt_hid_device_connect()` and checking if it returns ESP_OK

```
void reconnect_hidd(void* PvParameters){
    (void) PvParameters;
    char *TAG = "Reconnect_HIDD";
    esp_bd_addr_t address = {0};
    esp_err_t error;
    error = get_saved_bt_devive(&address);
    if(error != ESP_OK) {
        ESP_LOGW(TAG, "Error getting saved bt device address: %s", esp_err_to_name(error));
    } else {
        /*Try to reconnect 3 times with 5sec break*/
        ESP_LOGI(TAG, "Trying to connect to %02x:%02x:%02x:%02x:%02x:%02x", address[0], address[1],
                        address[2], address[3], address[4], address[5]);
        for(uint8_t i = 0; i < MAX_RECONNECT_ATTEMPTS; i++){
            error = esp_bt_hid_device_connect(address);
            if(error == ESP_OK) {
                ESP_LOGI(TAG, "Connected to last remembered device!");
                break;
            }
            ESP_LOGW(TAG, "Error connecting to the BT device: %s", esp_err_to_name(error));
            vTaskDelay(pdTICKS_TO_MS(5000));
        }
    }
    vTaskDelete(NULL);
}
```

The problem is that `esp_bt_hid_device_connect()` returns ESP_OK even if the device is not available (Turned off BT on my laptop). While getting the BD address saving process right, I had a wrong address in the NVS partition (random bytes) and even then the API function returned ESP_OK. Is there any way to try to connect to the device and get information if it there was actually a connection established?

**Logs:**
```
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 21:45:10
I (27) boot: chip revision: 
I (31) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4d470h (316528) map
I (210) esp_image: segment 1: paddr=0005d498 vaddr=3ffbdb60 size=02b80h ( 11136) load
I (215) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=90558h (591192) map
I (430) esp_image: segment 3: paddr=000f0580 vaddr=3ffc06e0 size=01c0ch (  7180) load
I (433) esp_image: segment 4: paddr=000f2194 vaddr=40080000 size=19e00h (105984) load
I (479) esp_image: segment 5: paddr=0010bf9c vaddr=50000000 size=00010h (    16) load
I (492) boot: Loaded app from partition at offset 0x10000
I (492) boot: Disabling RNG early entropy source...
I (504) psram: This chip is ESP32-D0WD
I (505) spiram: Found 64MBit SPI RAM device
I (505) spiram: SPI RAM mode: flash 40m sram 40m
I (509) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (517) cpu_start: Pro cpu up.
I (520) cpu_start: Starting app cpu, entry point is 0x40081430
0x40081430: call_start_cpu1 at D:/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (1414) spiram: SPI SRAM memory test OK
I (1422) cpu_start: Pro cpu start user code
I (1422) cpu_start: cpu freq: 160000000
I (1422) cpu_start: Application information:
I (1425) cpu_start: Project name:     Touchdeck
I (1431) cpu_start: App version:      81c4b3c-dirty
I (1436) cpu_start: Compile time:     Dec  1 2022 21:00:58
I (1442) cpu_start: ELF file SHA256:  29b85afd45876892...
I (1448) cpu_start: ESP-IDF:          v4.4.2-dirty
I (1454) heap_init: Initializing. RAM available for dynamic allocation:
I (1461) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1467) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1473) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1480) heap_init: At 3FFD0AB0 len 0000F550 (61 KiB): DRAM
I (1486) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1492) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1499) heap_init: At 40099E00 len 00006200 (24 KiB): IRAM
I (1505) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1514) spi_flash: detected chip: generic
I (1518) spi_flash: flash io: dio
I (1524) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1533) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1573) BTDM_INIT: BT controller compile version [bf9c077]
I (1573) system_api: Base MAC address is not set
I (1573) system_api: read default base MAC address from EFUSE
I (1583) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2263) esp_bt_gap_cb: BT GAP event: 10
I (2263) bt_init: register hid device callback
I (2263) bt_init: starting hid device
I (2273) esp_bt_hidd_cb: setting hid parameters
I (2273) esp_bt_gap_cb: BT GAP event: 10
I (2273) esp_bt_hidd_cb: setting hid parameters success!
I (2283) esp_bt_hidd_cb: setting to connectable, discoverable
I (2323) Reconnect_HIDD: Trying to connect to b0:fc:36:f4:37:7a
I (2323) Reconnect_HIDD: Connected to last remembered device!
I (2333) esp_bt_hidd_cb: connecting...
I (2343) send_macro_task: started
I (2343) app_main: NVS partition initialized
I (2353) lvgl_helpers: Display hor size: 480, ver size: 320
I (2353) lvgl_helpers: Display buffer size: 19200
I (2353) lvgl_helpers: Initializing SPI master for display
I (2363) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (2363) lvgl_helpers: MISO pin: -1, MOSI pin: 13, SCLK pin: 14, IO2/WP pin: -1, IO3/HD pin: -1
I (2373) lvgl_helpers: Max transfer size: 57600 (bytes)
I (2383) lvgl_helpers: Initializing SPI bus...
I (2383) disp_spi: Adding SPI device
I (2393) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 15
I (2603) ILI9488: ILI9488 initialization.
I (2903) ILI9488: Enabling backlight.
I (2903) ILI9488: Display orientation: LANDSCAPE
I (2903) ILI9488: 0x36 command value: 0x28
I (2903) lvgl_helpers: Initializing SPI master for touch
I (2913) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (2923) lvgl_helpers: MISO pin: 19, MOSI pin: 23, SCLK pin: 18, IO2/WP pin: -1, IO3/HD pin: -1
I (2923) lvgl_helpers: Max transfer size: 0 (bytes)
I (2933) lvgl_helpers: Initializing SPI bus...
I (2943) XPT2046: XPT2046 Initialization
I (2943) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2963) main_menu: initialized
I (2983) button_menu: ID: 1 initialized
I (3003) button_menu: ID: 2 initialized
W (7463) BT_HIDD: hidd_l2cif_connect_cfm: connection failed, now disconnect
W (7463) BT_HIDD: hidd_conn_disconnect: already disconnected
W (7473) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
```

boblane1, 2022-12-05:
Hi @artik15,

The Bluetooth stack on ESP32 is asynchronous, so you need to check the real connection state in the HID Device event callback in the `ESP_HIDD_OPEN_EVT`, like the following code shows:
```c
    ...
    case ESP_HIDD_OPEN_EVT:
        if (param->open.status == ESP_HIDD_SUCCESS) {
            if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTING) {
                ESP_LOGI(TAG, "connecting...");
            } else if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTED) {
                ESP_LOGI(TAG, "connected to %02x:%02x:%02x:%02x:%02x:%02x", param->open.bd_addr[0],
                         param->open.bd_addr[1], param->open.bd_addr[2], param->open.bd_addr[3], param->open.bd_addr[4],
                         param->open.bd_addr[5]);
                bt_app_task_start_up();
                ESP_LOGI(TAG, "making self non-discoverable and non-connectable.");
                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
            } else {
                ESP_LOGE(TAG, "unknown connection status");
            }
        } else {
            ESP_LOGE(TAG, "open failed!");
        }
        break;
    ...
```
The return of `esp_bt_hid_device_connect()` just indicates the result of sending the connection command.

artik15, 2022-12-05:
Hi @boblane1,
Thanks for the answer. So there is no other way to get close to instant status of the connection when trying to connect? If not then what is the event that is called when I send the connect request but the connection is not established? Just `ESP_HIDD_OPEN_EVT`? So a working solution would be to send a msg to the reconnect_hidd() task to try connecting again.
Another thing is that when I use the API function `esp_bt_hid_device_connect()`  ESP blocks the connection and I cannot connect with a real device. Is there a way to force the connection at that time?

Thanks in advance!

artik15, 2022-12-08:
@boblane1 
Sorry I haven't answered earlier, was busy.
How can I use the patch? Please explain or give me a guide. Thanks!

boblane1, 2022-12-09:
@artik15 

Just use the command `git apply test.patch`.

artik15, 2022-12-23:
Sorry, been away. Managed to create a mechanism that sends clean reports to check the connection. Thanks :D

closed 2022-12-23 by artik15