espressif/esp-idf #10307: Guru meditation on nvs_close, nvs_erase and nvs_deinit

ValterMinute, 2022-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

Invoking nvs_close, nvs_flash_deinit and nvs_flash_erase should perform the required operation without crashing the system.

### What is the actual behavior?

Sometimes I get a Guru meditation when invoking nvs_close, nvs_flash_deinit and nvs_flash_erase (I call them to clear configuration on my device).
The issue does not happen 100% of the times but for sure it happens only when I made some changes to the data (using nvs_set_blob or nvs_set_str followed by nvs_commit)

The error message is the following:

Guru Meditation Error: Core 0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core 0 register dump:
MEPC : 0x42035a72 RA : 0x42035a4e SP : 0x3fc9ea50 GP : 0x3fc96a00
0x42035a72: intrusive_list::erase(intrusive_list::iterator) at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/intrusive_list.h:195
(inlined by) nvs_close at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:308

0x42035a4e: nvs_close at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:302

TP : 0x3fc77e38 T0 : 0x4200ed76 T1 : 0x10000000 T2 : 0x00000000

nvs_api address may change, but all the errors reference intrusive_list::erase(intrusive_list::iterator).

### Steps to reproduce.

I tried with a plain reproducer, performing all the operations in sequence, but I didn't manage to get a crash.

### Debug Logs.

```plain
Guru Meditation Error: Core 0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core 0 register dump:
MEPC : 0x42035a72 RA : 0x42035a4e SP : 0x3fc9ea50 GP : 0x3fc96a00
0x42035a72: intrusive_list::erase(intrusive_list::iterator) at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/intrusive_list.h:195
(inlined by) nvs_close at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:308

0x42035a4e: nvs_close at C:/Users/valter.minute/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:302

TP : 0x3fc77e38 T0 : 0x4200ed76 T1 : 0x10000000 T2 : 0x00000000
```

### More Information.

Currently, my application is calling nvs_flash_init at startup, it uses wifi (doing scans only) and BT. 
Sometimes it writes data to NVS (configuration parameters written using nvs_set_blog/nvs_set_string and nvs_commit).
I have a command to clean configuration that invokes nvs_close, nvs_flash_deinit, and nvs_flash_erase to remove all NVS content and this is where I get the error.
nvs_flash_init and nvs_open on different namespaces (2) are invoked at startup.
Flash write is called from different tasks, but there are no concurrent calls.
When a reset to factory command is received via BT the application closes the namespaces, de-initializes flash (nvs_flash_deinit), and erases it (nvs_flash_erase), forcing a reboot afterward.
I tried also to invoke nvs_flash_erase without first closing/deinitializing, but the result didn't change, unfortunately.

ValterMinute, 2022-12-05:
I forgot to attach registers values to the dump:
S0/FP   : 0x3fcc80b4  S1      : 0x3fc9eaa0  A0      : 0x00000001  A1      : 0xffffffff
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000007  A5      : 0x3fcc80e4
A6      : 0x60023000  A7      : 0x0000000a  S2      : 0x3fc9fb38  S3      : 0x3fc9ea98
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x0000000b

igrr, 2022-12-05:
@ValterMinute Could you please attach the complete stack trace you get after the register dump? (You can get it if you run the program using `idf.py monitor` or the equivalent option in an IDE.)
In the absence your .elf file, the register values do not provide a lot of useful information, unfortunately.

> I tried with a plain reproducer, performing all the operations in sequence, but I didn't manage to get a crash.

Could you try to describe what is different between your application and a "plain reproducer", to try to narrow down the possible cause of the issue? For example, are you using NVS APIs from multiple application tasks? Is the amount of heap memory always sufficient? Have you tried enabling heap poisoning to check if this could be caused by heap corruption? This or any other information might help us make this issue more actionable.

ValterMinute, 2022-12-06:
I am investigating heap corruption, trying to narrow this down in my code base.
Thanks a lot for the hint.
I think that the crash is a symptom, not the real issue.

closed 2022-12-06 by ValterMinute