espressif/esp-idf #10309: On crash the backtrace log in terminal output cannot be parsed by address resolver

sabbath24, 2022-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

On crash the backtrace log in terminal output cannot be parsed, because the logger in [components\esp_system\port\arch\xtensa\debug_helpers.c](https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/arch/xtensa/debug_helpers.c) does not put a space between the fist and second stacktrace enrty fieds. I am fixed this problem in my local machine, just patch the debug_helpers.c.
How can I create a pull-request to send this patch?

igrr, 2022-12-05:
@sabbath24 which IDF version are you using?

There was a PR https://github.com/espressif/esp-idf/pull/9138 related to fixing the backtrace formatting, we have merged it recently: https://github.com/espressif/esp-idf/commit/9b8c558e63d95b491355a22e69f247834e899b47.

sabbath24, 2022-12-05:
My version is v4.4.2 is was pulled by the espidf vscode extension.

My patch for the file is:
```
C:\Users\sabbath\esp\esp-idf>git diff components/esp_system/port/arch/xtensa/debug_helpers.c
diff --git a/components/esp_system/port/arch/xtensa/debug_helpers.c b/components/esp_system/port/arch/xtensa/debug_helpers.c
index 0a819d5d73..af26b8a493 100644
--- a/components/esp_system/port/arch/xtensa/debug_helpers.c
+++ b/components/esp_system/port/arch/xtensa/debug_helpers.c
@@ -87,8 +87,8 @@ esp_err_t IRAM_ATTR esp_backtrace_print_from_frame(int depth, const esp_backtrac
         if (!esp_backtrace_get_next_frame(&stk_frame)) {    //Get previous stack frame
             corrupted = true;
         }
-        print_entry(esp_cpu_process_stack_pc(stk_frame.pc), stk_frame.sp, panic);
         print_str(" ", panic);
+        print_entry(esp_cpu_process_stack_pc(stk_frame.pc), stk_frame.sp, panic);
     }

     //Print backtrace termination marker

C:\Users\sabbath\esp\esp-idf>
```

sabbath24, 2022-12-05:
Okay I see now the https://github.com/espressif/esp-idf/commit/9b8c558e63d95b491355a22e69f247834e899b47 commit, so they fixed in the v5.1 development branch, in the printenrty function... I am sorry but I did not realize this earlier.

igrr, 2022-12-05:
@sabbath24 No problem, I can try to backport this commit release/v4.4. Will update this issue when the backport is done!

igrr, 2022-12-05:
@sabbath24 Please ignore my previous comment. The fix is also in release/v4.4, commit is https://github.com/espressif/esp-idf/commit/68dd8997. This commit is already part of v4.4.3 bugfix release.

closed 2022-12-05 by igrr