espressif/esp-idf #10310: error: implicit declaration of function 'soc_has_cache_lock_bug' when ESP_INT_WDT option is disabled

stkw0, 2022-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It compiles the software successfully. 

### What is the actual behavior?

The next error is returned

```
esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c: In function 'esp_startup_start_app':
esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:486:13: error: implicit declaration of function 'soc_has_cache_lock_bug' [-Werror=implicit-function-declaration]
  486 |     assert(!soc_has_cache_lock_bug() && "ESP32 Rev 3 + Dual Core + PSRAM requires INT WDT enabled in project config!");
```

### Steps to reproduce.

1. Make a copy of the hello world example
2. Run idf.py menuconfig and disable ESP_INT_WDT
3. Run idf.py build

### Build or installation Logs.

```plain
esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c: In function 'esp_startup_start_app':
esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:486:13: error: implicit declaration of function 'soc_has_cache_lock_bug' [-Werror=implicit-function-declaration]
  486 |     assert(!soc_has_cache_lock_bug() && "ESP32 Rev 3 + Dual Core + PSRAM requires INT WDT enabled in project config!");
```
```

ma89vik, 2022-12-06:
Hi,

The issue seems to be that `esp_chip_info.h` is missing from `port.c`, we'll fix that.

As for making `INT_WDT` depend on `SPIRAM` in kconfig, I think the problem here is that we do not know the ECO version at compile time. so this will have to stay as a run-time check for now I think.

ESP-Marius, 2022-12-06:
Hi, 

The issue seems to be that esp_chip_info.h is missing from port.c, we'll fix that. 

As for making INT_WDT depend on SPIRAM in kconfig, I think the problem here is that we do not know the ECO version at compile time. so this will have to stay as a run-time check for now I think.

closed 2022-12-06 by espressif-bot