espressif/esp-idf #10312: Unable to reflash ESP32 on encrypted flash enabled

tsctrl, 2022-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

i was unable to flash ESP32 on encrypted flash enabled, i have tried using command mentioned in the documentation, build my own encrypted firmware and flash all with the same results. i did not have this issue, or basically its work on older 4.3.0 idf version

step to replicate: in development mode:
1. enable secure boot. burn bootloader and flash keys.
2. generate secure bootloader with same key: idf.py bootloader
2. reflasing bootloader: esptool.py 0x0 build/bootloader/bootloader-reflash-digest.bin
3. reflash all partition (exclude bootloader) with non encrypted firmware: esptool.py 0x10000 build\partition_table\partition-table.bin 0x18000 build\ota_data_initial.bin 0x20000 build\firmware.bin
4. check if device are bootable. Yes its bootable and the hardware are work fine until this step!
5. enable flash encryption development mode in menuconfig and build.
[image]
6. run summary to check bits and burn FLASH_CRYPT_CONFIG efuse.
[image]
7. run command to encrypt firmware: espsecure.py encrypt_flash_data --keyfile build/app_keys.bin --address 0x1000 -o bootloader_encrypted.bin build\bootloader\bootloader.bin
8. restart 

other approach command use with the same result: 
1. use idf.py encrypted-app-flash monitor to build and encrypt the images in step 7.
2. run command to encrypt firmware: espsecure.py encrypt_flash_data --keyfile /key.bin --address 0x1000 -o bootloader_encrypted.bin build\bootloader\bootloader.bin and flash bootloader_encrypted.bin to 0x1000 in step 7

result:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```
documentation is here:
[flash-encryption](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#flash-encryption-configuration)

tsctrl, 2022-12-05:
last time i could see the flash encryption processing status after i restart, but now seem like it didnt do it

chipweinberger, 2022-12-05:
You forgot to mention, which ESP-IDF are you using?

tsctrl, 2022-12-06:
idf 4.4.1

after playaround with the 'behavior' of the encryption mechanism. few rule which need to obey here...
few issue i notice here where the eFuse does not increment automatically on reboot after i reflash the plaintext for the secureboot bootloader and all partition with plaintext. enabling the fuse to encrypted does not encrypt the firmware during boot. i have change the flash encryption to enable, rebuild and flash the plaintext and left the cnt to even number. not sure what was the mechanism to force the esp to encrypt the plaintext during restart or maybe only for apps partition and not bootloader?

so all approach above to disable the encryption up to 3 times above and reflash plaintext with automatic esp encryption during restart are not possible or having issue. i am looking to have this approach as it does not require us to encrypt all partition each.
for now i have resolve this issue by flashing the encrypted digest and all partitions. and also noticed i have to use the apps key and not by using the bootloader key to encrypt the bootloader digest.

may close this issue, possible this is the correct behavior or possible enhancement or clarification.

tsctrl, 2022-12-06:
possible with plaintext bootloader digest will not be encrypted by hardware on restart. only for apps partition? so need manual encrypted bootloader digest to be flash first then the restart and auto CNT efuse increment and firmware encryption during restart will work?

mahavirj, 2022-12-06:
@tsctrl 

Based on the `FLASH_CRYPT_CNT` value as 0x7, device would expect encrypted images on the flash (includes bootloader, partition table, application). Please ensure that you encrypt the images on the host and then flash them. Please see a relevant issue here https://github.com/espressif/esp-idf/issues/9499

You may also use `merge_bin` option of `esptool` to create a single binary file and then encrypt it before flashing at offset 0x0.

mahavirj, 2022-12-14:
Closing the issue, please feel free to re-open if you have any questions

closed 2022-12-14 by mahavirj