espressif/esp-idf #10313: invalid "info" message ?

dannybackx, 2022-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I sometimes get a message like
I (19:59:02.637) HTTP_CLIENT: Body received in fetch header state, 0x3ffd8c02, 254

which I suspect should be ESP_LOGD, not ESP_LOGI ?

Either that, or documentation should describe what this means.

This is on a client library that does https calls to an internet based service.

### What is the actual behavior?

sends this message

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
D (19:59:02.545) HTTP_CLIENT: HEADER=Content-Type:application/json
D (19:59:02.551) Acme: HttpEvent: header Content-Type value application/json
D (19:59:02.558) HTTP_CLIENT: HEADER=Content-Length:659
D (19:59:02.563) Acme: HttpEvent: header Content-Length value 659
D (19:59:02.569) HTTP_CLIENT: HEADER=Connection:keep-alive
D (19:59:02.575) Acme: HttpEvent: header Connection value keep-alive
D (19:59:02.581) HTTP_CLIENT: HEADER=Cache-Control:public, max-age=0, no-cache
D (19:59:02.589) Acme: HttpEvent: header Cache-Control value public, max-age=0, no-cache
D (19:59:02.597) HTTP_CLIENT: HEADER=X-Frame-Options:DENY
D (19:59:02.602) Acme: HttpEvent: header X-Frame-Options value DENY
D (19:59:02.609) HTTP_CLIENT: HEADER=Strict-Transport-Security:max-age=604800
D (19:59:02.616) Acme: HttpEvent: header Strict-Transport-Security value max-age=604800
D (19:59:02.624) HTTP_CLIENT: http_on_headers_complete, status=200, offset=258, nread=258
D (19:59:02.632) HTTP_CLIENT: http_on_body 254
I (19:59:02.637) HTTP_CLIENT: Body received in fetch header state, 0x3ffd8c02, 254
D (19:59:02.645) Acme: HttpEvent HTTP_EVENT_ON_DATA (len 254)
D (19:59:02.651) HTTP_CLIENT: content_length = 659
D (19:59:02.656) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=659
D (19:59:02.663) transport_base: remain data in cache, need to read again
D (19:59:02.670) HTTP_CLIENT: need_read=405, byte_to_read=405, rlen=405, ridx=254
D (19:59:02.677) HTTP_CLIENT: http_on_body 405
D (19:59:02.682) Acme: HttpEvent HTTP_EVENT_ON_DATA (len 405)
D (19:59:02.688) HTTP_CLIENT: http_on_message_complete, parser=3ffd84bc
D (19:59:02.694) Acme: PerformWebQuery -> {
  "RzI4oPJpCfE": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417",
  "keyChange": "https://acme-v02.api.letsencrypt.org/acme/key-change",
  "meta": {
    "caaIdentities": [
      "letsencrypt.org"
    ],
    "termsOfService": "https://letsencrypt.org/documents/LE-SA-v1.3-September-21-2022.pdf",
```

### More Information.

Porting an existing application from 4.4 to 5.0

hmalpani, 2022-12-26:
Hello @dannybackx 
This info log was added as it appears in some cases when the body is also received with the header when performing an HTTP request. This was set as an info log intentionally. It shouldn't affect any application as such.

dannybackx, 2022-12-26:
so please document it

closed 2023-01-16 by espressif-bot