espressif/esp-idf #10319: I don't know DFU download mode PHY behavior in doc. And I don't enter the DFU mode in windows

WangShuoran, 2022-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When in ESP32S3, the doc show:
By default, the USB_SERIAL_JTAG module is connected to the ESP32-S3’s internal USB PHY, while the USB OTG peripheral can be used only if an external USB PHY is connected. Since DFU is provided via the USB OTG peripheral, it cannot be used through the internal PHY in this configuration.

Does this mean that if DFU is used, the value of efuse must be changed?

And if USB_PHY_SEL change, the USB_SERIAL_JTAG can not available permanent?

If permanent unavailable, I think in the doc must give any warning.

igrr, 2022-12-07:
FWIW, I think the next sentence on that page already says that this change is permanent:

> However, users can permanently switch the internal USB PHY to work with USB OTG peripheral instead of USB_SERIAL_JTAG by burning the USB_PHY_SEL eFuse.

WangShuoran, 2022-12-09:
I'm sorry I'm stupid. But I google any same keyword but I do not know any behavior in esp32s3.
In Technical Reference Manual:
[image]
And
[image]
[image]

In the technical manual, I understand that the default connection behavior of the internal integrated transceiver is modified by using efuse, and another peripheral device should be available later, because the integrated transceiver is shared through time division multiplexing.
在技术手册中，我理解到的是，使用efuse修改的是内部集成收发器默认连接的行为，应该后续另一个外设也可以使用，是因为通过时分复用共用集成收发器。

DFU can not follow this rule? When the esp32s3 open, it will can not print something with USB_SERIAL_JTAG?

igrr, 2022-12-09:
There is a single USB PHY (transceiver) in the chip. Either the USB_OTG peripheral or USB_SERIAL_JTAG peripheral can be internally connected to the built-in USB PHY. Software can switch the PHY between the two peripherals (which i guess is what the TRM refers to when it says about the time division multiplexing.) However, the USB host (PC) will see different devices on the bus depending on which USB peripheral is connected, and will have to perform enumeration each time. It will not look to the PC user that CDC and DFU devices are simultaneously available in the system.

If you need to use USB_SERIAL_JTAG and USB_OTG peripherals concurrently, you can connect one of them to an external USB PHY (transceiver) chip. One example of such chip is SP5301.

WangShuoran, 2022-12-09:
I seem to understand something. I forgot about USB host (PC) perform enumeration.
But if as you say. when I use tinyusb let esp32s3 become USB slave. It must config USB_PHY_SEL? But I remember I didn't burn USB_PHY_SEL, and the USB host (PC) can Interact with tinyusb.

So, when esp32s3 uses tinyusb as the slave, does it automatically set the default connection of the internal integrated transceiver to USB OTG peripheral?
所以，在esp32s3使用tinyusb作为从机时，是否自动完成了设置内部集成收发器默认连接为USB OTG peripheral?

When esp32s3 is used as a USB host, do you complete the above similar behaviors?
在esp32s3作为USB Host的时候，是否也完成了上面类似行为？

leeebo, 2023-12-05:
@WangShuoran When you initialize the USB Device driver or USB Host driver, the USB-PHY will be switched to the USB-OTG during initialization

Back to your question: If the ESP32-S3 wants to implement DFU downloads, you need to burn efuse to switch the USB-PHY connection to the USB-OTG, because the download functions is implement in the ROM Code (which cannot be changed).

But please take care before you burn the efuse bit：For ESP32-S3 series chips manufactured before DateCode 2219 and modules and development boards with the PW No. before PW-2022-06-XXXX, the EFUSE_DIS_USB_OTG_DOWNLOAD_MODE (BLK0 B19[7]) bit of eFuse is set by default and cannot be modified. Therefore, the USB-OTG-Download function is unavailable for these products. (espressif has been released PCN about the issue)

StevenMacias, 2023-12-11:
@leeebo Thanks for your comment. I lost a couple of days trying to make my ESP32-S3-DevKitC-1 to go into DFU mode. For those interested on how to figure out the Date Code you can check the 'Data Matrix Convention' section on the [Module Packaging Information](https://www.espressif.com/sites/default/files/documentation/espressif_module_packaging_information_en.pdf). Then I installed the 'Code Scan - Scan any bar code' app and read the Data Matrix engraved in the SoC package. In my case my DevKit was from `2149` and indeed the `USB_OTG_DOWNLOAD_MODE` was set to 1:

```
espefuse.py --port /dev/ttyACM1 summary
...
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Set this bit to disable download through USB-OTG   = True R/W (0b1)
...
```
I tried to find the PCN that you mentioned but I couldn't. Can you share the link if available? Thanks!

leeebo, 2023-12-12:
@StevenMacias I can't find the exact [PCN](https://www.espressif.com/en/support/documents/pcns) either, but find the description in [esp32-s3_errata](https://www.espressif.com/sites/default/files/documentation/esp32-s3_errata_en.pdf)

igrr, 2023-12-12:
It is this one, it was published as an advisory because fixing the previously-not-working USB download modes could have some security implications: https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-004%20Security%20Advisory%20for%20USB_OTG%20%26%20USB_Serial_JTAG%20Download%20Functions%20of%20ESP32-S3%20Series%20Products%20EN.pdf

tore-espressif, 2024-12-16:
The original issue seems to be resolved, we will close this issue.

Feel free to reopen if you have any more qurestions