espressif/esp-idf #10320: [ESP32S3] No reboot after panic on Core 1 (Core 0 works fine)

nx518, 2022-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

External 3.3V

### What is the expected behavior?

Clean Reboot after Crash

### What is the actual behavior?

ESP32S3 Core 1 ->  ❌ freezes / hangs after "Rebooting..." , no reboot
ESP32S3 Core 0 -> ✅works fine
ESP32 Core 1    -> ✅works fine
ESP32 Core 0    -> ✅works fine

### Steps to reproduce.

Run this code, to provoke panic. 
If it runs on core 1 it freezes / hangs, on core 0 it works fine.

```
#include <stdint.h>
#include <string.h>

#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_task.h"

#define TASK_PINNED_CORE_ID     1   /*< Run panic task on core 1 */

void vTaskCode(void *pvParameters)
{
    for (;;)
    {
        int test = *(int *)0x1; //*(int *)0x1 = 0;
        ESP_LOGI("vTaskCode", "value of address: %d", test);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main()
{
    TaskHandle_t xHandle = NULL;
    xTaskCreatePinnedToCore(vTaskCode, "LoadProhibitedTask", 2048, NULL, 10, &xHandle, TASK_PINNED_CORE_ID);

    while (1)
    {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
```

### Debug Logs.

```plain
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
I (25) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 16:46:01
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) boot_comm: chip revision: 1, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=083a4h ( 33700) map
I (112) esp_image: segment 1: paddr=000183cc vaddr=3fc91000 size=02a70h ( 10864) load
I (117) esp_image: segment 2: paddr=0001ae44 vaddr=40374000 size=051d4h ( 20948) load
I (128) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18f94h (102292) map
I (150) esp_image: segment 4: paddr=00038fbc vaddr=403791d4 size=07dd4h ( 32212) load
I (157) esp_image: segment 5: paddr=00040d98 vaddr=50000000 size=00010h (    16) load
I (163) boot: Loaded app from partition at offset 0x10000
I (163) boot: Disabling RNG early entropy source...
I (178) cpu_start: Pro cpu up.
I (179) cpu_start: Starting app cpu, entry point is 0x4037518c
0x4037518c: call_start_cpu1 at /home/pi/git/Firmware/180718/esp-idf_ABL/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (193) cpu_start: Pro cpu start user code
I (193) cpu_start: cpu freq: 160000000 Hz
I (193) cpu_start: Application information:
I (196) cpu_start: Project name:     180718_Test_Panic_LoadProhibite
I (203) cpu_start: App version:      7a61bf3-dirty
I (209) cpu_start: Compile time:     Dec  6 2022 16:47:21
I (215) cpu_start: ELF file SHA256:  2a1167f3cd2221c3...
I (221) cpu_start: ESP-IDF:          v5.0
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (233) heap_init: At 3FC94480 len 00055290 (340 KiB): D/IRAM
I (239) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (246) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (252) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (259) spi_flash: detected chip: generic
I (263) spi_flash: flash io: dio
W (267) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the bi
nary image header.
I (281) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420078e1  PS      : 0x00060030  A0      : 0x8037c4bc  A1      : 0x3fcea4d0
0x420078e1: vTaskCode at /home/pi/git/Firmware/180718_Test_Panic_LoadProhibited/main/main.c:15

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8037c7b6  A9      : 0x00000001
A10     : 0x00060023  A11     : 0x00000003  A12     : 0x00060023  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x420078de:0x3fcea4d0 0x4037c4b9:0x3fcea4f0
0x420078de: vTaskCode at /home/pi/git/Firmware/180718_Test_Panic_LoadProhibited/main/main.c:12

0x4037c4b9: vPortTaskWrapper at /home/pi/git/Firmware/180718/esp-idf_ABL/components/freertos/FreeRTOS-Kernel/portable/xt
ensa/port.c:151

ELF file SHA256: 2a1167f3cd2221c3

Rebooting...
```

### More Information.

Never recovers after "Rebooting..."

esp-wzh, 2022-12-13:
@nx518 Thanks for your report!
This bug been fixed on master by https://github.com/espressif/esp-idf/commit/0d345c814a178ce71fc2877e7e44497450967b1e and will be backported to release/v5.0 & release/v4.4  soon. If there's any problem feel free to reopen this issue!

AxelLin, 2022-12-13:
@esp-wzh 
From the commit of the fix, it looks like the issue has impact on all dual-core SoCs.
Could you confirm if v4.3 also needs fix?

igrr, 2022-12-13:
The bug affected ESP32-S3 only. (CONFIG_IDF_TARGET_ESP32 was defined for ESP32, but not defined for ESP32-S3; changing that to !CONFIG_FREERTOS_UNICORE fixed the issue.)
For this reason, we won't backport this to release/v4.3. That release only had preview support for ESP32-S3-beta2.
For the other two branches, the fix will be released in v4.4.5 and v5.0.1.

AxelLin, 2022-12-14:
> The bug affected ESP32-S3 only. (CONFIG_IDF_TARGET_ESP32 was defined for ESP32, but not defined for ESP32-S3; changing that to !CONFIG_FREERTOS_UNICORE fixed the issue.) For this reason, we won't backport this to release/v4.3. 

Because the change is not just changing CONFIG_IDF_TARGET_ESP32 to !CONFIG_FREERTOS_UNICORE.
It also changed the code from "esp_cpu_unstall(PRO_CPU_NUM);" to "unstall all other cores".
Anyway thanks for clarify this bug affected esp32-s3 only.

QuentinFarizon, 2023-04-05:
@esp-wzh I am on v5.0.1, that include thix fix, but I still get a full freeze (no reboot) when there is a crash on Core 1 :
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420b340b  PS      : 0x00060630  A0      : 0x8200c917  A1      : 0x3fcaede0  
0x420b340b: esp_http_client_set_method at /home/quentin/esp/esp-idf/components/esp_http_client/esp_http_client.c:984

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3c0de960  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3c0de95a  A8      : 0x8204dc3c  A9      : 0x3fcaeda0  
A10     : 0x00000000  A11     : 0x3c0c63c5  A12     : 0x3c0deb16  A13     : 0x3fc9dc90  
A14     : 0x3c0c63c5  A15     : 0xff000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000006c  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x420b3408:0x3fcaede0 0x4200c914:0x3fcaee00
0x420b3408: esp_http_client_set_method at /home/quentin/esp/esp-idf/components/esp_http_client/esp_http_client.c:983

0x4200c914: check_and_perform_dfu_task at /home/quentin/Code/Boks/boks-link/main/dfu/dfu.c:125 (discriminator 13)

ELF file SHA256: 6ba4d053f6140d04

CPU halted.
```

I reproduced by passing a badly formed URL to esp_http_client_perform

igrr, 2023-04-05:
> CPU halted.

Could you please attach your sdkconfig @QuentinFarizon?

QuentinFarizon, 2023-04-05:
Sorry @igrr , it was a misconfigured CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT :+1:

MrDexterX, 2023-08-27:
Is it possible to configure CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT using Arduino IDE?