espressif/esp-idf #10321: The panic_abort is triggered when Bluetooth and WiFi mesh are turned on at the same time

wishinlife, 2022-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

sh_

### Power Supply used.

USB

### What is the expected behavior?

Not occurred panic_abort.

### What is the actual behavior?

If the CONFIG_SW_COEXIST_ENABLE is turned on, there are two scenarios in which a panic_abort is triggered:
1. Bluetooth is start after wifi mesh;
2. wifi mesh set  ps disable.

If the CONFIG_SW_COEXIST_ENABLE is turned off, the bluetooth start after wifi mesh and set wifi mesh ps disable will not occurred panic_abort.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

wmy-espressif, 2022-12-09:
Hi @wishinlife , will you please answer my questions:
1. Which chip are you using, ESP32, ESP32-C3 or others?
2. Do you have backtrace/coredump of the panic abort, and can you provide the .elf file for your project? I would like to investigate where the software crashes

wmy-espressif, 2022-12-12:
Hi @wishinlife , I have confirmed with my colleague. Wi-Fi mesh relies on Wi-Fi SoftAP and STAtion roles. The coexistence of  Wi-Fi SoftAP and Bluetooth is not well supported at current stage, so you may meet severe performance issues for Wi-Fi mesh and Bluetooth, even when the crash issue is resolved. 

Therefore, please don't assume that Wi-Fi Mesh and Bluetooth coexistence scenario is well supported, and don't use the simultaneously. 

For this issue, I guess the root cause is related to this operation: `wifi mesh set ps disable`, in the source code, when coexistence is enabled, we don't allow the Wi-Fi power save mode to be set to ps disable, the crash is probably caused by a parameter check in an assert statement. To verify this, you can try not do this setting: `Wi-Fi ps disable`

wishinlife, 2022-12-20:
@wmy-espressif Hi, i use esp32-s3 chip, if set CONFIG_SW_COEXIST_ENABLE turned off or Wi-Fi ps is enable , this issue will not occur.
 Is Wi-Fi mesh networking and Bluetooth coexistence not officially recommended?

atemiz, 2022-12-22:
@wmy-espressif We also want to use Wi-Fi Mesh and BLE together in our current project (ESP32-C3, IDF v5.0). 
It seems to work with mesh PS function, although there are stability issues. 
Will support for coexistence of Wi-Fi SoftAP & BLE become stable in the future or is this issue due to hardware limitations and will never be fully stable?

wmy-espressif, 2022-12-26:
Hi @wishinlife ,

> Hi, i use esp32-s3 chip, if set CONFIG_SW_COEXIST_ENABLE turned off or Wi-Fi ps is enable , this issue will not occur.

The option `CONFIG_SW_COEXIST_ENABLE` is required to be enabled for all espressif chips if you want to use coexistence. There was once a so called `hardware coexistence mode` for ESP32 but is deprecated now.

> Is Wi-Fi mesh networking and Bluetooth coexistence not officially recommended?

Not it is not recommended. Is I had explained earlier, Wi-Fi mesh relies on softAP + station, but the coexistence performance in case of Wi-Fi softAP scenario is not well supported yet at current stage. We cannot ensure a specific level of performance in that coexistence condition for now.

AxelLin, 2022-12-26:
> Not it is not recommended. Is I had explained earlier, Wi-Fi mesh relies on softAP + station, but the coexistence performance in case of Wi-Fi softAP scenario is not well supported yet at current [...]

@wmy-espressif 

It's known issue since 2020, https://github.com/espressif/esp-idf/issues/4940#issuecomment-636586168
It has been 2 years, so I'm surprise that coexistence with softAP is still not well supported at current stage.
Is this hardware limitation that there is no chance to improve it in the future?

wishinlife, 2022-12-27:
> 

Hi @wmy-espressif, Thank you for your answer.
My need is to configure the parameters of the device via Bluetooth, and there is less time for data transmission. At the same time, the use of wifi mesh is to collect data from the device, the bandwidth requirements are not high, can exceed 1Mb/s can be, but we have high requirements for the stable work of the chip, as long as it can run stably, the data transmission performance is lower can also meet my needs, can the current situation meet my use requirements?

Espressif-liuuuu, 2022-12-27:
hi @wishinlife , thanks for your question. I saw you raised this question in https://github.com/espressif/esp-idf/issues/5357, and sorry for the late response, lets talk about it here.  
We checked the firmware internally and found the limitation check, which leads to assert, did happen if your chip works as a root in mesh with BLE at the same time. However, Im not sure why its OK if BLE starts before Wi-Fi. Im stiil wondering the order of calling so that we could reproduce it locally. You could simpily drag the ZIP of project into dialog box, or take the project in your own repo and refer it here. many thanks:).
And your need could be meet if no heavily BT in use, e.g no BLE scan with almost same interval and window, no heavy traffic in BLE connection. 
The difficulty of coexistence with softAP and BLE is that there isnt too many approaches now for softAP to stop traffic from station, which is neccessary in heavy traffic secanaries. We did some demo for couple of customers and the solution is in test. It would be supported in next year so that the coexistence of softAP and BT would be officially guaranteed. @atemiz @AxelLin

wishinlife, 2023-01-05:
Hi @Espressif-liuuuu, it is only necessary to start bluetooth controller first, and then start wifi mesh as a root, which will not trigger this assert.

Espressif-liuuuu, 2023-01-13:
hi, @wishinlife , We have cleared the situation of the assert, let me clarify that:
Basically there is limitation that Wi-Fi ps must be enabled if Wi-Fi and BT both enabled, the check would be done at the momemt  you change sleep type, and the momemt both modules enabled. Assertion happens when the check failed.
However, the check wouldnt be done after Wi-Fi mesh start, beacause of some internal reason. So if you would like to disable mesh ps or Wi-Fi ps, pls do the configuration in orders below:
1. Wi-Fi configurations (keep ps on) and BT configurations. The order could be switched.
2. `esp_mesh_start` and bt controller enable
3. set ps mode to 0 to close ps.

In that case, the check would not be done, and the assert would not happen.

Leadrive, 2023-04-29:
@Espressif-liuuuu 过去半年了，请问这个问题解决了吗？

Espressif-liuuuu, 2023-05-22:
hi @Leadrive , 您指的是mesh共存时触发crash的问题吗? 这个问题可以通过
https://github.com/espressif/esp-idf/issues/10321#issuecomment-1381709670
的方式避免

Leadrive, 2023-06-03:
@Espressif-liuuuu 不行的，就是按这个来，有些板子正常，有些就一直报异常。关键是这个异常还导致重启。想避免都不行。
估计是你们出厂固件写死了，想修复 都修复不了。
````
I (11329) wifi:Set ps type: 0

E (11329) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

abort() was called at PC 0x42117633 on core 0
0x42117633: pm_set_sleep_type at ??:?
                                                                                                                                                                                                                                   

Backtrace: 0x40375d8e:0x3fcf53c0 0x40382b0d:0x3fcf53e0 0x40388ec2:0x3fcf5400 0x42117633:0x3fcf5470 0x4210606c:0x3fcf5490 0x42104452:0x3fcf54b0 0x4214a275:0x3fcf54d0
0x40375d8e: panic_abort at /home/develop/esp/esp-idf/components/esp_system/panic.c:408
                                                                                                                                                                                                                                   
0x40382b0d: esp_system_abort at /home/develop/esp/esp-idf/components/esp_system/esp_system.c:137
                                                                                                                                                                                                                                   
0x40388ec2: abort at /home/develop/esp/esp-idf/components/newlib/abort.c:46
                                                                                                                                                                                                                                   
0x42117633: pm_set_sleep_type at ??:?
                                                                                                                                                                                                                                   
0x4210606c: wifi_set_ps_process at ??:?
                                                                                                                                                                                                                                   
0x42104452: ieee80211_ioctl_process at ??:?
                                                                                                                                                                                                                                   
0x4214a275: ppTask at ??:?
````

Leadrive, 2023-06-03:
@Espressif-liuuuu 要构建wifi-mesh，大家都一样配置才行，否则会出现连不上，断连的情况。
这个小配置，真是搞死人，麻烦得很

Leadrive, 2023-06-03:
非root，在启动mesh后设置不会报异常。root节点，等连上外网，时间同步后再设置，就不报错了。

Espressif-liuuuu, 2023-06-05:
@wishinlife  好的, 已了解您的需求. 这里的限制已经在修改中, 计划改成warning和其他自适应的方式, 在几周内应该可以sync至github.

KonssnoK, 2023-08-11:
> hi @wishinlife , thanks for your question. I saw you raised this question in #5357, and sorry for the late response, lets talk about it here. We checked the firmware internally and found the [...]

hello @Espressif-liuuuu ,
we would be interested in this coexistence too.
Does it require specific code from the latest branches (we use v4.4) to try this coexistence or is it enough to use the workaround you said above ?
Thanks

Espressif-liuuuu, 2023-08-14:
> @Leadrive 好的, 已了解您的需求. 这里的限制已经在修改中, 计划改成warning和其他自适应的方式, 在几周内应该可以sync至github.

Hi @Leadrive , 这一限制已在以下master commit中取消
https://github.com/espressif/esp-idf/commit/851446909211813b5e3c0beb1baa0d9cf43959b3

Espressif-liuuuu, 2023-08-14:
> hello @Espressif-liuuuu , we would be interested in this coexistence too. Does it require specific code from the latest branches (we use v4.4) to try this coexistence or is it enough to use the [...]

Hi @KonssnoK , which one do you want to know more? Do you mean the limitation of Wi-Fi sleep, or about the softAP strategy?  
The limitation has been removed on master, and you may stiil need to call API in specifical order on old branch to avoid the issue.  
If you want to konw more about the softAP strategy, lets open a new ticket and talk about the detail :)
Thanks

closed 2023-08-14 by Alvin1Zhang

KonssnoK, 2023-08-14:
@Espressif-liuuuu 

Well, we would like to know more about both :)

Limitation of wifi sleep:

We currently use Wifi-Mesh without PS. 
We would like to read a BLE beacon once every few minutes.
If we try to put PS on the mesh it will completely disrupt the latency of the network itself.
We just want to passively read a beacon. Could this be an issue for the softAP traffic?

Solution on the coexistence:
Yes, we would be interested, do i need to to specify anything in particular in the new ticket?