espressif/esp-idf #10322: Build issue when using compiler size optimization option and maximum log level is debug or verbose

deviousprophet, 2022-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Build without issue

### What is the actual behavior?

Fail to build with an unexpected error as below:
```
.../esp-idf/components/hal/esp32/include/hal/ledc_ll.h:57:27: error: 'glb_clk' may be used uninitialized in this function [-Werror=maybe-uninitialized]
```

### Steps to reproduce.

1. Create an example project using VSCode
2. `idf.py menuconfig`
3. `Compiler Options` -> `Optimization Level` -> `Optimize for size (-Os)`
4. `Component config` -> `Log output` -> `Maximum log verbosity` -> `Debug` or `Verbose`
5. Save config
6. Build

### Build or installation Logs.

```plain
-- Adding linker script C:/Users/tripr/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Users/tripr/esp/esp-idf/components/bootloader C:/Users/tripr/esp/esp-idf/components/bootloader_support C:/Users/tripr/esp/esp-idf/components/efuse C:/Users/tripr/esp/esp-idf/components/esp_app_format C:/Users/tripr/esp/esp-idf/components/esp_common C:/Users/tripr/esp/esp-idf/components/esp_hw_support C:/Users/tripr/esp/esp-idf/components/esp_rom C:/Users/tripr/esp/esp-idf/components/esp_system C:/Users/tripr/esp/esp-idf/components/esptool_py C:/Users/tripr/esp/esp-idf/components/freertos C:/Users/tripr/esp/esp-idf/components/hal C:/Users/tripr/esp/esp-idf/components/log C:/Users/tripr/esp/esp-idf/components/bootloader/subproject/main C:/Users/tripr/esp/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/tripr/esp/esp-idf/components/newlib C:/Users/tripr/esp/esp-idf/components/partition_table C:/Users/tripr/esp/esp-idf/components/soc C:/Users/tripr/esp/esp-idf/components/spi_flash C:/Users/tripr/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/tripr/Desktop/sample_project/build/bootloader
[650/840] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/ledc.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/ledc.c.obj
C:\Users\tripr\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -IC:/Users/tripr/Desktop/sample_project/build/config -IC:/Users/tripr/esp/esp-idf/components/driver/include -IC:/Users/tripr/esp/esp-idf/components/driver/deprecated -IC:/Users/tripr/esp/esp-idf/components/driver/esp32/include -IC:/Users/tripr/esp/esp-idf/components/driver/include/driver -IC:/Users/tripr/esp/esp-idf/components/newlib/platform_include -IC:/Users/tripr/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/tripr/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/tripr/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/tripr/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/tripr/esp/esp-idf/components/esp_hw_support/include -IC:/Users/tripr/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/tripr/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/tripr/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/tripr/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -IC:/Users/tripr/esp/esp-idf/components/heap/include -IC:/Users/tripr/esp/esp-idf/components/log/include -IC:/Users/tripr/esp/esp-idf/components/soc/include -IC:/Users/tripr/esp/esp-idf/components/soc/esp32/. -IC:/Users/tripr/esp/esp-idf/components/soc/esp32/include -IC:/Users/tripr/esp/esp-idf/components/hal/esp32/include -IC:/Users/tripr/esp/esp-idf/components/hal/include -IC:/Users/tripr/esp/esp-idf/components/hal/platform_port/include -IC:/Users/tripr/esp/esp-idf/components/esp_rom/include -IC:/Users/tripr/esp/esp-idf/components/esp_rom/include/esp32 -IC:/Users/tripr/esp/esp-idf/components/esp_rom/esp32 -IC:/Users/tripr/esp/esp-idf/components/esp_common/include -IC:/Users/tripr/esp/esp-idf/components/esp_system/include -IC:/Users/tripr/esp/esp-idf/components/esp_system/port/soc -IC:/Users/tripr/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/tripr/esp/esp-idf/components/xtensa/include -IC:/Users/tripr/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/tripr/esp/esp-idf/components/lwip/include -IC:/Users/tripr/esp/esp-idf/components/lwip/include/apps -IC:/Users/tripr/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/tripr/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/tripr/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/tripr/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/tripr/esp/esp-idf/components/esp_pm/include -IC:/Users/tripr/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/tripr/esp/esp-idf/components/efuse/include -IC:/Users/tripr/esp/esp-idf/components/efuse/esp32/include -IC:/Users/tripr/esp/esp-idf/components/esp_timer/include -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Os -freorder-blocks -fmacro-prefix-map=C:/Users/tripr/Desktop/sample_project=. -fmacro-prefix-map=C:/Users/tripr/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/ledc.c.obj -MF esp-idf\driver\CMakeFiles\__idf_driver.dir\ledc.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/ledc.c.obj -c C:/Users/tripr/esp/esp-idf/components/driver/ledc.c
In file included from C:/Users/tripr/esp/esp-idf/components/hal/include/hal/ledc_hal.h:18,
                 from C:/Users/tripr/esp/esp-idf/components/driver/ledc.c:16:
C:/Users/tripr/esp/esp-idf/components/driver/ledc.c: In function 'ledc_set_timer_div':
C:/Users/tripr/esp/esp-idf/components/hal/esp32/include/hal/ledc_ll.h:57:27: error: 'glb_clk' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   57 |     hw->conf.slow_clk_sel = slow_clk_sel;
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:/Users/tripr/esp/esp-idf/components/driver/ledc.c:496:25: note: 'glb_clk' was declared here
  496 |     ledc_slow_clk_sel_t glb_clk;
      |                         ^~~~~~~
cc1.exe: some warnings being treated as errors
[663/840] Performing build step for 'bootloader'
[1/101] Generating project_elf_src_esp32.c
[2/101] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj
[3/101] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2c_periph.c.obj
```

AxelLin, 2022-12-11:
The same compile error in v4.3 branch if build with debug.

listout, 2023-01-02:
Can we please have a fix for this? I cannot move my projects from v4.4.3 to v5.0 due this issue.

closed 2023-01-13 by espressif-bot