espressif/esp-idf #10325: usb_host_interface_claim throw an error when connected a special tsc printer on esp32-s3

yourchanges, 2022-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

here is the test code:
```
void check_interface_desc_printer(const void *p)
{
  const usb_intf_desc_t *intf = (const usb_intf_desc_t *)p;

  // USB Printer Class Specification 1.1
  if ((intf->bInterfaceClass == USB_CLASS_PRINTER) &&
      (intf->bInterfaceSubClass == 1))
  {
    /* Protocol
     * 00 Reserved, undefined.
     * 01 Unidirectional interface.
     * 02 Bi-directional interface.
     * 03 1284.4 compatible bi-directional interface.
     * 04-FEh Reserved for future use.
     * FFh Vendor-specific printers do not use class-specific protocols.
     */
    // No idea how to support 1284.4 so ignore it.
    if ((intf->bInterfaceProtocol == 1) || (intf->bInterfaceProtocol == 2)) {
      isBiDirectional = (intf->bInterfaceProtocol == 2);
      if (isBiDirectional) {
        if (intf->bNumEndpoints < 2) {
          isPrinter = false;
          return;
        }
      }
      else {
        if (intf->bNumEndpoints < 1) {
          isPrinter = false;
          return;
        }
      }
      isPrinter = true;
      ESP_LOGI("", "Claiming a %s-directional printer!", (isBiDirectional)?"bi":"uni");
      esp_err_t err = usb_host_interface_claim(Client_Handle, Device_Handle,
          intf->bInterfaceNumber, intf->bAlternateSetting);
      if (err != ESP_OK) ESP_LOGI("", "usb_host_interface_claim failed: %x", err);
    }
  }
}
```
here is output
```
esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c54
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x2900
entry 0x403c98c0
[    86][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   133][I][usbhhelp.hpp:79] usbh_setup(): [] usb_host_install: 0
[   133][I][usbhhelp.hpp:90] usbh_setup(): [] usb_host_client_register: 0
[   518][I][usbhhelp.hpp:40] _client_event_callback(): [] New device address: 1
[   518][I][usbhhelp.hpp:49] _client_event_callback(): [] speed: 1 dev_addr 1 vMaxPacketSize0 64 bConfigurationValue 1
[   524][I][show_desc.hpp:44] show_dev_desc(): [] bLength: 18
[   529][I][show_desc.hpp:45] show_dev_desc(): [] bDescriptorType(device): 1
[   536][I][show_desc.hpp:46] show_dev_desc(): [] bcdUSB: 0x110
[   542][I][show_desc.hpp:47] show_dev_desc(): [] bDeviceClass: 0x00
[   548][I][show_desc.hpp:48] show_dev_desc(): [] bDeviceSubClass: 0x00
[   554][I][show_desc.hpp:49] show_dev_desc(): [] bDeviceProtocol: 0x00
[   561][I][show_desc.hpp:50] show_dev_desc(): [] bMaxPacketSize0: 64
[   567][I][show_desc.hpp:51] show_dev_desc(): [] idVendor: 0x483
[   573][I][show_desc.hpp:52] show_dev_desc(): [] idProduct: 0x5720
[   579][I][show_desc.hpp:53] show_dev_desc(): [] bcdDevice: 0x0
[   584][I][show_desc.hpp:54] show_dev_desc(): [] iManufacturer: 1
[   590][I][show_desc.hpp:55] show_dev_desc(): [] iProduct: 2
[   596][I][show_desc.hpp:56] show_dev_desc(): [] iSerialNumber: 1
[   602][I][show_desc.hpp:57] show_dev_desc(): [] bNumConfigurations: 1
[   608][I][show_desc.hpp:64] show_config_desc(): [] bLength: 9
[   614][I][show_desc.hpp:65] show_config_desc(): [] bDescriptorType(config): 2
[   621][I][show_desc.hpp:66] show_config_desc(): [] wTotalLength: 32
[   627][I][show_desc.hpp:67] show_config_desc(): [] bNumInterfaces: 1
[   633][I][show_desc.hpp:68] show_config_desc(): [] bConfigurationValue: 1
[   640][I][show_desc.hpp:69] show_config_desc(): [] iConfiguration: 0
[   646][I][show_desc.hpp:74] show_config_desc(): [] bmAttributes(Self Powered): 0xc0
[   653][I][show_desc.hpp:75] show_config_desc(): [] bMaxPower: 50 = 100 mA
[   660][I][show_desc.hpp:82] show_interface_desc(): [] bLength: 9
[   666][I][show_desc.hpp:83] show_interface_desc(): [] bDescriptorType (interface): 4
[   674][I][show_desc.hpp:84] show_interface_desc(): [] bInterfaceNumber: 1
[   680][I][show_desc.hpp:85] show_interface_desc(): [] bAlternateSetting: 0
[   687][I][show_desc.hpp:86] show_interface_desc(): [] bNumEndpoints: 2
[   694][I][show_desc.hpp:87] show_interface_desc(): [] bInterfaceClass: 0x07
[   700][I][show_desc.hpp:88] show_interface_desc(): [] bInterfaceSubClass: 0x01
[   708][I][show_desc.hpp:89] show_interface_desc(): [] bInterfaceProtocol: 0x02
[   715][I][show_desc.hpp:90] show_interface_desc(): [] iInterface: 0
[   721][I][usbhprinter.ino:67] check_interface_desc_printer(): [] Claiming a bi-directional printer!
[   730][I][usbhprinter.ino:70] check_interface_desc_printer(): [] usb_host_interface_claim failed: 105
[   739][I][show_desc.hpp:100] show_endpoint_desc(): [] bLength: 7
[   745][I][show_desc.hpp:101] show_endpoint_desc(): [] bDescriptorType (endpoint): 5
[   752][I][show_desc.hpp:104] show_endpoint_desc(): [] bEndpointAddress(In): 0x81
[   760][I][show_desc.hpp:107] show_endpoint_desc(): [] bmAttributes(Bulk): 0x02
[   767][I][show_desc.hpp:108] show_endpoint_desc(): [] wMaxPacketSize: 64
[   773][I][show_desc.hpp:109] show_endpoint_desc(): [] bInterval: 0
[   779][I][usbhprinter.ino:99] prepare_endpoints(): [] usb_host_transfer_submit In fail: 105
[   788][I][show_desc.hpp:100] show_endpoint_desc(): [] bLength: 7
[   794][I][show_desc.hpp:101] show_endpoint_desc(): [] bDescriptorType (endpoint): 5
[   801][I][show_desc.hpp:104] show_endpoint_desc(): [] bEndpointAddress(Out): 0x03
[   809][I][show_desc.hpp:107] show_endpoint_desc(): [] bmAttributes(Bulk): 0x02
[   816][I][show_desc.hpp:108] show_endpoint_desc(): [] wMaxPacketSize: 64
[   822][I][show_desc.hpp:109] show_endpoint_desc(): [] bInterval: 0
[   828][I][usbhprinter.ino:109] prepare_endpoints(): [] Out data_buffer_size: 512
[257770][I][usbhprinter.ino:179] loop(): [] usb_host_transfer_submit Out fail: 105

```
here is the related code:
```
esp_err_t err = usb_host_interface_claim(gUsbClientHandle(), gUsbDeviceHandle(),
        intf->bInterfaceNumber, intf->bAlternateSetting);
```

so  when we call usb write, there also be an error "[257770][I][usbhprinter.ino:179] loop(): [] usb_host_transfer_submit Out fail: 105"

any ideas about this problem?

yourchanges, 2022-12-14:
follow the https://www.beyondlogic.org/usbnutshell/usb5.shtml  I think mostly info is included in the pre-post logs, and here is the info when the printer connected a linux machine:
```
[2022-12 14 11:19:14 2022] usb 1-2.2: new full-speed USB device number 20 using xhci_hcd
[2022-12 14 11:19:14 2022] usb 1-2.2: config 1 has an invalid interface number: 1 but max is 0
[2022-12 14 11:19:14 2022] usb 1-2.2: config 1 has no interface number 0
[2022-12 14 11:19:14 2022] usb 1-2.2: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 0.00
[2022-12 14 11:19:14 2022] usb 1-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=1
[2022-12 14 11:19:14 2022] usb 1-2.2: Product: Datong Printer
[2022-12 14 11:19:14 2022] usb 1-2.2: Manufacturer: Datongsmart
[2022-12 14 11:19:14 2022] usb 1-2.2: SerialNumber: Datongsmart
[2022-12 14 11:19:14 2022] usblp 1-2.2:1.1: usblp2: USB Bidirectional printer dev 20 if 1 alt 0 proto 2 vid 0x0483 pid 0x5720

```

Dazza0, 2022-12-14:
@yourchanges 

```
[   674][I][show_desc.hpp:84] show_interface_desc(): [] bInterfaceNumber: 1
```
```
[2022-12 14 11:19:14 2022] usb 1-2.2: config 1 has no interface number 0
```
From your log, it looks like the interface's number (i.e., `bInterfaceNumber` is `1`) but you're attempting to claim interface number `0`.

yourchanges, 2022-12-15:
Our code already use what we just got:
```
esp_err_t err = usb_host_interface_claim(gUsbClientHandle(), gUsbDeviceHandle(),
        intf->bInterfaceNumber, intf->bAlternateSetting);
```

so should we force the bInterfaceNumber to 0?

yourchanges, 2022-12-15:
It seems the not working printer is a full-speed printer, but the worked printer is just the high usb printer, it related or not?

tore-espressif, 2025-06-25:
@yourchanges I sorry for this issue not being answered in very long time.

There was a bug in older versions of esp-idf that could not open interfaces if they were not indexed from 1. This is long fixed.

Is this issue still relevant?

yourchanges, 2025-06-25:
this issue can be closed

closed 2025-06-25 by tore-espressif