espressif/esp-idf #10326: 'How to Improve Wi-Fi Performance' needs improvement

higaski, 2022-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The section 'How to Improve Wi-Fi Performance' in the [WiFi Driver documentation](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-guides/wifi.html#how-to-improve-wi-fi-performance) needs some improvement.

There are some mistakes e.g. the docs say to set the value LWIP_IRAM_OPTIMIZATION to 13 in order to optimize throughput yet that value is a boolean one and can therefor only be set 0 or 1. The section specifically about PSRAM usage mentions the values TCP_SND_BUF_DEFAULT and TCP_WND_DEFAULT and suggests to set them to 85K. However this option can only be enabled once CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP has been enabled.

hansw123, 2024-03-04:
hi @higaski 
Sorry for the delay in getting back to you, but we've received your question and will fix the problem in the next update, where the 13 in LWIP_IRAM_OPTIMIZATION stands for the amount of ram space used to turn on the option.

Indastri, 2024-10-14:
Hi, This issue should be solved now. Closing, if the issue persists please reopen this or create a new issue.

Cheers!

closed 2024-10-14 by Indastri