espressif/esp-idf #10330: Failed to read coredump

whong0951, 2022-12-08:
OS: Linux
IDF version: esp-idf V4.4.3
Error traceback:
```
espcoredump.py v0.4-dev
Traceback (most recent call last):
  File "/home/whong/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/construct/lib/containers.py", line 100, in __getattr__
    return self[name]
KeyError: 'crc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/espcoredump.py", line 350, in <module>
    temp_core_files = info_corefile()
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/espcoredump.py", line 149, in info_corefile
    core_elf_path, target, temp_files = get_core_dump_elf(e_machine=exe_elf.e_machine)
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/espcoredump.py", line 75, in get_core_dump_elf
    loader.create_corefile(exe_name=args.prog, e_machine=e_machine)
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/corefile/loader.py", line 239, in create_corefile
    self._validate_dump_file()
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/corefile/loader.py", line 214, in _validate_dump_file
    self._crc_validate()
  File "/home/whong/workspace/esp/esp-idf-v4.4/components/espcoredump/corefile/loader.py", line 223, in _crc_validate
    'Invalid core dump CRC %x, should be %x' % (data_crc, self.core_src.crc))  # type: ignore
  File "/home/whong/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/construct/lib/containers.py", line 102, in __getattr__
    raise AttributeError(name)
AttributeError: crc
```

Problem code:
https://github.com/espressif/esp-idf/blob/6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7/components/espcoredump/corefile/loader.py#L218-L223

fix method:
replace **self.core_src.crc** with **self.core_src.checksum** in line 223.

devprofile98, 2023-07-29:
Hi, did you find the problem?

erhankur, 2023-07-29:
@whong0951 can you share your `sdkconfig` and `partition_table.csv` files?

closed 2023-08-04 by espressif-bot