espressif/esp-idf #10333: ble_gap_adv_start Returns with Error BLE_ERR_INV_HCI_CMD_PARMS on ESP32S3

matt001k, 2022-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an application that utilizes a custom BLE GATTS interface for provisioning a device.
This exact application works on an ESP32 but when porting the code to an ESP32S3 the same application always reports this error.
What I have done to troubleshoot:
- Compare bleperph example code to my current implementation to notice any blaring differences
- Upgrade IDF version from 4.4.2 to 5.0
- Validated SDKConfigs are the same between the ESP32 app and the ESP32S3 app
- Ensured I am not sending extended advertising data

I can attempt to show some of my code but will require approval first.
If anyone else has experienced this please let me know of any workarounds.

xiewenxiang, 2022-12-09:
The error code `BLE_ERR_INV_HCI_CMD_PARMS ` indicates the the parameter invalid, so you can modify it based on the ESP-IDF BLE_50 demo, or you can provide your current code for review.

matt001k, 2022-12-09:
`void BLEGAPAdvertiseConfig(uint8_t *advData, uint32_t advDataLen, uint8_t *scanRespData, uint32_t scanRespDataLen)
{
    esp_err_t ret;

    ret = ble_gap_adv_set_data(advData, advDataLen);
    if (ret != ESP_OK)
    {
        LogInterfaceLogInfo(BLE_GAP_TAG, "Cannot set advertising data err: %d", ret);
    }

    ret = ble_gap_adv_rsp_set_data(scanRespData, scanRespDataLen);
    if (ret != ESP_OK)
    {
        LogInterfaceLogInfo(BLE_GAP_TAG, "Cannot set scan response data err: %d", ret);
    }
}

void BLEGAPAdvertiseStart(bleGAPAdvertiseParameters_t advParameters)
{
    esp_err_t ret;
    struct ble_gap_adv_params advStartCnf;

    if (!BLEReady())
    {
        BLE_NOT_READY;
    }

    advStartCnf.channel_map = advParameters.channelSel;
    advStartCnf.conn_mode = advParameters.connectionMode;
    advStartCnf.disc_mode = advParameters.discoverableMode;
    advStartCnf.filter_policy = advParameters.filterPolicy;
    advStartCnf.high_duty_cycle = advParameters.highDutyOn;
    advStartCnf.itvl_max = advParameters.intervalMax;
    advStartCnf.itvl_min = advParameters.intervalMin;
    ret = ble_gap_adv_start(advParameters.bdaAddressType, (const ble_addr_t *)advParameters.bdaDirect, advParameters.duration,
                            &advStartCnf, gapEventHandler, NULL);
    if (ret != ESP_OK)
    {
        LogInterfaceLogInfo(BLE_GAP_TAG, "Cannot start advertising err: %d", ret);
    }

    if (bleGAPCb != NULL)
    {
        bleGAPCb(BLE_GAP_EVENT_ADV_START, (bleGAPEventInfo_t)
        {
            0
        });
    }
}
`

Here is a snippet of our code.
BLEGAPAdvertise is called with `advData` of `advDataLen` at 31 bytes and `scanRespData` of `scanRespDataLen` of 31 bytes.
Both  `ble_gap_adv_set_data` and `ble_gap_adv_rsp_set_data` return 0 meaning the parameters are ok.
When calling `BLEGAPAdvertiseStart` the following is configured:
- `ble_gap_adv_params`
    - `.conn_mode` = `BLE_GAP_CONN_MODE_UND`
    - `.disc_mode` = `BLE_GAP_DISC_MODE_GEN`
    - `.itvl_min` = `0x20-0xF`
    - `.itvl_max` = `0x20+0xF`
    - `.channel_map` = `ADV_CHNL_ALL`
    - `.filter_policy` = `BLE_SCAN_FILTER_ALLOW_ALL`
    - `.high_duty_cycle` = `false`
- `own_addr_type` = `BLE_ADDR_TYPE_PUBLIC`
- `direct_addr` = `NULL`
- `duration_ms` = `INT32_MAX`
- `cb` = `gapEventHandler`
- `cb_arg` = `NULL`
We are returning 0x212 from `ble_gap_adv_start`.

Hope this provides enough information...

rahult-github, 2022-12-30:
Hi @matt001k ,
 
 `.itvl_min = 0x20-0xF
.itvl_max = 0x20+0xF`

Here,  itvl_min gets set to 0x11 and itvl_max gets set to 0x2F . 

Can you please update the itvl_min to atleast 0x0020 and try ?  I think, it is out of the range of the values supported in spec . 

[image]

rahult-github, 2023-04-07:
@matt001k   ,  Thanks for reporting, would you please help share if any further updates for the issue?

rahult-github, 2023-05-12:
Closing this issue. Feel free to reopen in case of further comments

closed 2023-05-12 by rahult-github