espressif/esp-idf #10335: [Flash Encryption][v4.4.3][ESP32-S3] Encryption gets stuck when virtual fuses are enabled?

chipweinberger, 2022-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-215-g9f3f82f54b

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

Flash gets encrypted on first boot.

### What is the actual behavior?

We get stuck after this log:

`I (50828) boot: Disabling RNG early entropy source...`

### Steps to reproduce.

1. I think it might have to do with virtual fuses being enabled
2. I enable flash encryption (Release Mode)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10191884/sdkconfig.txt)

### Debug Logs.

```plain
export IDF_PATH=/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_monitor.py -p /dev/cu.SLAB_USBto%                                                                               
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % export IDF_PATH=/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_e
nv/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_monitor.py -p /dev/cu.SLAB_USBtoUART -b 460800 --toolchain
-prefix xtensa-esp32s3-elf- --target esp32s3 /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jJamAp
p.elf
--- idf_monitor on /dev/cu.SLAB_USBtoUART 460800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
xx�xxxx��x��x��x��x�x�xx�������x��xx�������x�����������x�x�xxx��x�x�x��x�xx�x���������������x�x�x��x�x�x���x��xx��xxxxx�x��x�x�xx�x�x�x�x���xx�x��xxx�xx�xx�xxxx�x�xx�x�xx�x�x�xxx�xx�x��xxx�xxxx����xxxxx��xx�x�x�xxx��x��x�xx��x�x�xxxx��x��x�x��x�x�x�x�xx�x���x����xx�x��������x�xxx�x���x��������x�x�x�xx�x���x�������x�����x�xxx�x���x��x����x�x�x�xx�x���x�������xx���x�xxx�x���x��x������x�xx�x�x�x�x�x��x�������x���������x�x����x����������x�xxx�x������xx��x�x�x�xxx�x�xx�������x������x�xxxx�xxx���x��x�x�xxxx�x�x�x�x�xxx�x�x�x�x�x�xxx�xxx�xx�x�x�x���xx�xxxx�xxx��x�x�x�x���x�x��x�xx�x�x�x�x�xx�x�xx�x��x�x�x�x�x�x�x�x�xxx�xxx�x��x�xxx�x�x�x�x��x�xxx�x���x�x�x�x�x�x�x��x�xxx�xx�x�x���xxxxxx�x��x�x�xx��x�x�x�xxxx�xxx��x�xx���x�x�x�xx�����x�x�x��x����x����������x�xxx�x������xx�x�x�xxx�xxx����x�xx�x��x�x�x���x��x�x�xxxx�x�x�x�xxx�xxx��x�x�xxxx�x���x�x�x��x���xxxx�x�x�x�xxx�xxx�x���x�x�x��xI (54) boot: ESP-IDF v4.4.3-215-g9f3f82f54b-dirty 2nd stage bootloader
I (55) boot: compile time 20:36:08
I (55) boot: chip revision: 0
I (55) boot.esp32s3: Boot SPI Speed : 80MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (58) boot.esp32s3: SPI Flash Size : 8MB
I (59) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 otadata          OTA data         01 00 00010000 00002000
I (65) boot:  1 phy_init         RF data          01 01 00012000 00001000
[... 106 lines cut ...]
I (33399) flash_encrypt: Done encrypting
I (33399) flash_encrypt: Encrypting partition 21 at offset 0x2d8000 (length 0x4000)...
I (33603) flash_encrypt: Done encrypting
I (33603) flash_encrypt: Encrypting partition 22 at offset 0x2dc000 (length 0x10000)...
I (34417) flash_encrypt: Done encrypting
I (34417) flash_encrypt: Encrypting partition 23 at offset 0x2ec000 (length 0x4000)...
I (34631) flash_encrypt: Done encrypting
I (34631) flash_encrypt: Encrypting partition 24 at offset 0x2f0000 (length 0x4000)...
I (34844) flash_encrypt: Done encrypting
I (34844) flash_encrypt: Encrypting partition 25 at offset 0x2f4000 (length 0x2000)...
I (34956) flash_encrypt: Done encrypting
I (34957) flash_encrypt: Encrypting partition 26 at offset 0x2f6000 (length 0x2000)...
I (35074) flash_encrypt: Done encrypting
I (35074) esp_image: segment 0: paddr=00300020 vaddr=3c190020 size=db628h (898600) map
I (35235) esp_image: segment 1: paddr=003db650 vaddr=3fca2ff0 size=049c8h ( 18888) 
I (35239) esp_image: segment 2: paddr=003e0020 vaddr=42000020 size=18c6c0h (1623744) map
I (35531) esp_image: segment 3: paddr=0056c6e8 vaddr=3fca79b8 size=02c74h ( 11380) 
I (35533) esp_image: segment 4: paddr=0056f364 vaddr=40374000 size=1efe4h (126948) 
I (35556) flash_encrypt: Encrypting partition 28 at offset 0x300000 (length 0x500000)...
I (50825) flash_encrypt: Done encrypting
I (50825) flash_encrypt: Setting CRYPT_CNT for permanent encryption
W (50825) efuse: Virtual efuses enabled: Not really burning eFuses
I (50827) flash_encrypt: Flash encryption completed
I (50828) boot: Disabling RNG early entropy source...
```

chipweinberger, 2022-12-09:
Upon trying to flash again, after disabling virtual fuses, I'm now stuck here:

```
 *  Executing task in folder jamcorder_app: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.SLAB_USBtoUART -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x300000 jJamApp.bin 0xf000 partition_table/partition-table.bin 0x10000 ota_data_initial.bin 

esptool.py v3.3.2
Serial port /dev/cu.SLAB_USBtoUART
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e0:59:60
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00000000 to 0x00009fff...
Flash will be erased from 0x00300000 to 0x0058efff...
Flash will be erased from 0x0000f000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00011fff...
Erasing flash...
Took 0.60s to erase flash block
Wrote 39936 bytes at 0x00000000 in 3.4 seconds (93.4 kbit/s)...
Hash of data verified.
Erasing flash...
Took 8.50s to erase flash block
Wrote 2681856 bytes at 0x00300000 in 233.1 seconds (92.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.09s to erase flash block
Wrote 3072 bytes at 0x0000f000 in 0.2 seconds (99.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.16s to erase flash block
Wrote 8192 bytes at 0x00010000 in 0.7 seconds (97.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

```

chipweinberger, 2022-12-09:
After forcibly rebooting, it appears flash encryption finally succeeded (with virtual fuses disabled):

```
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/pyt
hon_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_monitor.py -p 
/dev/cu.SLAB_USBtoUART -b 460800 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Volumes/User/MBP-Google-Driv
e/jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jJamApp.elf
--- idf_monitor on /dev/cu.SLAB_USBtoUART 460800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
xx�xxxx��x��x��x��x�x�xx�������x��xx�������x�����������x�x�xxx��x�x�x��x�xx�x���������������x�x�x��x�x�x���x��xx��xxxxx�x��x�x�xx�x�x�x�x���xx�x��xxx�xx�xx�xxxx�x�xx�x�xx�x�x�xxx�xx�x��xxx�xxxx����xxxxx��xx�x�x�xxx��x��x�xx��x�x�xxxx��x��x�x��x�x�x�x�xx�x���x����xx�x�x���x�x�xxx�x���x��x���xx��x�x�xx�x���x�������x�����x�xxx�x���x��x���x�x�x�xx�x���x�������xx���x�xxx�x���x�������xx��xx�x�x�x�x�x��x�������x��������x�xI (34) boot: ESP-IDF v4.4.3-215-g9f3f82f54b-dirty 2nd stage bootloader
I (34) boot: compile time 21:05:07
I (35) boot: chip revision: 0
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
I (39) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (41) boot: ## Label            Usage          Type ST Offset   Length
I (43) boot:  0 otadata          OTA data         01 00 00010000 00002000
I (45) boot:  1 phy_init         RF data          01 01 00012000 00001000
I (46) boot:  2 coredump         Unknown data     01 03 00013000 00010000
I (48) boot:  3 nvs_key          NVS keys         01 04 00023000 00001000
I (50) boot:  4 nvs              WiFi data        01 02 00024000 00024000
I (52) boot:  5 jCoredumpExtra   Unknown data     01 06 00048000 00008000
I (54) boot:  6 jBootArgs        Unknown data     01 06 00050000 00004000
I (56) boot:  7 jBootVer         Unknown data     01 06 00054000 00004000
I (58) boot:  8 jDevCmds         Unknown data     01 06 00058000 00004000
I (60) boot:  9 jJfuArgs         Unknown data     01 06 0005c000 00004000
I (61) boot: 10 jJfuApp          OTA app          00 11 00060000 00226000
I (63) boot: 11 jLifeStats       Unknown data     01 06 00286000 0003c000
I (65) boot: 12 jEspDsCaCert     Unknown data     01 06 002c2000 00002000
I (67) boot: 13 jEspDsIssCert    Unknown data     01 06 002c4000 00002000
I (69) boot: 14 jEspDsCipherIv   Unknown data     01 06 002c6000 00002000
I (71) boot: 15 jStsafeCaCert    Unknown data     01 06 002c8000 00002000
I (73) boot: 16 jStsafeIssCert   Unknown data     01 06 002ca000 00002000
I (74) boot: 17 jStsafeHmac      Unknown data     01 06 002cc000 00002000
I (76) boot: 18 jOtaAesKey       Unknown data     01 06 002ce000 00002000
I (78) boot: 19 jDeviceTree      Unknown data     01 06 002d0000 00004000
I (80) boot: 20 jFactory         Unknown data     01 06 002d4000 00004000
I (82) boot: 21 jImmortal        Unknown data     01 06 002d8000 00004000
I (84) boot: 22 jExtra1          Unknown data     01 06 002dc000 00010000
I (86) boot: 23 jExtra2          Unknown data     01 06 002ec000 00004000
I (87) boot: 24 jExtra3          Unknown data     01 06 002f0000 00004000
I (89) boot: 25 jExtra4          Unknown data     01 06 002f4000 00002000
I (91) boot: 26 jExtra5          Unknown data     01 06 002f6000 00002000
I (93) boot: 27 jJamArgs         Unknown data     01 06 002f8000 00004000
I (95) boot: 28 jJamApp          OTA app          00 10 00300000 00500000
I (97) boot: End of partition table
I (98) boot: No factory image, trying OTA 0
I (99) boot: bootloader_utility_get_selected_boot_partition: 0
I (101) boot ver: read json
I (102) boot: found: jBootVer, offset:0x54000 size:16384
I (103) boot ver: read {"jBootVer":"0.2","compileDate":"Dec  8 2022","compileTime":"20:36:10","ESP-IDF":"v4.4.3-215-g9f3f82f54b-dirty"}
W (106) boot ver: need to write idf version
I (108) boot: found: jBootVer, offset:0x54000 size:16384
I (154) boot ver: write: JSON116{"jBootVer":"0.2","compileDate":"Dec  8 2022","compileTime":"21:05:09","ESP-IDF":"v4.4.3-215-g9f3f82f54b-dirty"}
I (155) boot: Clearing ota_data Partition...
I (156) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (262) boot_comm:  0 otadata          data  00010000 00002000 [yes]
I (262) boot_comm:  1 phy_init         data  00012000 00001000 [no]
I (263) boot_comm:  2 coredump         data  00013000 00010000 [no]
I (264) boot_comm:  3 nvs_key          data  00023000 00001000 [no]
I (266) boot_comm:  4 nvs              data  00024000 00024000 [no]
I (268) boot_comm:  5 jCoredumpExtra   data  00048000 00008000 [no]
I (270) boot_comm:  6 jBootArgs        data  00050000 00004000 [no]
I (271) boot_comm:  7 jBootVer         data  00054000 00004000 [no]
I (273) boot_comm:  8 jDevCmds         data  00058000 00004000 [no]
I (275) boot_comm:  9 jJfuArgs         data  0005c000 00004000 [no]
I (277) boot_comm: 11 jLifeStats       data  00286000 0003c000 [no]
[... 177 lines cut ...]
I (2846) cpu_start: App version:      0.5
I (2847) cpu_start: Compile time:     Dec  8 2022 21:04:59
I (2848) cpu_start: ELF file SHA256:  4fe4045fda0dbe9f...
I (2850) cpu_start: ESP-IDF:          v4.4.3-215-g9f3f82f54b-dirty
[... cut ...]
I (3149) sleep: Configure to isolate all GPIO pins in sleep state
I (3149) sleep: Enable automatic switching of GPIO sleep configuration
I (3151) esp_core_dump_flash: Init core dump to flash
I (3153) esp_core_dump_flash: Found partition 'coredump' @ 13000 65536 bytes
I (3155) coexist: coexist rom version e7ae62f
I (3156) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (3158) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (3158) pd main: app_main() called
```

chipweinberger, 2022-12-09:
I'm also seeing that '--before default_reset --after no_reset' is sometimes specified in the VSC commands. might be related.

mahavirj, 2022-12-09:
> I (35556) flash_encrypt: Encrypting partition 28 at offset 0x300000 (length 0x500000)...
I (50825) flash_encrypt: Done encrypting
I (50825) flash_encrypt: Setting CRYPT_CNT for permanent encryption
W (50825) efuse: Virtual efuses enabled: Not really burning eFuses
I (50827) flash_encrypt: Flash encryption completed

For this case, all the contents (partitions) were encrypted on the flash but `FLASH_CRYPT_CNT` was not really updated. Ideally device would reset after this step but since flash encryption was never really enabled, it went to unpack and load the app. At this moment, booting up the application failed because it was already encrypted. This happened as virtual eFuse config was kept enabled.

Probably problem was fixed with correct `FLASH_CRYPT_CNT` getting populated after virtual efuse config was disabled.

Running `espefuse.py --summary --chip <chip>` should give the clear idea on what eFuses were really programmed on the device. For release mode, `esptool.py --no-stub get_security_info` command should help.

chipweinberger, 2022-12-09:
Perhaps we can add a better log or warning for the user?

```
W (50825) efuse: Virtual efuses enabled: Not really burning eFuses
I (50827) flash_encrypt: Flash encryption not completed, as virtual fuses must be disabled
```

chipweinberger, 2022-12-16:
PR: https://github.com/espressif/esp-idf/issues/10335

closed 2022-12-16 by chipweinberger