espressif/esp-idf #10336: wifi mesh set network duty cycle not effective on root

wishinlife, 2022-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Run esp_mesh_set_network_duty_cycle (new_nwk_duty, -1, MESH_PS_NETWORK_DUTY_APPLIED_ENTIRE) on the root node, the duty cycle of the nodes that have joined the network does not change, and only takes effect for the nodes that join later.
According to the documentation, I think I changed the duty cycle of the entire network, was it a bug or did I have to do esp_mesh_set_network_duty_cycle on every node?

esp-idf: v5.0, esp-wifi updated to master branch https://github.com/espressif/esp-idf/commit/6cbedbfa5767a2230e05a9c6696a8b749476f60d

zhangyanjiaoesp, 2024-03-19:
@wishinlife The use of the API seems to be fine, do you still have the problem?

Sherry616, 2024-04-24:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-24 by Sherry616