espressif/esp-idf #10339: esp_wifi_get_config() fail to get sta password after WPS connect

AxelLin, 2022-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2186-g454aeb3a48

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E custom board

### Power Supply used.

USB

### What is the expected behavior?

Test examples/wifi/wps/main/wps.c with WPS_TYPE_PBC.
The example can connect to AP via WPS.

However, when I added below code to check the received SSID and password it shows wrong password setting.

```
diff --git a/examples/wifi/wps/main/wps.c b/examples/wifi/wps/main/wps.c
index 8ba96a3c20..5babf8cbc9 100644
--- a/examples/wifi/wps/main/wps.c
+++ b/examples/wifi/wps/main/wps.c
@@ -108,6 +108,10 @@ static void wifi_event_handler(void* arg, esp_event_base_t event_base,
                  */
                 ESP_ERROR_CHECK(esp_wifi_wps_disable());
                 esp_wifi_connect();
+
+                wifi_config_t conf= { };
+                esp_wifi_get_config(ESP_IF_WIFI_STA, &conf);
+                ESP_LOGW(TAG, "WPS:: SSID=%s PASSWORD=%s", conf.sta.ssid, conf.sta.password);
             }
             break;
         case WIFI_EVENT_STA_WPS_ER_FAILED:

```

I got below message:
W (32960) example_wps: WPS:: SSID=IGSAP PASSWORD=f2f26130fc3c9f9ccbc2e0faf1c3cc33ec5ac5c3a42e5fd3b409ed37862c5d29

Also note the same code works if build with v4.3 branch.

### What is the actual behavior?

I would expect to see the correct password in above output.

### Steps to reproduce.

Just test WPS.

### Debug Logs.

```plain
I (855) example_wps: start wps...
I (856) example_wps: WIFI_EVENT_STA_START
I (8388) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (8889) wifi:state: init -> auth (b0)
I (8893) wifi:state: auth -> assoc (0)
I (8904) wifi:state: assoc -> run (10)
I (28906) wifi:state: run -> init (cc00)
I (28907) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (28909) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (31319) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (31329) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (31331) wifi:state: init -> auth (b0)
I (31337) wifi:state: auth -> assoc (0)
I (31348) wifi:state: assoc -> run (10)
I (31951) wifi:state: run -> init (17a0)
I (31953) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (31955) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (31956) example_wps: Failed to connect!
I (32952) example_wps: WIFI_EVENT_STA_WPS_ER_SUCCESS
W (32960) example_wps: WPS:: SSID=IGSAP PASSWORD=f2f26130fc3c9f9ccbc2e0faf1c3cc33ec5ac5c3a42e5fd3b409ed37862c5d29
I (32967) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (32984) wifi:state: init -> auth (b0)
I (32987) wifi:state: auth -> assoc (0)
I (32999) wifi:state: assoc -> run (10)
I (33008) wifi:connected with IGSAP, aid = 2, channel 1, 40U, bssid = 18:a6:f7:ab:ca:66
I (33008) wifi:security: WPA2-PSK, phy: bgn, rssi: -42
I (33019) wifi:pm start, type: 1

I (33032) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (34024) esp_netif_handlers: sta ip: 192.168.0.100, mask: 255.255.255.0, gw: 192.168.0.1
I (34025) example_wps: got ip: 192.168.0.100
```

AxelLin, 2022-12-09:
Summary of the test with different branch:
v4.3.4-236-g310571138e       OK
v4.4.3-316-ge86181704a       OK
v5.0-159-g75c3a69feb           FAIL
v5.1-dev-2186-g454aeb3a48 FAIL

BTW, I notice a few WIFI_EVENT_STA_DISCONNECTED event before getting WIFI_EVENT_STA_WPS_ER_SUCCESS.
Is that WIFI_EVENT_STA_DISCONNECTED events normal?

AxelLin, 2022-12-12:
BTW, just notice that I only receive AP credential so there is no event_data in my test case.

AxelLin, 2022-12-12:
@kapilkedawat 
1. This is a breaking change because now I cannot get password via esp_wifi_get_config() which was working in v4.x branches.
2. So how to convert it back to raw passowrd?

kapilkedawat, 2022-12-12:
> @kapilkedawat
> 
>     1. This is a breaking change because now I cannot get password via esp_wifi_get_config() which was working in v4.x branches.
there is no changes in esp_wifi_get_config, its supposed to print the values configured using esp_wifi_set_config, In this case since psk is configured(by the AP, not by station), its printing psk.

> 
>     2. So how to convert it back to raw passowrd?

Please note, esp devices are not converting the passpharse to psk, it is done by AP, device is just configuring the values that it got from AP. See this optimization in hostapd for details:
https://w1.fi/cgit/hostap/tree/src/wps/wps_registrar.c#n2546

Incase you prefer to get passphrase, you can use this patch(there is no actual benefit of getting the passphrase instead of psk for any usecase IMO).

--- a/components/wpa_supplicant/esp_supplicant/src/esp_wps.c
+++ b/components/wpa_supplicant/esp_supplicant/src/esp_wps.c
@@ -1400,6 +1400,7 @@ static int wifi_station_wps_init(const esp_wps_config_t *config)
         goto _err;
     }
 
+    cfg.wps->config_methods = WPS_CONFIG_DISPLAY | WPS_CONFIG_PHY_DISPLAY;
     os_memcpy(cfg.wps->uuid, sm->uuid, WPS_UUID_LEN);
     if ((sm->wps = wps_init(&cfg)) == NULL) {         /* alloc wps_data */
         goto _err;

AxelLin, 2022-12-12:
> Please note, esp devices are not converting the passpharse to psk, it is done by AP, device is just configuring the values that it got from AP. See this optimization in hostapd for details: [...]

The user applicatin can show the password.
While testing with the same AP, application built with esp-idf-v5.x shows different password than application built with v4.x.
This is indeed a **breaking change.** (It's nothing to do with benefit or not, it's an unexpected behavior)
You need to provide an user configuration in esp-idf to allow backward compatible to avoid changing user application's behavior.

AxelLin, 2022-12-19:
@kapilkedawat 

Please fix this **regression**, since you are aware it's a breakage. (Fixed by setting cfg.wps->config_methods = WPS_CONFIG_DISPLAY | WPS_CONFIG_PHY_DISPLAY;)
**You cannot break existing application's behavior.**

AxelLin, 2023-01-03:
> Please note, esp devices are not converting the passpharse to psk, it is done by AP, device is just configuring the values that it got from AP. See this optimization in hostapd for details: https://w1.fi/cgit/hostap/tree/src/wps/wps_registrar.c#n2546
> 
The problem is not about AP side.

> Incase you prefer to get passphrase, you can use this patch(there is no actual benefit of getting the passphrase instead of psk for any usecase IMO).

See, your patch shows the change is in device side.
The result is a surprise for users upgrading existing application from v4.x to v5.x.
(especially v4.4 will reach maintenance period soon, some users will rebuild existing application with newer stable branch).

AxelLin, 2023-02-04:
@kapilkedawat

Now the only way to avoid changing application behavor when build with esp-idf-5.0+ is to patch the esp-idf.
Would you fix this regression?

AxelLin, 2023-02-21:
> @kapilkedawat
> 
> Now the only way to avoid changing application behavor when build with esp-idf-5.0+ is to patch the esp-idf. Would you fix this regression?

@kapilkedawat 

I don't get your response for 2 months.
So I'm wondering if it is true that espressif no longer fix regression report now?
In this case, it's not about preference at all.
It's esp-idf-5 changed behavior so application cannot keep the same behavior as built with esp-idf-4.x.

@igrr

AxelLin, 2023-03-21:
Any update? Both v5.0 and v5.1 branches need fix!

closed 2023-04-03 by espressif-bot