espressif/esp-idf #10340: [idf.py] 'flash' command does not support Flash Encryption (Release) Mode, soft bricks device

chipweinberger, 2022-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-215-g9f3f82f54b

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Please see my related issue in **esptool.py:** https://github.com/espressif/esptool/issues/809

`idf.py flash` should detect when your chip in Secure Download Mode and force you to specify a flag `--apply-encryption-key /path/to/key.bin` or `--flash-raw`

At the very least, we should have the ability to set a key-file when using`idf.py flash`.

**Note:** `idf.py encrypted-flash` does not support release mode, and IMO, encryption should be supported in the normal 'idf.py flash' command using flags. 

### What is the actual behavior?

`idf.py flash` flashes the binary as plaintext, causing the device to not boot. Not very user-friendly!

### Steps to reproduce.

1. Enable Flash Encryption (Release)
2. burn all relevant device fuses
3. use 'idf.py flash'

mahavirj, 2022-12-13:
>  If that was true, it should be simple to detect when flash encryption is enabled, and return an error asking to specify a keyfile.

That won't help in the 1st workflow I described in my earlier comment. So idea here is that, once "release" mode of the flash enc is enabled (and it happens on the manufacturing line), only way remains to update the firmware is over the OTA update channel. If we are really need a way to update the firmware over UART DL mode then I think that can always happen using "Development" mode of the flash enc feature.

chipweinberger, 2022-12-13:
Right, for the 1st workflow it would still report an error, telling you to specify a keyfile (which you do not have). or a --force-flash command, to flash anyway (which would hard brick the device).

This still seems like a big improvement. Especially for the 1st workflow, where you do not have the key and could ruin your device.

**Use Case:**  Of course "Development" Mode (3rd workflow...), is the most ideal for development. But I want to produce beta devices for users that use Release Mode (development mode is not secure), and want to retain the ability to fully update them over UART (bootloader, partition table, etc, if I need to.). For this, the current `idf.py` tools  are not ergonomic.

mahavirj, 2023-02-08:
@chipweinberger 

> Use Case 2: Not hard bricking device in 1st workflow. Am I wrong in saying that for the 1st workflow, accidentally calling idf.py flash will ruin your device without warning? I have never tried...

Yes. We have now added a fix for this in the esptool through https://github.com/espressif/esptool/commit/0be5fcd903bdcad5bcb4d16dd13321787ad0a59c

Please check once, and it that looks good then probably we can close this ticket (implementing the host key based encryption can be handled in the project specific IDF extension, as I had commented earlier)

chipweinberger, 2023-02-08:
That PR looks wonderful!

Thank you for circling back and fixing this. That's a great improvement.

> "WARNING: Detected flash encryption and "
"secure download mode enabled."
"Flashing plaintext binary may brick your device! "
"Use --force to override the warning."

I would say: 

"

> "WARNING: Detected flash encryption and "
"secure download mode enabled."
"Flashing plaintext binary may brick your device!"
"Use --force to override this warning after you've ensured the binary is pre-encrypted with host generated keys."

mahavirj, 2023-02-08:
I will put this feedback internally. Interim, docs have bit more information with dedicated section for this: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#encrypted-flash-protection

Thank you for your feedback and suggestions :+1: 

CC @Harshal5

mahavirj, 2023-02-08:
> question - why does secure download mode matter? I would expect we just check for flash encryption release mode. Perhaps that is how we check for release vs development mode?

Yes, the change primarily detects if encrypted writes are still supported through UART DL mode or not. In case, they are supported then potentially device could be recovered, irrespective of the FLASH_CRYPT_CNT state.

closed 2023-02-10 by mahavirj