espressif/esp-idf #10341: ESP-NOW does not workig with WiFi (STA) mode

rajneeshsinha, 2022-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I am working on ESP-IDF release version-4.4 and having requirements to put the ESP32 device on APSTA mode mean, device should work in AP mode (start web page) and in STA mode connect to router to get internet access. With these both mode device work correctly but after initializing ESP-NOW it doesn't works (Rx data call back handler never called).

I have tried following things before posting this issue here:

1. Start AP to launch web page and start STA together to connect with router --> ESP-NOW not working
2. Start only AP to launch web page and init ESP-NOW (start esp-now)  --> ESP-NOW working
3. Start only STA to connect with router and init ESP-NOW (start esp-now)  --> ESP-NOW not working
4. Start only STA to connect with router and read channel number and set this channel number to init ESP-NOW (start esp-now)  --> ESP-NOW not working
5. Start only STA to connect with router and hard-coded channel number to 1 and set this channel number to init ESP-NOW (start esp-now)  --> ESP-NOW not working
6. From above step-1 to step-5 test, I also tested ESPNOW in both station and softap mode  --> ESP-NOW not working

I already checked and following links:
- https://github.com/jonathanrandall/esp32-esp32-now-wifi
- https://github.com/jonasbystrom/ESP-Now-Sensor-system-with-WiFi

Any suggestions will be appreciated.

kriegste, 2022-12-09:
Have you tried to disable modem sleep, i.e.

`esp_wifi_set_ps(WIFI_PS_NONE);`

Xiehanxin, 2022-12-12:
hi @rajneeshsinha. have you tried the ESP-NOW example, and could you please provide some error log

goatzillax, 2023-12-11:
Congratudolences, guys!  This core functionality has been broken for at least a year!

Xiehanxin, 2023-12-20:
hi @goatzillax this issue is fixed in IDF 5.0 and later version. if you want tou use it on 4.3/4.4, you should init esp-now first ,then connect to AP, and the esp-now can send packet normally. and we will backport to 4.4

goatzillax, 2023-12-20:
@Xiehanxin I wonder if there is some sort of miscommunication here.

I'm on 2.0.14 of the ESP32 package which apparently is IDF 4.4  (correct me if I'm wrong).

I just tried moving esp_now_init *before* connecting the *RX* to an AP, and it still fails.  My testcase is otherwise the same.  I press the button, it disconnects, and then the ESP-NOW messages start showing up.

So as far as I can tell, moving esp_now_init before WiFi.begin() is *not* a workaround for IDF 4.4.

I've updated my testcase.

https://github.com/goatzillax/espnow_crosstest

Can you show me the testcase you used to verify functionality?

zhangyanjiaoesp, 2024-03-26:
@goatzillax I noticed that your ESPNOW packets are broadcast packets. When STA is connected to wifi, it will only receive the broadcast packet whose bssid is equal to the associated AP bssid. Therefore, packet receiving fails. Suggest you use unicast to send.

zhangyanjiaoesp, 2024-03-26:
@goatzillax In this https://github.com/goatzillax/espnow_crosstest testcase, I don't see the `esp_now_add_peer()` in the `rx_esp32_c3.ino` if you send unicast.

goatzillax, 2024-03-26:
> @goatzillax In this https://github.com/goatzillax/espnow_crosstest testcase, I don't see the `esp_now_add_peer()` in the `rx_esp32_c3.ino` if you send unicast.

I am almost certain that requirement was removed in the ESP32 version of the libraries.  I tried to make that painfully clear in the file naming -- the receiver is an ESP32-C3 and the transmitter is an ESP8266, and their libraries are not the same although they should be able to communicate with each other over ESP-NOW.

https://github.com/espressif/arduino-esp32/issues/8912

Again, I don't think you understand the bug, testcase, and now the API.

The testcase boots up with the RX in STA mode.  No messages are received.  When I press the button, it disconnects from STA and goes into SoftAP mode, and messages are then received.  All without having called esp_now_add_peer.

This is why I suspect espressif's own testcase aren't being run.  I don't think they'll even compile due to API changes.

goatzillax, 2024-03-26:
Just checked the docs, it does not indicate that esp_now_add_peer is required to *receive* messages.

zhangyanjiaoesp, 2024-03-26:
> Just checked the docs, it does not indicate that esp_now_add_peer is required to _receive_ messages. In fact I don't think that was ever a requirement as I'm looking back at some old code. It was the master/slave notion they did away with.

Sorry, I made a mistake, it's not required to add peer when receiving.

zhangyanjiaoesp, 2024-03-26:
https://github.com/espressif/arduino-esp32/issues/8912 I will test the case on Arduino. By the way, can you please have a try on Arduino 3.0.0 (which is based on IDF v5.1)?

PepeTux, 2024-03-31:
Hello all, [here there is a working code](https://www.esp32.com/viewtopic.php?f=19&t=12992#), it have both transmitter and receiver, running on the ESP32, to set the module as WIFI_STATION and connect to an AP, must be configured as:
`WiFi.mode(WIFI_AP_STA );`

As the [ESP-Now IDF ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html)said:  it must be set as the channel that station or softap is on. 
[image]

The transmitter have a search channel routine to find the receiver channel, 

Tested on a ESP32-C3 (revision v0.4).
Arduino IDE 2.3.2

Enjoy coding !

Jose Luis

zhangyanjiaoesp, 2024-06-03:
@rajneeshsinha @goatzillax do you still have this problem?

goatzillax, 2024-06-04:
I tried the esp_wifi_set_promiscuous workaround that PepeTux mentioned, but that did not work for me.

I don't know what you mean by Arduino 3.0.0.  Do you mean the arduino-esp32 3.0.0 board package?  I tried that version, and it is still broken in the exact same way.

I have not seen a test case in any suite that indicates anybody actually understands the bug in the first place.

No test case == not fixed

leonardodanelutti, 2024-06-07:
I have the same issue, once esp_wifi_connect() is called, esp now stops to work.

zhangyanjiaoesp, 2024-06-12:
> I don't know what you mean by Arduino 3.0.0. Do you mean the arduino-esp32 3.0.0 board package? I tried that version, and it is still broken in the exact same way.

Yes, I mean the arduino-esp32 3.0.0 board package.

zhangyanjiaoesp, 2024-06-12:
> I have the same issue, once esp_wifi_connect() is called, esp now stops to work.

@leonardodanelutti can you provide a demo to test?

leonardodanelutti, 2024-06-12:
Sorry @zhangyanjiaoesp, I am using Rust in my project, but I am doing the same things as @rajneeshsinha and @goatzillax on IDF 5.1:

- start the Wi-Fi as STA
- Start ESP-NOW (at this point, ESP-NOW packets are being received)
- Connect to the Wi-Fi AP.

At this point, ESP-NOW stops working if the AP's channel is not 1.

The problem seems to be that ESP-NOW and the Wi-Fi must be on the same channel, but it is impossible to set a channel other than 1 on the ESP-NOW transmitter (error: peer channel is not equal to home channel, send fail!).

zhangyanjiaoesp, 2024-06-12:
@leonardodanelutti When the STA connects to AP, it will change it's home channel same to the AP.  For example, the default channel for STA is channel 1, when STA connects to a channel 5 AP, the STA will change the home channel to 5. So if the home channel of your STA and AP is different, you need to  call `esp_wifi_set_channel()` to set channel of the STA same to AP, at the same time, the channel of the peers should also be consistent.

zhangyanjiaoesp, 2024-06-12:
@goatzillax I have tested your code on Arduino 3.0.0, and I didn't reproduce your issue.

leonardodanelutti, 2024-06-12:
@zhangyanjiaoesp I know that, but as I said for me it is impossible to change the channel of the peer in the ESP-NOW transmitting device. As soon as I call `esp_now_send` this error is raised: "peer channel is not equal to home channel, send fail!".

zhangyanjiaoesp, 2024-06-12:
@leonardodanelutti The `peer channel` in log `peer channel is not equal to home channel, send fail!` means the `peer->channel` when you call `esp_now_add_peer(peer)`, you can call `esp_now_mod_peer(peer)` to change the `peer->channel` if you have added it.

zhangyanjiaoesp, 2024-06-12:
> > @goatzillax I have tested your code on Arduino 3.0.0, and I didn't reproduce your issue.
> 
> That's interesting. The TX side just sends a message whenever reset is pressed so it's pretty simple.
> 
> So when you ran the RX code, you connected it to a STA, then while still connected to the STA (confirmed with a ping from another machine), you pressed reset on the TX device and saw it blink the LED and/or watched the serial port output and it confirmed "msg received"?

yes

leonardodanelutti, 2024-06-12:
@zhangyanjiaoesp the error message is displayed when I call `esp_now_send` not when I add the peer, even if I set the same channel in the STA config

leonardodanelutti, 2024-06-12:
@zhangyanjiaoesp seem that this should be fixed on [5.2](https://github.com/espressif/esp-idf/releases/tag/v5.2), I will test that

zhangyanjiaoesp, 2024-06-12:
> @zhangyanjiaoesp the error message is displayed when I call `esp_now_send` not when I add the peer, even if I set the same channel in the STA config

@leonardodanelutti I know the log displayed when call `esp_now_send()`, when you want to send data to peer, it will check the channel first. Your issue seems to be caused by inconsistent channels, you only need to set the STA channel, AP channel , peer channel to be consistent can solve the problem.

goatzillax, 2024-06-12:
> yes

Huh.  One last thing -- did you change the TX to transmit to a specific MAC (I.e. make it use unicast)?  I only put the broadcast in as a placeholder.

zhangyanjiaoesp, 2024-06-12:
> Huh. One last thing -- did you change the TX to transmit to a specific MAC (I.e. make it use unicast)? I only put the broadcast in as a placeholder.

yes,  I have changed it to my communicate device MAC

goatzillax, 2024-06-17:
OK.  Well, all I can do is leave this here.

I updated my testcase to make it slightly harder to mess up; all you have to do is set your SSID/password and channel in the RX code.

I literally made a video of what I'm seeing:

https://www.youtube.com/watch?v=dEBRfFWL9lQ

When the code starts up, it connects to an AP and no ESP-NOW messages are received.

I press the button, it disconnects, and it starts receiving messages.

The serial console text from the video:

`
ESP-IDF major 5
ESP-IDF minor 1
MAC Address:  B4:E6:2D:E9:FE:6E
192.168.2.166
Wi-Fi Channel: 3
Initializing ESP-NOW
disconnecting WiFi
msg received
msg received
`

Can you point out where in your test suite this exact functionality is actually tested?

zhangyanjiaoesp, 2025-02-25:
@goatzillax Do you still have this issue?

goatzillax, 2025-02-26:
Last I checked (with arduino-esp32 3.0.0) it was still broken for me.

Was there a specific fix for this added since then?

zhangyanjiaoesp, 2025-02-27:
@goatzillax 
Just retested using your [latest code ](https://github.com/goatzillax/espnow_crosstest/tree/master/rx_esp32_c3) on esp32C3, arduino-esp32 3.0.0, still can't reproduce your issue.
The C3's log is the following:
```
ESP-ROM:esp32c3-eco7-20230720
Build:Jul 20 2023
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2880
entry 0x403cc710

ESP-IDF major 5
ESP-IDF minor 1
B4:E6:2D:E9:FE:6E
192.168.0.198
Wi-Fi Channel: 5

Initializing ESP-NOW
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received
disconnecting WiFi //ESP-NOW data packets can be received regardless of whether C3 is connected or not.
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received
msg received

```

My TX device was using ESP32, and the TX code was using the test example in side, the log is following:
```
I (4498) wifi:mode : sta (4c:75:25:bb:86:8c)
I (4498) wifi:enable tsf

+MODE:OK
+WIFI:STA_START

:>channel -G
+CHANNEL:5,none

:>
:>espnow -I
I (31618) ESPNOW: espnow [version: 2.0] init
+ESPNOW:OK

:>
:>espnow -P -t Add -i 0 -m B4:E6:2D:E9:FE:6E    //add peer

+ESPNOW:OK
espnow -S -m B4:E6:2D:E9:FE:6E -l 100 -n 10 -j 10 // send data to peer, length is 100bytes, number is 10, and send interval is 10ms

+ESPNOWSEND:OK
espnow -S -m B4:E6:2D:E9:FE:6E -l 100 -n 10 -j 10 //send after c3 disconnected

+ESPNOWSEND:OK

```

goatzillax, 2025-02-28:
Still broken for me.  Again, I am sending from an **ESP8266** to an ESP32-C3.

On the **ESP8266** side:

```
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386

fpm close 1 
mode : sta(xx:xx:xx:xx:xx:xx)
add if0
Delivery fail
del if0
usl
enter deep sleep

SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386

fpm close 1 
mode : sta(08:3a:8d:cc:c4:ec)
add if0
Delivery success
del if0
usl
enter deep sleep

```

On the ESP32-C3 Side:

```
ESP-IDF major 5
ESP-IDF minor 3
MAC Address:  B4:E6:2D:E9:FE:6E
192.168.2.166
Wi-Fi Channel: 2
Initializing ESP-NOW
disconnecting WiFi
msg received

```

As you can see I updated all the way to arduino-esp32 3.1.3.

I enabled verbose on the RX side but there were no messages when the message failed.  Obviously whatever is happening isn't being caught by current debugging.

Is there some basic incompatibility with sending ESP-NOW messages from an ESP8266 to an ESP32-C3?

zhangyanjiaoesp, 2025-02-28:
@goatzillax 
Can you try using an ESP32 series device as the sender?

The `tx_esp8266` seems too old, I didn't find the `esp_now_set_self_role()` and `esp_now_add_peer()`(with 5 parameters) in the latest ESP8266 wifi code.

zhangyanjiaoesp, 2025-02-28:
@goatzillax 
Now I can reproduce your problem using ESP8266 for TX (even with the latest version), I will check the reason.

goatzillax, 2025-03-03:
@zhangyanjiaoesp Thanks for debugging this.

Do you know how this would get propagated to the ESP8266 Arduino core?

I understand if ESP8266 is in maintenance mode, but having a properly working ESP-NOW should still give all these devices plenty of useful life.

me-no-dev, 2025-03-04:
@goatzillax Espressif is not the maintainer of the ESP8266 Arduino core. We have no control over it.

zhangyanjiaoesp, 2025-03-11:
@goatzillax Do you have any other questions? If not, we will close this issue.

goatzillax, 2025-03-11:
Is there an actual description of what broke?  I see the description in the commit, but it doesn't seem to make sense to me.  It mentions "broadcast" but my case is specifically unicast.  So I don't understand the fix.

zhangyanjiaoesp, 2025-03-12:
@goatzillax 
Please take a look at the doc (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#frame-format).

[image]

[image]

In the original 8266 code, the addr3 in the mac header is not broadcast address, which cause the problem. The fix change the addr3 in the mac header to broadcast address. This is not the same thing as what you mentioned about sending unicast or broadcast packets.

goatzillax, 2025-03-12:
Ah, I understand now, thanks for the explanation.  That's all I had so if it's fixed go ahead and close it.

closed 2025-03-12 by Alvin1Zhang